sidoh/esp8266_milight_hub #432: After wifi restart lag problems

poudenes, 2019-04-13:
Hi all,

When wifi is reboot (for a reason) and u don't restart the hub then I get lots of lag problems. After reboot the hub and then domoticz all problems are gone.

Question : is it possible to give the hub a reboot command from a http line?

I want do a restart of domoticz, first restart hub then wait 5 sec and then Restart domoticz...

Greets
Peter

sidoh, 2019-04-13:
Huh, weird.  Shouldn't be the case.  Would be curious to see Serial logs after a wifi reboot.

Anyway, yes, it's possible.  Check out the [REST API documentation](https://github.com/sidoh/esp8266_milight_hub#rest-endpoints).  Example curl command:

```
$ curl -X POST -H 'Content-Type: application/json' -d '{"command":"restart"}' http://milight-hub/system
```

poudenes, 2019-04-13:
Thanks,  

Serial log is not easy. Module is build in somewhere so can't take it out easily :)

sidoh, 2019-04-13:
Cool.  Let me know if that changes.  Will close for now. :)

closed 2019-04-13 by sidoh