sidoh/esp8266_milight_hub #433: Thoughts: API Documentation using Postman Tool

shamasis, 2019-04-13:
Hi @sidoh,

Your work is amazing. I have recently set this up at home on an ESP8266 and am in the process of using `homebridge-http` plugin to access the lights via Siri. In the process of configuring the APIs, I had to test the same. Since I work at a company called Postman, which makes API management tools, I setup a small documentation of the APIs here: https://documenter.getpostman.com/view/33232/S1ENzKNF

This could be useful to others too.

shamasis, 2019-04-13:
Also, I could not find a plugin for HomeBridge (though I found one for HomeAssistant.) So, I published one: https://github.com/shamasis/homebridge-milight-hub

PS: This is just a very basic one... planning to add all features.

sidoh, 2019-04-14:
Love it!  Would be happy to add the Postman documentation to the README.  Is there a way for me to copy the workspace you've got, or for you to add me to yours?  Would like to be able to edit if I make changes to the API.

Plugin looks great.  Would be happy to add it to the [wiki](https://github.com/sidoh/esp8266_milight_hub/wiki).

shamasis, 2019-04-19:
Great. I have cleaned up milight-hub readme and released v1. That should make adding to wiki / readme feasible now. I have been running this myself for a week now and did not see any crashes.

Regarding the Postman Collection ... we can do either: you click on Run in Postman button on top and import it in your workspace and invite me (`mail+milight@sshamasis.net`) to it or I can continue maintaining it whenever you change APIs. I will also look into creating a Postman team and inviting you to it to save some of your time. The last option seems coolest.

PS: I think I have a feature request to add APIs to return list of registered bulbs. Maybe I will try and send a PR myself. Should I create a separate issue?

sidoh, 2019-04-19:
> Great. I have cleaned up milight-hub readme and released v1. That should make adding to wiki / readme feasible now. I have been running this myself for a week now and did not see any crashes. [...]

Appreciate it!

Makes sense.  Postman team sounds good to me.

> PS: I think I have a feature request to add APIs to return list of registered bulbs. Maybe I will try and send a PR myself. Should I create a separate issue?

Yeah, let's create a separate issue to discuss.  Sounds like this makes sense, but there are a few ways we could go with the semantics of the response.

sidoh, 2019-07-30:
Hi @shamasis, I've gone a potentially different route -- spec-ed the API using the openapi3 spec and set up some CI generation with ReDoc.  I liked having it in a platform-independent file that I can stuff into git.

Happy to incorporate Postman into this somehow if it makes sense.  Let me know :)

closed 2019-09-01 by sidoh