sidoh/esp8266_milight_hub #434: Nodemcuv3

AlexKusnezov, 2019-04-15:
hey guys, I have the nodemcuv3 and NRF24L01+  . I flashed it with https://github.com/sidoh/esp8266_milight_hub/releases/download/1.9.0-rc2/esp8266_milight_hub_nodemcuv2-1.9.0-rc2.bin on windows with estool. Everything went well and I can access the webinterface, but I just can't get a bulb to pair with it. The bulb I'm trying to pair is fut013. What am I doing wrong? Is there something I have to adjust in the precompiled bin or within the estool to get it flashed properly?

JMacIV, 2019-04-15:
One thing that tripped me up with these bulbs is that they will only pair if you physically power cycle the bulb on, then quickly hit the pair button on web interface. The bulbs stop looking for a pairing request after about 5 seconds of being on. It’s simple, but give that a try first. 

Also double check that you have a valid device_id (typically hex, like 0x1B0B) inputted and a group selected before pressing pair. 

If that does not work, then you could be looking at an issue with the NRF board - connection or hardware. Do you have existing bulbs? If so, you can turn on the sniffing feature and send commands via the Mi-Light app to the existing iBox controller. Your esp should be able to detect the traffic, which will tell you if the NRF board is working.

AlexKusnezov, 2019-04-15:
@JMacIV thanks I'll try that. Which parameters should I look for, when sniffing, in order to tell if NRF is working or not? Btw there's an LED on the NRF, which glows red constantly.

JMacIV, 2019-04-15:
When sniffing is enabled on the ESP, you should see commands sent to the bulbs show up there. Granted, this requires you to have existing bulbs configured via a Mi-Light controller like the iBox 1/2 and then send the commands that way. 

That is good news that the NRF is getting power. The one I used does not have an LED, but there is a lot of variation in these chips.

AlexKusnezov, 2019-04-15:
@JMacIV thank you for explaining. Currently I have only the fut013 bulb. I was hoping that I can just pair it with the esp in order to avoid buying original hubs and remotes at all.

JMacIV, 2019-04-15:
> @JMacIV thank you for explaining. Currently I have only the fut013 bulb. I was hoping that I can just pair it with the esp in order to avoid buying original hubs and remotes at all.

Ahh gotcha. Were you able to try and set a device_id, pick a group and then try to pair by powering off bulb, back on, and pressing pair within 5 seconds?

Once you have done that, we can try other stuff. Here is the full pairing guide. https://github.com/sidoh/esp8266_milight_hub/wiki/Pairing-new-bulbs

sidoh, 2019-04-15:
Thanks for your help, @JMacIV!

@AlexKusnezov, if the pairing procedure JMacIV linked isn't working, check out the troubleshooting article:

https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting

AlexKusnezov, 2019-04-16:
@sidoh @JMacIV I've figured it out. I bought the without the wireless module, this was the problem :https://www.amazon.de/NRF24L01-drahtloses-Breakout-Adapter-Intellective/dp/B06Y62C4H7/ref=sr_1_16?keywords=NRF24L01%2B&qid=1555412924&s=gateway&sr=8-16

Sorry for being dumb.

JMacIV, 2019-04-16:
No worries! Glad you got it figured out. 

That almost happened to me too. US Amazon has a very similar listing, I almost bought what you did too! Had the wrong one in my cart and everything!

@sidoh - Good to close.

sidoh, 2019-04-17:
👍 thanks for your help, @JMacIV!  Much appreciated.

Glad you got it working, @AlexKusnezov.

closed 2019-04-17 by sidoh

Kuaaaly, 2019-07-26:
Hi there,

Sorry to re-up this topic but I might be encountering a similar issue.
Is [this](https://www.amazon.fr/AZDelivery-NRF24L01-Antenne-emetteur-recepteur-Récepteur/dp/B06XJN417D/ref=sr_1_11?__mk_fr_FR=ÅMÅŽÕÑ&keywords=NRF24L01&qid=1564158962&s=gateway&sr=8-11) supposed to work ?

I've the exact same problem (can't pair bulb - FUT104) and start wondering if I bought the good stuff... It's my first time in anything : IoT, ESP8266 (Lolin NodeMCU v3), esptool.py, PlatformIO, etc...

Thank you in advance for your help!