sidoh/esp8266_milight_hub #436: Unpair once hub dies?

bworby, 2019-04-18:
So I have this made, and used it for a year or so, then the board died. How can I unpair my lights to pair to my new nodeMCU board?

sidoh, 2019-04-18:
IDs are spoofed.  You don't need to re-pair your bulbs.  Just re-use the same ID you had entered with your old one.

bworby, 2019-04-18:
I got that .. .but no idea what the ids were :(.. should have saved it somewhere. Any other thoughts?

sidoh, 2019-04-18:
I'd try guessing a few obvious things.  Maybe you used like... `1`? 😀 

If that doesn't work, I'm not aware of a way to unpair bulbs without knowing the device ID.

If I were in this situation, I'd probably binary search through the ~65k possible IDs.  Probably easiest with some bash/MQTT:

```bash
for i in $(seq 0 32767); do 
  mosquitto_pub -h mqtt-server -p 1883 -u username -P password -t "milight/${i}/rgb_cct/1" -m '{"status":"on"}'
  sleep 0.1
done
```

If bulb responds, it's in the first half, and search half of that space.  If not, search 32768-65535.  And so on.

kinda crappy, but can't really think of a better way. :-\

bworby, 2019-04-18:
oh goody :) ha. thanks for your attempts though, appreciate it. The latest version is newer than the one I installed a year ago. The web UI used to save the "hub" ids but no longer appears too?

bworby, 2019-04-18:
I believe I started with 0xFFF(X)? something and went backward. So I may try guessing a little first then do it programmatically if that fails.

sidoh, 2019-04-18:
In the 1.9.0 branch, IDs are automatically saved in the dropdown when you enter them.  1.9.0 is still pre-release, but RC3 is pretty stable.  I've used it in my own setup for a few weeks.  Just waiting a bit to make sure any kinks are worked out.

sidoh, 2019-04-18:
Godspeed!  Wish I had a better way to help you =)

bworby, 2019-04-18:
no prob, ill figure it out.  but what if a user were to break there non-home-grown hub? would they be hosed?

sidoh, 2019-04-18:
As far as I know, short of building something like this to search for the IDs, yeah. :X

bworby, 2019-04-18:
oh man ok. I thought they would have had some fail safe. Have a good one! thanks for all the quick great replies!

sidoh, 2019-04-18:
good luck!  gonna close, but feel free to continue the discussion if needed.

closed 2019-04-18 by sidoh

JMacIV, 2019-04-18:
Hey @sidoh - How long have the MQTT messages been set to retained in your code?

If awhile more than a year ago and @bworby used MQTT with the bulbs, then could just check the retained messages on MQTT server. Would have the device_id there

bworby, 2019-04-18:
thanks for the idea, I actually remembered I created my own interface a while back and put it on my server. It had the IDs saved! I changed my boards IP on it and I am getting successful responses back from this setup. but the lights are not responding? Ill re-check my setup.

bworby, 2019-04-18:
Got it, it just that Home Assistant is not working with it. I think it is because it is still using the old PORT that was on the older version of this. I'll mess with it. Thanks everyone!

JMacIV, 2019-04-18:
Awesome! Glad to hear it!

bworby, 2019-04-19:
Hey guys, sorry if I should put this somewhere else, but wanted to see what you guys though of another issue. I did find that a couple newer bulbs were not in my old setup so the IDs were not known. So I did write a script to start iterating potential ids and try them. It was working great for a while then I started noticing responses were hanging. I stopped the script and went to bed :) this morning I tried it again and it still is not responding. I noticed the red light was solid on the ESP. I power cycled it, and it popped back up with fancy blue light blinking as it was before but not responses are MUCH slower or timeouts. Any ideas on this? I reset my APs and network gateway just to be sure it was not a network packet issue.

bworby, 2019-04-19:
I am now seeing 
"Failed to load resource: The network connection was lost."
in console when using the UI on the board. So thats interesting.

sidoh, 2019-04-19:
What resource is it failing to load?  Can you copy/paste the whole error?

bworby, 2019-04-19:
That was just from the browser console:

[Error] Failed to load resource: The network connection was lost. (4, line 0) http://milight-hub.local/gateways/65391/rgbw/4

sidoh, 2019-04-19:
Not sure what that means.  Perhaps a crash?  I'd suggest looking at Serial logs.

bworby, 2019-04-20:
Well after hours of trying to figure it out, I remembered I had a bag of the NRF24s and swapped it out. And now its working again! ha. hmmmm thanks again for your time for this amateur nerd...

sidoh, 2019-04-20:
Glad you got it working!  For what it's worth, I've never had an ESP8266 or an nRF go bad in my several years of using them.  If yours actually are going bad, I might suggest making sure the power supply you're using is high quality.

jespertheend, 2019-12-19:
Uhm, I reckon this is probably way too late but in case anyone else has this issue. You can reset the milight bulbs by toggling the power on and off five times. You need to hold the power in the on/off state for about 2 seconds every time you toggle. The lights will start flashing red indicating that they have been unpaired.

You can also just pair the lights to a new id/group anyway. No need to unpair them first, just make sure you press the pair button within 2 seconds after turning the lights on. You can pair lights to multiple ids actually, not sure what the limit is. I have all my lights paired to one single id and I have paired them again to a separate id for each light individually. So I have one switch to control everything.

Not sure if this works for every milight model but I've found this to work for me at least.