sidoh/esp8266_milight_hub #437: Feature request: hass mqtt discovery suppport

danpowpow, 2019-04-19:
Hiya,
Your bridge is great. It's been running my lights for a year now with no issues - thanks!
You may or may not be aware of some big(ish) developments in hass. Things such as areas (think rooms). 
https://github.com/home-assistant/architecture/issues/94
These new features depend on 'integration' within hass, and mqtt devices integrate through a discovery mode
https://www.home-assistant.io/docs/mqtt/discovery/
do you think it would be possible to add this discovery feature to espmh?
Thanks again for great software!

sidoh, 2019-04-19:
Yeah, seems like a reasonable thing to look into.  Seems mostly straightforward, although would probably depend on something like #326.

Realistically speaking, will probably be a little while before I look into this.  Not planning on adding any more features to 1.9.0, so this would be for the next release.

Would be happy to look at a PR if it's something someone would like into before then, though. :)

oakbrad, 2019-06-26:
Thank you for this feature!

I have upgraded to 1.10.0 dev3. Gave all my lights aliases, changed my MQTT topics to use :device_alias, and then enabled HA discovery. Works perfectly!

danpowpow, 2019-06-27:
this is great news... so soon aswell :-) thanks for the implementation @sidoh
thanks aswell for your tips on setup @oakbrad

closed 2019-08-30 by sidoh