sidoh/esp8266_milight_hub #439: Ha Bridge Example

dougdalton, 2019-04-20:
Does someone have an example of how to do a ha bridge http json request that works with basic authorization required by the milight hub? I have used off as my example below

I have done an a http request with the target item : 
http://username:password@192.168.2.100/gateways/0x5685/rgbw/1

with http verb put and the http headers: 
[{"status": "off"}]
 
Content type : application/json

but I get a spark.matching error, what its the correct way to do this?

sidoh, 2019-04-20:
> http headers:
> [{"status": "off"}]

Not sure if this is what you meant, but these shouldn't be headers.  It should be the body of the request.

See examples here:

https://github.com/sidoh/esp8266_milight_hub#examples

dougdalton, 2019-04-22:
that was the issue,  thank you

closed 2019-04-22 by sidoh