sidoh/esp8266_milight_hub #442: Wemos D1 mini PRO / Won't save basic configs in UI

MrSiO, 2019-04-22:
Hi.
I have flashed a D1 mini PRO first with 1.8.7 then with 1.9.0-rc.4 but an issue persist.
Wifi config sticks, but all other configs like username, password, hostname, device id, led options etc... all disappear after browser refresh.  Able to log-in without password after power off/on.

All MQTT credentials also disappear from the web UI, but it keeps transmitting to broker...

More info:
1.9.0-rc.4  .bin image
used NodeMcu Firmware Programmer
Default configs except set to 16M of flash.

Thx all for your help.

sidoh, 2019-04-22:
Huh, didn't know there were 16M flash chips.

This is rough, sorry the upgrade bit you.  I upgraded the Arduino SDK version in [1.9.0-rc1](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.9.0-rc1).  That's my best guess for the source of the issue.

It's still connecting to the broker?  That's super weird.  What's the output of `/settings`?

Is there anything interesting in Serial logs?

MrSiO, 2019-04-22:
Hi! Thx for quick reply!
Problem is fixed.  

I believe using the BIN file for D1_mini was not a great idea in the first place...
What I did now is install Atom Linux (first time user) and download the 1.9.0-rc.4 source.

I modified the platformio.ini with the following infos:

```
[env:d1_mini_pro]
platform = ${common.platform}
framework = ${common.framework}
board = d1_mini_pro
upload_speed = 115200
build_flags = ${common.build_flags} -Wl,-Tesp8266.flash.4m1m.ld -D FIRMWARE_VARIANT=nodemcuv2
extra_scripts = ${common.extra_scripts}
lib_deps =
  ${common.lib_deps_builtin}
  ${common.lib_deps_external}
test_ignore = ${common.test_ignore}
```
Uploaded and now everything is ok!
My added devices are kept in config. So is my hostname and other configs like MQTT.
In the PIN fields, I always had to populate manually... now they are pre-populaed and worked great!

Only issue I faced is when I ask it to REBOOT.. it never came back online... 
Had to physically power off/on. No big deal

REQUEST:  Could you add a BIN file for the D1_mini_PRO ?  They are getting cheaper and WIFI range is way better.  Thx alot !!

sidoh, 2019-04-22:
Glad it's working!

Makes sense to add `d1_mini_pro`.

sidoh, 2019-04-22:
For what it's worth, seems like only 4MB of the flash is actually supported with the Arduino SDK: https://github.com/esp8266/Arduino/issues/2581.

MrSiO, 2019-05-13:
Great job releasing the .bin for the D1_mini_pro! Works great!!

closed 2019-05-28 by sidoh