sidoh/esp8266_milight_hub #444: Lag problems

poudenes, 2019-04-22:
Hi All,

I don't know what this can be. But when I use the Milight Hub a lot (with Domoticz) then there its creates a lag problem. It takes more and more time before the bulbs are going on or offf, or change color, etc.

When I reboot the hub then the lag is gone.

sidoh, 2019-04-22:
What integration are you using?  I've seen this happen when the network stack on the ESP8266 gets overwhelmed and TCP responds by throttling.

This thing is not designed for high throughput.  There are a few things you can do to improve performance, but don't expect to get more than a couple of updates a second out:

https://github.com/sidoh/esp8266_milight_hub/wiki/Performance-Tuning

sidoh, 2019-05-19:
I'm going to close this since I haven't heard back.

closed 2019-05-19 by sidoh