sidoh/esp8266_milight_hub #446: Compatibility with Mi-Light clone

cristianlivella, 2019-04-23:
Hi. I tried to buy a Mi-Light iBox2 to control my RGB stripes via WiFi, but I can't pair it with my RGB controllers. So I checked better, and I noticed that my controllers and remotes are apparently like Mi-Light FUT025/FUT098, but there aren't "Mi-Light" logos on them, so I think they are clones, and they don't work with iBox2. You think that I could use ESP8266+NRF24L01 to control my RGB stripes? Thanks

sidoh, 2019-04-23:
Would imagine it's possible in theory.  Probably not something this project in particular can help with, though. :)

sidoh, 2019-04-24:
Gonna close this because I think it's out of scope.  But happy to discuss further if you'd like.

closed 2019-04-24 by sidoh