sidoh/esp8266_milight_hub #448: Force wifi to BG

stefanheijnen, 2019-04-26:
I have been fighting with my esp8266 boards (mostly ESPEasy, a few Tasmota and 2x espmilighthub) for the past 2 months to keep a stable wifi connection. They often lost there connection with errors like:

(201) No AP found
(2) Auth expire
(4) Assoc expire
(200) Beacon timeout

Or they had problems getting an IP address, even though my pfSense firewall was giving them 1.

Rebooting has no effect, I just have to sit and wait till they reconnect to wifi, sometimes this takes hours to a day.

I have solved the DHCP part with assigning static IP addresses. The wifi errors have been resolved on my ESPEasy boards since they added an option to force wifi to the B/G band.

[image]

https://github.com/letscontrolit/ESPEasy/issues/2012

https://github.com/letscontrolit/ESPEasy/issues/1987#issuecomment-455843832

https://github.com/letscontrolit/ESPEasy/commit/6643eb4aaa653476948b25299a92d421e65aa528

My AP is an [Ubiquiti UniFi AP AC PRO](https://www.ui.com/unifi/unifi-ap-ac-pro/). I have tried a lot of different configurations, but with no effect, so I don't believe my problem is with the AP.

It would be nice to have the "Force wifi to BG" option in espmilighthub.

sidoh, 2019-04-28:
This is the first I'm hearing of this particular sort of behavior, which makes it feel like it's got to be some kind of environmental thing.  If it's not your AP, could it be a signal strength thing?

Probably doesn't hurt to have the option, happy to have it added.  Will probably not add it for 1.9 as this branch is already in RC, though.

stefanheijnen, 2019-04-28:
Good to hear! If I can test anything for you, let me know.

stefanheijnen, 2019-06-25:
Great! I will test it in the following days.

stefanheijnen, 2019-06-26:
I've updated 1 hub to 1.10.0-dev3. I enabled wifi B and rebooted. Devices doesn't come back up.

My AP doesn't see the device.

Serial log:

```
����ERROR: Failed to connect to MQTT server
*WM: Adding parameter
*WM: staticIP
*WM: Adding parameter
*WM: netmask
*WM: Adding parameter
*WM: gateway
We have a static IP: 10.0.2.95
                              *WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 10.0.2.95
*WM: Using last saved values, should be faster
```
After a few minutes WiFi manger starts it's own AP and I reconnect to my WiFi network. An option to disable WiFi B is missing. After a reboot it still doesn't show up and gives the same error. Although it says "Wifi connected succesfully"

```
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 10.0.2.95
*WM: Already connected. Bailing out.
Wifi connected succesfully

Setup complete (version 1.10.0-dev.3)
                                     ERROR: Failed to connect to MQTT server
ERROR: Failed to connect to MQTT server
ERROR: Failed to connect to MQTT server
ERROR: Failed to connect to MQTT server
ERROR: Failed to connect to MQTT server
ERROR: Failed to connect to MQTT server
ERROR: Failed to connect to MQTT server
ERROR: Failed to connect to MQTT server
$�%�=���ERROR: Failed to connect to MQTT server
*WM: Adding parameter
*WM: staticIP
*WM: Adding parameter
*WM: netmask
*WM: Adding parameter
*WM: gateway
We have a static IP: 10.0.2.95
                              *WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 10.0.2.95
*WM: Using last saved values, should be faster

```

sidoh, 2019-06-26:
The setting forces it into 802.11b mode.  does your router support this?

stefanheijnen, 2019-06-26:
Yes it does.

[image]

sidoh, 2019-06-26:
Looks like ESPEasy forces it into 802.11g rather than b.  And by default uses n.

Could you try _disabling_ the new setting?  This will force the ESP into g-mode.

stefanheijnen, 2019-06-26:
Can I change that setting through the serial console somehow, or should I reflash?

sidoh, 2019-06-26:
Probably easiest to reflash.

stefanheijnen, 2019-07-02:
> This will force the ESP into g-mode.

This works.

How about users who don't have issues with wifi N, are they forced to run their hub in G mode?

sidoh, 2019-07-02:
Yeah, this feature probably needs to be tweaked. I'll change it to either be G/N, or B/G/N.

stefanheijnen, 2019-07-03:
OK. For now I've updated all 3 of my hubs to the latest and have them running in G mode. I'll let you know in a few weeks what my experience is.

sidoh, 2019-07-03:
The dev.5 release has a setting to force b, g, or n mode.

closed 2019-08-30 by sidoh

denouche, 2020-12-20:
hello @stefanheijnen , did you solve your wifi connection issues with your Ubiquiti access point? I also have that kind of problems, what did you do to solve it? Thanks!

stefanheijnen, 2020-12-21:
I use the latest beta / rc firmware. Currently 4.3.23 in the ap's and the controller 6.0.x. But honestly, I won't be buying Unify again. They had a good name, but it's trash nowadays. Hardware is OK, but their software sucks, full of bugs that they don't acknowledge or fix. Just browse their forum.