sidoh/esp8266_milight_hub #449: Sending does not work

galinette2000, 2019-04-26:
Hi,

I have a problem sending commands on a new set-up.

I am using a NodeMCUv3 and a nRF24L01, connected with jump cables. I have triple checked that the wiring is correct (and used the same colors as you did in the tutorial for avoiding messing things up). The firmware was compiled and flashed using Atom + PlatformIO as described.

I can receive my FUT089 commands without any problem, and find my device ID, using packet sniffing.

But : I cannot send anything, neither using the remote ID, or by pairing the light bulbs.

I have followed the TroubleShooting guide. It fails when trying to sniff self-sent packets, while the guide said it should see them. But the guide doesn't say what this implies.

I have attached the settings and about pages outputs.

[settings.txt](https://github.com/sidoh/esp8266_milight_hub/files/3122374/settings.txt)
[about.txt](https://github.com/sidoh/esp8266_milight_hub/files/3122375/about.txt)

sidoh, 2019-04-27:
Sounds like a wiring issue:

https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting#wiring

> Bad wiring can cause crashes or hangs when sending commands.

Check Serial logs.  If there's a WDT rest, it's likely because one or more of the connections are faulty.  Try reseating them or switching cables if you're using jumper wires.

galinette2000, 2019-04-28:
I checked the serial monitor in Atom. Whenever I send a command, I get:

> Soft WDT reset
> ctx: cont
> sp: 3ffff9b0 end: 3fffffd0 offset: 01b0
> (Lots of hex dump)
> <<<stack<<<
> H!�ԅ���)(?*WM:
> *WM: AutoConnect
> *WM: Connecting as wifi client...
> *WM: Already connected. Bailing out.
> *WM: IP Address:
> *WM: 192.168.0.45
> Wifi connected succesfully
> 
> Setup complete (version 1.8.8)

But I removed all the jump cables, took new ones, re-cabled, and nothing changed. Is there any other possible cause? I will try changing the power supply, and also with another nodeMCU & radio.

galinette2000, 2019-04-28:
Hi,

I have tried connecting "CE" to "D4" on the NodeMCUv3, and in the setting set "2" for the GPIO (as D4 is GPIO2). This works now!

And, I have tried back connecting CE to D2 and set GPIO4, as it was by default, and it does not work.

So, either my D2 pin is dead (but this is a brand new NodeMCUv3 purchased for this project, and never connected wrong), either the D2 pin on the v3 is reserved for something else? I have seen that you used D4 on earlier versions, why did you change?

sidoh, 2019-04-28:
No idea re: D2.  I doubt the pin is reserved.  Would suggest testing with a simple blink sketch or something.

Context for the change here: https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.8.6

Gonna close, but happy to continue the discussion or re-open if there's anything else.

closed 2019-04-28 by sidoh