sidoh/esp8266_milight_hub #452: New full MQTT Domoticz Integration

galinette2000, 2019-05-02:
I just made a Domoticz plugin for the hub. It uses MQTT, but not through the built-in Domoticz MQTT. Instead, it can directly communicate with the ESP8266 hub using its native MQTT implementation.

I don't have all Milight lamps and remotes, but it supports most features, including color, tone, disco mode, etc etc... And it is fully bidirectionnal, which means if you use your legacy remotes, the state will be updated in Domoticz!

Devices are auto-detected (don't forget to allow new devices in Domoticz settings, and use your legacy remote, this will add devices)

Feel free to test. Where should we store this, do you want to add it to the project git repo, or should it be in a dedicated repo?

[esp8266_milight_hub.zip](https://github.com/sidoh/esp8266_milight_hub/files/3138707/esp8266_milight_hub.zip)

galinette2000, 2019-05-02:
I created a topic on the Domoticz forum:
http://www.domoticz.com/forum/viewtopic.php?f=65&t=27771

sidoh, 2019-05-02:
This is awesome!  Thanks for doing this.

> Where should we store this, do you want to add it to the project git repo, or should it be in a dedicated repo?

I think it should be in its own repo.  I'll be happy to link to it from this one.  Do you want to set one up?

I'd also like to add some instructions to the [wiki](https://github.com/sidoh/esp8266_milight_hub/wiki).  If there's something I can copy/paste there, that'd be awesome.

poudenes, 2019-05-09:
Plugin is great. MQTT BI directional is super. I have post at Domoticz Forum lot of information about the compatibility with domoitcz's dzvent system. Maybe this can be check as well so everything become extremely high class combination with the HUB, Plugin and Domoticz... :D:D:D

poudenes, 2019-05-17:
@sidoh  here how to install the plugin:

Goto your Domoticz Server/setup

```
cd ~/domoticz/plugin/
wget -O plugin.zip http://www.domoticz.com/forum/download/file.php?id=19778
unzip plugin.zip
systemctl restart domoticz.service
```

Goto your Domoticz Frontend page

setup > hardware > scroll down in Type and select: ESP8266 Milight Hub
Give this hardware a name
Fill in all the info that been ask below
MQTT part:
mqtt_topic_pattern: milight/:device_id/:device_type/:group_id
mqtt_state_topic_pattern: milight/states/:hex_device_id/:device_type/:group_id

Hit Add

Now goto your Milight Hub then settings > MQTT

MQTT topic pattern: milight/:device_id/:device_type/:group_id
MQTT update topic pattern: milight/updates/:hex_device_id/:device_type/:group_id
MQTT state topic pattern: milight/states/:hex_device_id/:device_type/:group_id

hit Submit

Now go back to Domoticz and then Setup > Settings 
Below hit Allow for 5 minutes 
Then hit Apply Settings upper right corner

Now go back to Milight Hub and switch on/off every bulb you have configured.
You will see that every bulb will add as device inside Domoticz

Remember the Group All is not working yet. They still develop this part

Here you find some info when you use dzVents System inside domoticz.
Some set options are working some not:
http://www.domoticz.com/forum/viewtopic.php?f=65&t=27771&start=20#p213498

sidoh, 2019-05-18:
Posted: https://github.com/sidoh/esp8266_milight_hub/wiki/Domoticz

Thanks @poudness!  I made a few minor edits (mostly formatting), let me know if I messed anything up.

Thanks again @galinette2000!

closed 2019-05-18 by sidoh

poudenes, 2019-05-18:
Looks nice !! and yes thanks again @galinette2000