sidoh/esp8266_milight_hub #454: SetSleepMode for WiFi - I loose connection

agoralive, 2019-05-05:
Hi

I have a ESP8266 (https://www.aliexpress.com/item/1pcs-NodeMCU-V3-Lua-WIFI-module-integration-of-ESP8266-extra-memory-32M-Flash-USB-serial-CH340G/32813084968.html?spm=a2g0s.9042311.0.0.27424c4dZslw1E)

When flashed with esp8266_milight_hub  I am able to start the AP-mode (select wifi network), without any problems = NO PROBLEMS

Then when I choose my wireless network, enter my code click save the module restart just like it is suppose to do.  I can then find the IP address of the ESP connect to it and everything works perfect for 5 seconds.

This is when the WIFI connection get's lost!!!! The esp is running fine, the milight_hub is working (I see that the LED accepts/reacts when any milight remote is pushed). This means that maybe there is a problem with me esp and the wifi onboard chip???? BUT NO, during AP mode there is no time-out problems, and I have flashed this board with a tasmota .bin file and this is running for hours without issues.

For the sake of testing I have tried with another board (same link as above), and this produces the same error.

What is wrong, Can I somehow set the wifi.mode or setSleepMode to Off or something like that?

I have also tested the same firmware (milight_hub) with an older board with pin pre-mounted and here it works without problems.

My guess would be that I need to disable power saving on the wifi-board in order to be successful.

Do anybody have any ideas?

sidoh, 2019-05-05:
Is there anything interesting in serial logs?

agoralive, 2019-05-05:
I don't know, how do I extract these?

sidoh, 2019-05-05:
https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html

or if you have platformio installed...

```
$ pio device monitor
```

agoralive, 2019-05-05:
Hi

No nothing of interest:

*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.0.227
Wifi connected succesfully

Setup complete (version 1.9.0-dev10)

sidoh, 2019-05-05:
There is no power saving / sleep mode with this project.  It's designed to be always-on.  What makes you think it loses its connection?

agoralive, 2019-05-05:
The fact that it get's dropped from the router. I've tried a google mesh, TP-link router and a "home central" which I think is from huawei.

The fact is just that when I use this specific esp it looses connection to ALL my routers, when I use any other esp or other esp builds (not milight_hub) it stays on, also when the AP mode is active the connection stays on as long as I look at it, with it's own AP

sidoh, 2019-05-05:
Not sure what to tell you.  Stays connected to my router, and this is the first I'm hearing about this problem. :\

I guess possible it's related to #448?  Do you have force B/G mode set on Tasmota?

agoralive, 2019-05-05:
I would not know where to enable this, and the fact that Tasmota work on this new esp8266 board makes me believe that it is not the board which has an error, but more likely the code which in incompatible with the board....

Your coding works great with the elderly boards I own.... Could I send you a board to test this with?

agoralive, 2019-05-06:
@sidoh - This appears to be the same problem I am experiencing:
https://github.com/sidoh/esp8266_milight_hub/issues/406

agoralive, 2019-05-06:
@sidoh 
Could you have a look at this: https://github.com/esp8266/Arduino/issues/5784

Appears that maybe the Arduino Core 2.5 is the issue and that a rollback is required to fix the issue...

sidoh, 2019-05-06:
#406 was opened before [1.9.0-rc1](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.9.0-rc1), which is when the SDK was upgraded to 2.5.0.  So doubt it's exactly the same issue.

This looks like a promising lead.  Could you test it?  You'd just want to revert this commit and compile/upload:

https://github.com/sidoh/esp8266_milight_hub/commit/b27a22497d0ce042703ddf8173813ca0f427994d

agoralive, 2019-05-06:
@sidoh I'm trying but really having problems with the compiling...... Are you saying that any .bin file which is before v1.9 is using the old SDK?

If I try 1.8.8 this would be SDK 2.3?

sidoh, 2019-05-06:
yes, versions prior to 1.9 use the platformio build v1.8.0, which I believe is Arduino SDK v2.3.

agoralive, 2019-05-06:
Hi

It makes no difference which version I flash the boards with, the all react the same way with these bin files.

I guess there is some compatibility issues with this code and these board. I have tried a Wemos D1 Mini today and this also works great. I still wonder how Tasmota is able to make everything work????

sidoh, 2019-05-06:
https://github.com/sidoh/esp8266_milight_hub/issues/454#issuecomment-489449610

Can you see if you have B/G mode enabled?  Maybe it's on by default.

I have no way of reproducing this, so I'm not going to be very helpful.  This is something you're in a much better position to debug yourself.  Very happy to look into any solutions you find, though.

sidoh, 2019-05-06:
Another suggestion: try erasing flash and re-flashing the binary.  Would be surprised if this solves the problem, but always something you should try when you see screwy behavior.

agoralive, 2019-05-06:
I don't know about B/G Mode, but:

Core/SDK Version | 2_3_0/1.5.3(aec24ac9)
-- | --

agoralive, 2019-05-06:
I have tried to erase flash

sidoh, 2019-05-06:
The 1.8 branch is using v2.4.2.  You'll need to compile it yourself if you want to downgrade beyond that.

I'm running out of suggestions.  Think this is something you'll have to drive.  Happy to answer particular questions, though!

agoralive, 2019-05-06:
I just find it strange that the network connection ONLY falls out when it's connected to a network, but works PERFECT when running in AP mode?

agoralive, 2019-05-06:
Problem is the same on two different Networks (Google Wifi and a D-Link Covr) - Both mesh

sidoh, 2019-05-06:
I'd add, though, that it drops when it connects to _your_ network.  Works fine on mine!

Let me know if you end up figuring out what's wrong, and will be happy to incorporate a solution.  Also happy to help with particular questions.  But I'm not going to be able to figure this out for you.

sidoh, 2019-05-07:
btw - the platformio version that maps to SDK v2.3 is 1.5.0 ([platform releases](https://github.com/platformio/platform-espressif8266/releases?after=v1.7.0)).

That's the only meaningful difference I can see between this and Tasmota.  The release log there does seem to indicate that newer versions cause issues.

I've built against v2.4 on this project for a long time.  Really old versions might be built against v2.3.

If using v2.3 works for you, would be happy to add a release binary built against v2.3.

agoralive, 2019-05-12:
So.....

I tested the 1.5.0 Binaries for NodeMCUv2 - This keeps connection!!!!! Works great. Are there any WiFi changes since this release?

Could you try to compile usign 2.3 as you described above?

sidoh, 2019-05-12:
Nice, glad you're making progress on debugging.

> Are there any WiFi changes since this release?

Pretty sure not, but check the change logs. :)

> Could you try to compile usign 2.3 as you described above?

I would like you to try this. :)

I left some instructions/hints in these comments:

https://github.com/sidoh/esp8266_milight_hub/issues/454#issuecomment-489656893
https://github.com/sidoh/esp8266_milight_hub/issues/454#issuecomment-489900745

agoralive, 2019-05-12:
Hi

I appreciate your comments, but my knowledge at this moment is not sufficient enough to compile your code. I get many streange errors which I am still not able to correct, and also "changing" these Cores/SDK's I have no idea of how to do so.

sidoh, 2019-05-12:
Do you have PlatformIO installed? 

https://platformio.org/install/cli

It really is as simple as changing the version (one line) in `platformio.ini`.  Read the comments carefully, I think you'll be able to figure it out. :)

I can do it for you, but I'm preoccupied right now, and I'll probably not get to it for quite a while.

agoralive, 2019-05-12:
I am getting a lot of errors when compiling, is there a tutorial available?

sidoh, 2019-05-12:
Yes, I'm using platformio.  Tons of documentation.

https://docs.platformio.org/en/latest/core.html

agoralive, 2019-05-12:
Yeah I get errors all over the place, could I pay somebody to try and make this for me? Anybody?

agoralive, 2019-05-12:
Could you consider this compiled option for the next release?

sidoh, 2019-05-12:
Alright, just tried.

Looks like there are some pretty deep incompatibilities with 2.3.  All sorts of complaints about progmem macros not working in the same way.  There are upstream libraries I'm using that rely on >2.3 SDK as well.

I'd rather not add a bunch of crappy glue code to make it compatible with 2.3.  I would imagine 2.5 will fix connectivity issues eventually.

sidoh, 2019-05-12:
Here's one compiled against 2.4, but sounded like you already tried that, and it didn't work?

I don't think I'm going to support 2.3, sorry.

[milight-hub-sdk2.4.bin.zip](https://github.com/sidoh/esp8266_milight_hub/files/3170534/milight-hub-sdk2.4.bin.zip)

agoralive, 2019-05-13:
I will give this a try later today. Thank you for the effort.

agoralive, 2019-05-13:
I tried, but as you already mentioned I already tried the 2.4 and it does not work :(

sidoh, 2019-05-14:
Gotcha, makes sense.  Unfortunately I think we're at an impasse here.  I don't want to back-port to a really old version of the SDK unless there's a sizable fraction of the user base that's affected by this issue.

Does 1.5 have the features you need?  Lots of improvements since then, but maybe barebones will hold you over until the SDK folks patch this issue.

agoralive, 2019-05-17:
We did not find a solution, the code works but not with this board :(

closed 2019-05-17 by agoralive