sidoh/esp8266_milight_hub #456: WebUI: .css files loaded from the internet

hutmitfrosch, 2019-05-10:
Hi :-)
Thanks for that realy usefull tool, I am just discovering everything.

The first time when I started the UI, I had no internet connection - and everything looked broken, because the CSS is loaded from external servers.
Is this realy necessary, to make a lokal system be dependend on the availability of external servers and a working internet connection?

Another thing I discovered yet: Can it made be possible to save the ID, you want to use in the UI to send commands? The system seems to forget all IDs given to it - a list to select from would be great! This way, the UI could be part of the home automatisation controlls visible to the user and replace the remote controlls, if wanted :-)

Thanks in advance!

sidoh, 2019-05-10:
This is probably mostly a duplicate of #413.  You can read that issue for more context.

The short of it is that packing all of the upstream dependencies into the web bundle would eat up a bunch of space on flash.  There are 4MB to play with, and I'm not particularly keen on cramming 25% of that with Javascript.

The UI is the only thing that has an external dependency.  The REST API, MQTT, UDP, etc. are all entirely local.  The device can be configured via the REST API as well, so no need to even use the API if you don't want to.

IDs are auto-saved as of 1.9.0 (still in RC versions, but it's pretty stable).

hutmitfrosch, 2019-05-10:
Hi sidoh,
cool quick response :-)
Yea, I understand that storage problem. Did not think about it.

Will try to update to 1.9, thx!

sidoh, 2019-05-10:
Sweet.  Will close.  LMK if I've missed something.

closed 2019-05-10 by sidoh