sidoh/esp8266_milight_hub #457: Anyone using this with HomeSeer? MQTT is breaking my brain

OliPassey, 2019-05-12:
Got my hub working but i just can't work out how to configure MQTT, new to it and finding the wiki a bit light on detail - any advice?

I've got a broker working, and HomeSeer has a client talking to the Broker - i just can't work out how to configure the milight hub side of things or get it to respond to msgs.

sidoh, 2019-05-12:
I assume you've read

https://github.com/sidoh/esp8266_milight_hub#mqtt

?

It's reasonably detailed.  If you figure out HomeSeer-specific instructions, happy to add to the wiki.

OliPassey, 2019-05-12:
Yeah, i think im getting it now - slowly but surely.
What i've done, in case it helps someone else...

Setup the MiLight as per the wiki link above, copying exactly the phrases mentioned. 
Then in HomeSeer events, with the mcsMQTT plugin installed, use the "Send MQTT msg" event action to send a msg in the format:
milight/0x001/rgb_cct/0={"status":"ON"}

I guess i need to write loads of the above msgs's for all the scenes / settings i want to use and build those into events, to then be triggered by virtual devices.

closed 2019-05-12 by OliPassey