sidoh/esp8266_milight_hub #460: REST API and MQTT report CCT bulbs state OFF when in night_mode

Chrominator, 2019-05-19:
First of all, many thanks for this great dev.
Really amazing !

My CCT buld is currently in night_mode.
http://x.x.x.x/gateways/0x8B6C/cct/1 produces the following :
`{
  "state": "OFF",
  "brightness": 128,
  "level": 50,
  "kelvin": 60,
  "color_temp": 283,
  "bulb_mode": "white",
  "color": {
    "r": 255,
    "g": 255,
    "b": 255
  },
  "effect": "white_mode"
}`
Previous state was OFF.
Did not find a way to know that the bulb is in night mode.
I expected to have it at least in state ON, or better with an effect : night_mode.

State is reported ON in such situation with RGBW bulbs.

sidoh, 2019-05-19:
Looks like the night mode command is not properly decoded from the raw packet, I'll take a look.

sidoh, 2019-05-19:
Will be fixed in [1.9.0-rc.7](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.9.0-rc.7).

(Build is in progress, artifacts will be published when it completes.)

Thanks for reporting!

Chrominator, 2019-05-19:
Great !

Many thanks, I do appreciate your reactivity.
I've changed my MQTT flow and everything went well now.

Thanks again.

poudenes, 2019-05-26:
When change bulb from Webgui to night_mode you cant change the bulb to color, brightness, etc. You must turn it off and on again to use the other options.

sidoh, 2019-05-26:
Yeah, I think I'd say that's expected.  This is how the bulbs actually work: they have to be turned either on or off before you can change any other properties.

closed 2019-05-28 by sidoh