sidoh/esp8266_milight_hub #464: Wrong response to MQTT command

galinette2000, 2019-05-19:
When I send to MQTT:
`{"level": 74, "color": {"b": 255, "g": 233, "r": 139}}`
Then:
`{"command": "set_white", "temperature": 0, "level": 14}`

On the last command, it switches to white, but it seems the brightness level stays at 74 (high). However the interface reports the level to 14 properly, even if the lamp has a level of 74

The receiver is a FUT044 led strip controller (If I remember right, since it is embedded in the ceiling...)

sidoh, 2019-05-19:
I'm not able to reproduce this with the FUT089 remote type and a FUT105 bulb.  Reported state behaves as expected, and so does the bulb.

Perhaps the LED controller missed a packet?  State is produced from interpreting sent packets directly, so it's very unlikely that the state is correct if the right packets weren't sent out.

You could try:

1. Bumping up the number of repeats.
2. Sending on only a subset of radio channels and increasing the repeats more.  For example, I only send on MEDIUM and HIGH.  This cuts down the time it takes to send a single packet by ~1/3, which allows for 33% more repeats with no performance cost.
3. Ensuring the nRF24 has some space between the ESP8266 radio.  Interference is a real problem.

poudenes, 2019-05-20:
> Ensuring the nRF24 has some space between the ESP8266 radio. Interference is a real problem.

I have connect a antenna. The cable between is 1m long and the end of it is 30 min away from the ESP. Will this help also? Or MUST the nRF24 module as far as possible away from the ESP?

sidoh, 2019-05-20:
That should be fine.

Gonna close this 'cause I think it's expected behavior.  If it doesn't look that way to you, please feel free to reopen.

closed 2019-05-20 by sidoh

galinette2000, 2019-05-20:
No, the issue is 100% repeatable, and I'm 100% sure there is an issue. The lamp responds perfectly to any other color commands, so there is no radio problem. Only this specific MQTT command causes the brightness to stay at its previous level while the hub state takes the new level.

Interestingly, if I do:
`{"command": "set_white", "level": 14}`
it works

while
`{"command": "set_white", "temperature": 0, "level": 14}`
Always fails. The state stays to its previous level but the hub goes to 14.

Maybe it's this specific receiver type (FUT044) which does not like the packet.

sidoh, 2019-05-20:
Not sure what to tell you, I'm not able to reproduce this with my FUT105 bulbs and FUT089 remote type. 

State behaves exactly as expected with both commands you provided:

```
$ curl -X PUT -H 'Content-Type: application/json' -d '{"level": 74, "color": {"b": 255, "g": 233, "r": 139}}' 10.133.8.158/gateways/11/fut089/1
{"brightness":189,"bulb_mode":"color","color":{"r":140,"g":233,"b":255}}%

$ curl -X PUT -H 'Content-Type: application/json' -d '{"command": "set_white", "temperature": 0, "level": 14}' 10.133.8.158/gateways/11/fut089/1
{"brightness":36,"color_temp":153,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}%

$ curl -X PUT -H 'Content-Type: application/json' -d '{"level": 74, "color": {"b": 255, "g": 233, "r": 139}}' 10.133.8.158/gateways/11/fut089/1
{"brightness":189,"bulb_mode":"color","color":{"r":140,"g":233,"b":255}}%

$ curl -X PUT -H 'Content-Type: application/json' -d '{"command": "set_white", "level": 14}' 10.133.8.158/gateways/11/fut089/1
{"brightness":36,"color_temp":153,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}%
```

And the bulb does what I'd expect it to as well.

I'm happy to look into this, but I have to be able to reproduce it in order for that to happen.  I don't have access to the hardware you're using, unfortunately.  Any chance you can dig a little deeper on your own?

Like I said, state is interpreted directly from packets that are sent out.  So it's not really possible for state to get updated without the appropriate packet getting sent out.  Either your LED controller is missing the packet, or it's ignoring it.

_(note that brightness is on a [0,255] scale, while level is on a [0,100] scale.)_

galinette2000, 2019-05-20:
The FUT044 is a RGBW controller, which is either in RGB mode or in White mode (without temp control). Do you send temperature and brightness in the same packet? It seems that it ignores both because the temp is there. But it reacts to the white mode. And if I send white+brightness it works. It's only when sending white+temp+brightness that brightness is ignored.

sidoh, 2019-05-20:
The underlying RF packets can only contain one command, so no.  Multiple packets are sent at the same time in fast succession.

galinette2000, 2019-05-20:
So it seems that when sending a temp command to that controller, which does not support it, it ignores what comes next.

galinette2000, 2019-05-20:
Even stranger (and 100% repeatable):
`{"command": "set_white", "level": 14}`  **OK!**
`{"command": "set_white"}` then `{"command": "level": 14}` **Fails**

sidoh, 2019-05-20:
this is not valid JSON.  I would not expect it to work:

```json
{"command": "level": 14}
```

sidoh, 2019-05-21:
Example separately:

```
$ curl -X PUT -H 'Content-Type: application/json' -d '{"level": 74, "color": {"b": 255, "g": 233, "r": 139}}' 10.133.8.158/gateways/11/fut089/1
{"brightness":189,"bulb_mode":"color","color":{"r":140,"g":233,"b":255}}%

$ curl -X PUT -H 'Content-Type: application/json' -d '{"command": "set_white"}' 10.133.8.158/gateways/11/fut089/1
{"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}%

$ curl -X PUT -H 'Content-Type: application/json' -d '{"level": 14}' 10.133.8.158/gateways/11/fut089/1
{"brightness":36,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}%
```

together:

```
$ curl -X PUT -H 'Content-Type: application/json' -d '{"level": 74, "color": {"b": 255, "g": 233, "r": 139}}' 10.133.8.158/gateways/11/fut089/1
{"brightness":189,"bulb_mode":"color","color":{"r":140,"g":233,"b":255}}%

$ curl -X PUT -H 'Content-Type: application/json' -d '{"command":"set_white","level": 14}' 10.133.8.158/gateways/11/fut089/1
{"brightness":36,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}%
```

seems to work as expected here, right?