sidoh/esp8266_milight_hub #465: Wifi issues

PetricaM, 2019-05-20:
Hi Chris,

Any chance that Arduino SDK update introduced in 1.9.0 RC1 affected wifi watchdog?

I have two boards with ESPMH and sometimes one or both go offline (unsure, but I think this behavior started around the 1.9.0 mark) by no identifiable pattern. I also have other boards with ESP8266 chip and haven't seen major issues with them.

Sometimes boards still appear as connected (although not accessible by any computer) in the router webpage and I need to press the physical reset button or to remove power.

I did activate pushing updates over MQTT topic on one of ESPMH, however the other has only the command topic. 

Thanks

sidoh, 2019-05-20:
Hi Petrica,

Which version are you using?  I reverted back to v2.4 of the SDK in 1.9.0-rc.6 after reading a long comment thread about v2.5 having some pretty mysterious issues. *

I've been on v2.4 quite a long time.  The 1.8 branch was v2.4, for example:

```json
{"version":"1.8.8","variant":"d1_mini","free_heap":29360,"arduino_version":"2_4_2","reset_reason":"Software/System restart"}
```

from 1.9.0-rc.8:

```json
{"firmware":"milight-hub","version":"1.9.0-rc.8","ip_address":"10.133.8.194","reset_reason":"Software/System restart","variant":"d1_mini","free_heap":25160,"arduino_version":"2_4_2"}
```

Is there anything interesting in Serial logs?

[*] Context on reverting:
Sounded like the maintainers jumped the gun on incorporating a dev version of the upstream SDK.  I believe this was reverted in the most recent patch of 2.5, but didn't want to mess with it.

PetricaM, 2019-05-20:
I had 1.9.0 RC5 until this morning (now on RC8). 

Tried several times the serial connection; however, there was no wifi issue when connected to pc :)

Will get back case there are further problems on RC8.

Thanks

sidoh, 2019-05-20:
Ah, hopefully that's mystery solved then!

Gonna close because I'm discovering that I'm horrible with ticket hygiene.  But feel free to reopen if you're still having trouble.

closed 2019-05-20 by sidoh

PetricaM, 2019-05-25:
Just dropped to say that no wifi issues on RC8