sidoh/esp8266_milight_hub #466: More flash files for different modules

poudenes, 2019-05-21:
Hi,

I have some modules here (bought them on Wish) but they slight different then you now have add for flashing:

- ESPDUINO-32 ESP32 WEMOS R1 D32 
- new NodeMcu V3

Is it possible to create flash files for those as well?

sidoh, 2019-05-21:
Heya @poudenes,

For ESP32, there will need to be some meaningful code changes to support it.  See #405.   The SDKs have slightly different API syntaxes, some libraries are missing, some things work slightly differently, etc.  It's doable, but not as simple as adding another platform to the build.

My guess is that nodemcuv2 will work just fine for v3.  The only difference between the published platforms are flash sizes.  Some of them are even redundant (d1_mini and nodemcuv2, for example).

sidoh, 2019-05-21:
Give v2 a try and let me know if it doesn't work.  We can re-open the ticket if not.

closed 2019-05-21 by sidoh

JMacIV, 2019-05-21:
Side note on v2 and v3 Node chips - They use different USB drivers since have different USB chips onboard. @poudenes - If you have computer connection issues when using the new chips, that is probably why.