sidoh/esp8266_milight_hub #467: ESP8266 with NRF24L01+ often needs to issue command twice to get entire group to update

dougdalton, 2019-05-21:
I have noticed that I need to issue a command sometime 2 or 3 times to get all the lights in a group to change on a command,  its never the same light twice.  has anyone else had this issue?

poudenes, 2019-05-21:
I had the same problem. I changed some values and now all problems gone. Also create as much distance between ESP and nRF module

[image]

sidoh, 2019-05-21:
thanks @poudenes :)

Sounds like your bulbs are missing packets.  Suggestions (mostly repeating what poudenes said):

1. Make sure there's some distance between your ESP8266 and nRF24.  I noticed a dramatic improvement in performance when I separated the two with a ~20cm ribbon cable.
2. Increase the number of packet repeats (there's a setting for this).

poudenes, 2019-05-23:
@sidoh Can you tell me how many bulbs you can control in 1 action. I have 28 bulbs and in Domoticz I use dzVent Script to turn on and off the bulbs when a dummy switch is turn on or off.

I see that after 24 bulbs the hub stop handeling the actions and send back the MQTT state to Domoticz.

sidoh, 2019-05-24:
I'm not sure precisely what the limits are.  It's not really acceptable that the hub dies when it gets overwhelmed... just not how it should work.  But unfortunately it is.

Think the best thing I can do to address this is switch everything to be asynchronous.  Perhaps I'll look to do that in the next release.

poudenes, 2019-05-24:
It's not a big issue. (For me) I was test this and see what happen. When I want turn all the bulbs I turn on the groups per room :)

sidoh, 2019-05-24:
Will likely be the next thing I work on, but I work on my OSS stuff in spurts, and I have a few different projects that I work on.  Could be next month, could be a year from now. :X

poudenes, 2019-05-24:
You dare to work on it after a year 🤪

sidoh, 2019-05-27:
@dougdalton, did any of these suggestions help with your problem?

poudenes, 2019-06-07:
[image]

Will this module be better to use?

sidoh, 2019-06-07:
Maybe, although you'd want to make extra sure your power supply can handle it.  No personal experience with it.

Gonna close this ticket since I've not heard back, but happy to reopen.

closed 2019-06-07 by sidoh