sidoh/esp8266_milight_hub #468: 1.9.0 Errors in Domoticz log

poudenes, 2019-05-23:
Hi,

Don't know if I can post it here. I was do some test with a second Milight hub build and this latest beta firmware. The current stable is working correct but with this one I get errors:

```
2019-05-23 18:24:41.472  (Milight Hub) 001/Dressoir 1 - 0xAAAF/rgb_cct/1: State change: '{'state': 'OFF', 'status': 'OFF', 'brightness': 255, 'level': 100, 'hue': 232, 'saturation': 100, 'color': None, 'bulb_mode': 'color'}
2019-05-23 18:24:41.472  Error: (Milight Hub) 'onMessage' failed 'TypeError':'argument of type 'NoneType' is not iterable'.
2019-05-23 18:24:41.472  Error: (Milight Hub) ----> Line 675 in '/home/pi/domoticz/plugins/ESPMilight/plugin.py'
2019-05-23 18:24:41.472  Error: (Milight Hub) ----> Line 226 in '/home/pi/domoticz/plugins/ESPMilight/plugin.py'
2019-05-23 18:24:41.472  Error: (Milight Hub) ----> Line 150 in '/home/pi/domoticz/plugins/ESPMilight/plugin.py'
2019-05-23 18:24:41.472  Error: (Milight Hub) ----> Line 276 in '/home/pi/domoticz/plugins/ESPMilight/plugin.py'
2019-05-23 18:24:41.472  Error: (Milight Hub) ----> Line 641 in '/home/pi/domoticz/plugins/ESPMilight/plugin.py'
```

The bulbs go on and get their collor and brightness for this test

sidoh, 2019-05-24:
If the issue is with the plugin, let's put the issue on its board:

https://github.com/galinette2000/espmilighthub-domoticz

thank you!

closed 2019-05-24 by sidoh

poudenes, 2019-05-24:
But the plugin is working with current stable firmware

sidoh, 2019-05-24:
I'm guessing this is the issue:

```json
{'color': None}
```

Can you paste the output of `/settings`?  Make sure to sensor any passwords you've got.  Would also be useful to see the log from a version that works.

Tagging @galinette2000 as well in case you've got some leads.

poudenes, 2019-05-24:
**Settings from 1.9.0**

`{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":25,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.100.18:1883","mqtt_username":"XXXXXXXXXX","mqtt_password":"XXXXXXXXXX","mqtt_topic_pattern":"betamilight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"betamilight/updates/:hex_device_id/:device_type/:group_id","mqtt_state_topic_pattern":"betamilight/states/:hex_device_id/:device_type/:group_id","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":0,"state_flush_interval":0,"mqtt_state_rate_limit":50,"packet_repeat_throttle_sensitivity":10,"packet_repeat_throttle_threshold":1000,"packet_repeat_minimum":10,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow toggle","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","rf24_channels":["LOW","MID","HIGH"],"device_ids":[43681,43682,43683,43684,43690,43691,43692,43693,43694,43695,43706,43707],"gateway_configs":[],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type","oh_color"]}`

**Settings of 1.8.8**

`{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":25,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.100.18:1883","mqtt_username":"xxxxxxxxx","mqtt_password":"xxxxxxxx","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:hex_device_id/:device_type/:group_id ","mqtt_state_topic_pattern":"milight/states/:hex_device_id/:device_type/:group_id","discovery_port":48899,"listen_repeats":0,"state_flush_interval":0,"mqtt_state_rate_limit":50,"packet_repeat_throttle_sensitivity":10,"packet_repeat_throttle_threshold":1000,"packet_repeat_minimum":10,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow toggle","led_mode_packet":"Flicker","led_mode_packet_count":3,"device_ids":[43681,43682,43683,43684,43690,43691,43692,43693,43694,43695,43706,43707],"gateway_configs":[[43690,5980,6],[43691,5981,6],[43692,5982,6],[43693,5983,6],[43694,5984,6],[43695,5985,6],[43681,5986,6],[43682,5987,6],[43683,5988,6],[43706,5989,6],[43684,5990,6],[43707,5991,6]],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type"]}`

sidoh, 2019-05-24:
Try turning off the `oh_color` field:

```json
{
  "group_state_fields": [
    "state",
    "status",
    "brightness",
    "level",
    "hue",
    "saturation",
    "color",
    "mode",
    "kelvin",
    "color_temp",
    "bulb_mode",
    "computed_color",
    "effect",
    "device_id",
    "group_id",
    "device_type",
    "oh_color" <--- Delete this
  ]
}
```

You should probably trim that down quite a bit, actually.  Probably don't need all of color, computed_color, saturation, hue, etc.  computed_color by itself should be sufficient.

poudenes, 2019-05-24:
Ill go try that later... When I'm at home again!

poudenes, 2019-05-25:
Removed the line, now I don't see any errors appear in the log.
I did also the test to turn all the bulbs. But 1.9.0 sends max 11 commands, then it stop.
Current version sends 24. (Maybe this because Hub runs on a Wemos D1 and beta on a NodeMCU

poudenes, 2019-05-25:
Ohh and the groups all are working also! On / off.
Didn't try all the dzvents options with set.... Will do that later

sidoh, 2019-05-25:
👍 nice