sidoh/esp8266_milight_hub #471: Group 0 update Group 1 2 3 4

poudenes, 2019-05-29:
I saw a issue about this. But I think the way how Group 0 works is different.

When I turn on Group 0, I can change everything for all the bulbs that are in Group 1 2 3 4.
So when I turn Group 0 on, change settings I think that also those changes must pushed to Group 1 2 3 4.

When you turn off on of the Groups 1 2 3 4 than also Group 0 goes off. Because not all the groups are on anymore. Also when you turn on individual Group 1 2 3 4 and all are on then Group 0 must be on as well, because all groups are on.

When Group 1 2 3 4 are on of one of those and you turn on Group 0 and off again then all the groups 1 2 3 4 must be off as well, because you turn off Group 0.

Im right or not?

Because this is not active you get the problem with bi-direction sync in Domoticz:
I turn Group 1 2 3 on give them all different colours, then when I turn off Group 0 only Group 0 goes off (even when its already off) but the Groups 1 2 3 devices still have status ON.

How I solve this for now is add Group 0 into Group 1 2 3 4 as a slave device. When Group 1 2 3 4 goes on (or one of them) then Group 0 goes on as well without actions to Hub.
When I turn off Group 0, group 1 2 3 4 goes off in Hub but also in Domoticz without any action to Hub. 

Maybe a idea to look deeper into this?

sidoh, 2019-05-29:
Current group 0 behavior is probably not going to get better than it is right now.  When, for example, group 1 changes its state, it invalidates and clears the state of group 0.

sidoh, 2019-05-29:
summary of behavior is here:

https://github.com/sidoh/esp8266_milight_hub/pull/426

I understand that there are ways in which it might behave better, but it's already complicated enough as is, and I don't want to make it cleverer.

poudenes, 2019-05-29:
Understand. I'm not that smart to code things... So I will use the slave option in domoticz. I'm more then happy to have you as the programmer who created a great peace of work!!!

closed 2019-05-29 by sidoh

poudenes, 2019-05-31:
Issue solved. Updated the Plugin to 0.0.9 and now sync group all work :D

poudenes, 2019-05-31:
Didn't know where to post it or contact you

in Wiki you can remove the line:

Note: Remember the Group All is not working yet. They still develop this part