sidoh/esp8266_milight_hub #472: feathure request: nRF24 send channel MQTT optional selection

starkillerOG, 2019-06-09:
I love the ESP milight hub project, it works great and I use it to control most smart bulbs in my house. The resent 1.9.0 release is great and provides excellent control over the channels over witch the commands are sent. Tested it thoroughly and found no bugs.

I have 2 ESP_milight_hubs, and with this new 1.9.0 release I have put both of them on different RF channels and that seems to work fine.
However I also have a script that checks all my smart devices twice a day and it lets ESP_milight_hub_1 sent a dummy bulb message over RF and looks if ESP_milight_hub_2 receives it and vice versa. Now that both of the hubs are on different RF channels this does not work anymore...

Would it be possible to have an extra optional field in the MQTT JSON bulb command to specify if "min", "mid" or "high" channel is used. If that extra channel field would not be specified the channel(s) as selected in the settings would be used?

sidoh, 2019-06-12:
Glad the new features are working for you, and the request makes sense.

Seems possible, although it would muddle the interface.  Would it work for you to have the `POST /raw_commands/:device_type` support this?  Even then I'm not positive I'd want to do this.

Not ideal, but could you temporarily change the configuration of one of the devices when performing this test?  Maybe have the sender send on all channels?  All configs are changeable through the REST API.

starkillerOG, 2019-06-13:
I guess I can make it work with the `POST /raw_commands/:device_type` interface.
If you post a valid raw_command to this interface it will show up on the other esp milight hub i think.

Would be awesom if you could make that possible!

I don't like changing my configuration just because of this test, the test is supposed to improve reliability and notify me if any problems arrise. Fiddeling with the configuration will only decrease reliability I guess.
For now I just disabeld this particilar test in my system_check script. (still doing IP pings to check if both hubs are still connected to my network).

sidoh, 2019-06-13:
Makes sense.  I'll take a closer look when I begin work on the next version (not sure when this will be, though).

Could you make each ESP listen on the channel the other one sends on?

starkillerOG, 2019-06-13:
Thank you for looking into it in the future!

No it is not practical to make each ESP listen on the channel the other one sends on.
If ESP1 sends on CHAN A and B, ESP2 needs to send on CHAN C to cover all channels.
In order to let the ESP read its own commands I have them listening on the same channel that they are sending:
ESP1 listens on CHAN A and ESP2 listens on CHAN C.
It is only possible to listen to a single channel.

If ESP2 needs to receive the messages of ESP1 you end up with ESP1 sending on all 3 channels.
if ESP1 needs to receive the messages of ESP2 it would need to send on both CHAN A an CHAN C.

That way CHAN A and C are completly overlapping making the point of having diffrent hubs on seperated channels pretty useless....

starkillerOG, 2019-06-13:
For now I just don't preform this check.

Thank you for all your hard work @sidoh, you are awesom!
Keep it up ;)

sidoh, 2019-06-13:
>  In order to let the ESP read its own commands I have them listening on the same channel that they are sending:

This isn't necessary.  Sent packets are intercepted in software.  They're not actually picked up by the radio.  I actually think this might not be possible, as the nRF can not simultaneously send and receive.

sidoh, 2019-06-23:
@starkillerOG, are you able to try this suggestion?

> Could you make each ESP listen on the channel the other one sends on?

starkillerOG, 2019-06-24:
Yes I can make it work then

sidoh, 2019-06-24:
Sweet, sounds good.  Gonna close this issue.  Please feel free to reopen if that ends up not working for you.

The cost of implementing this is a bit more memory to allow for overriding of radio properties on a per-packet basis.  Not terrible, mostly just adds complexity.

closed 2019-06-24 by sidoh