sidoh/esp8266_milight_hub #474: Add state field requirements to wiki for HA

ronschaeffer, 2019-06-13:
Please add to the [wiki](https://github.com/sidoh/esp8266_milight_hub/wiki/HomeAssistant) which MQTT state fields are required for optimum integration with Home Assistant.

I would create a PR, but I'm unsure what it correct. I'm currently using the following, which appears to work properly.

```
"group_state_fields":[  
      "state",
      "brightness",
      "saturation",
      "color",
      "mode",
      "color_temp",
      "bulb_mode",
      "computed_color",
      "effect",
      "device_id",
      "group_id",
      "device_type"
   ]
```

sidoh, 2019-06-13:
There should be defaults for this setting, which is why it's not documented on that wiki article (note that they are documented on [the README](https://github.com/sidoh/esp8266_milight_hub#mqtt)).

However... just tried clearing my config, and there were no defaults set.  This was a regression introduced by [this commit](https://github.com/sidoh/esp8266_milight_hub/commit/ba9aa46f9de1b964aacfca3996077a9a11143fe6#diff-2fdbe0571957f7450928b01325277c7d), which cleaned up some internals.

Will push a fix.

sidoh, 2019-06-13:
Will be fixed in 1.9.1.

sidoh, 2019-06-13:
Fixed the bug and added a line about making sure that config is set on [the wiki page](https://github.com/sidoh/esp8266_milight_hub/wiki/HomeAssistant#esp8266_milight_hub).

Thank you for reporting!

closed 2019-06-13 by sidoh

ronschaeffer, 2019-06-13:
That was quick. Thanks!