sidoh/esp8266_milight_hub #476: REST endpoint JSON {"status": true} not working as of 1.9.0 rc6 and above

coehn1, 2019-06-14:
Hi,

I user diyHue to control my MiLights through the Hue app.

ditHue sends comands to the REST /gateways endpoint.

To toggle the lights on/off diyHue uses:
{"status": true} and {"status": false}

As of 1.9.0 rc6 this doesn't work anymore. The false/off stae works but the true/on doesn't work anymore. I've tried these firmware version:
1.8.8     -> workst
1.9.0 rc4 -> workst
1.9.0 rc5 -> workst
1.9.0 rc6 -> doesn't work
1.9.0 rc7 -> doesn't work
1.9.0 rc8 -> doesn't work
1.9.0 rc8 -> doesn't work
1.9.0     -> doesn't work
1.9.1     -> doesn't work

Could you please check this.

sidoh, 2019-06-14:
That was the ArduinoJson v6 upgrade, I think.

Can publish a fix.

closed 2019-06-14 by sidoh

Hypernova007, 2019-07-20:
There is still an issue with the status command:
On version 1.9.2 (nodemcuv2):
- {"status": true} and {"status": false} are working
- {"status": on} and {"status": off} are not working

sidoh, 2019-07-20:
`{"status":on}` and `{"status":off}` are not [valid JSON](https://www.w3schools.com/js/js_json_datatypes.asp).

Try `{"status":"on"}`.

Hypernova007, 2019-07-20:
I wrote it wrong on this forum, but in my code it is {"status": "on"} and {"status": "off"} . I apologise.

I did some more investigation: {"status": "off"} is also working in version 1.9.2
But {"status": "on"} is not working in version 1.9.2

sidoh, 2019-07-20:
I'm not able to reproduce.  Works fine for me on 1.9.2 and 1.10.0-dev.7.

```
$ curl -X POST -H 'content-type: application/json' -d '{"status":"on"}' 10.133.8.194/gateways/0x118d/rgb_cct/1
{"status":"ON","level":100,"hue":240,"saturation":41,"bulb_mode":"color"}%

$ curl -X POST -H 'content-type: application/json' -d '{"status":"off"}' 10.133.8.194/gateways/0x118d/rgb_cct/1
{"status":"OFF","level":100,"hue":240,"saturation":41,"bulb_mode":"color"}%

$ curl 10.133.8.194/about
{"firmware":"milight-hub","version":"1.9.2","ip_address":"10.133.8.194","reset_reason":"Software/System restart","variant":"d1_mini","free_heap":22192,"arduino_version":"2_4_2"}%
```

PR #477 (2019-06-14): Bugfix: boolean values for state/status field are not properly recognized
Fixes #476.