sidoh/esp8266_milight_hub #480: KeyError: 'state'

Sn33jk, 2019-06-18:
Greetings, 
i keep getting error below when trying to setup the Entertainment zone in Hue app on iPhone

ERROR - light 10 is unreachable
Traceback (most recent call last):
  File "/opt/hue-emulator/functions/lightRequest.py", line 215, in syncWithLights
    if light_data["state"] == "ON":
KeyError: 'state'

updated the nodeNCU to 1.9.2.

sometimes Entertainment zone works, sometimes doesnt. Sometimes the room scenes works all, sometimes only a few. But syncing entertainment zone with Razer Chroma doesnt work.

sidoh, 2019-06-18:
I'm not familiar with the hue emulator, so difficult for me to say what's going on.  Would suggest opening an issue on that project.

If what you're trying to do is sending more than a couple of packets a second, what you're doing is probably not going to work very well in v1.9.  The v1.10 dev release may work better, but this is still a pretty early dev version.

sidoh, 2019-06-19:
@Sn33jk, if you're able to provide further info, happy to take a closer look.  Otherwise, I'll close this issue in ~a week (happy to continue discussion after that, of course).

closed 2019-07-01 by sidoh