sidoh/esp8266_milight_hub #485: "nRF24 Listen Channel" setting in v. 1.10.0-dev.4

IvoTr, 2019-07-04:
Hi Sidoh,

It's a great project!

Just to let you know that in 1.10.0-dev.4 version the "nRF24 Listen Channel" setting on each reboot returns to the default value.

Also I would like to suggest you to add one more (Enable/Disable) setting in the Radio section  for constant sniffing (updating the button on the top strip of the Web UI respectively). Thank you very much in advance.

Regards,
Ivo

sidoh, 2019-07-04:
Hi Ivo,

I'm having a hard time replicating this -- i can set it to any of LOW, MID, HIGH, and the setting sticks after a reboot.  I'm also on 1.10.0-dev.4.  Are you able to post the output of `/settings`?  (Make sure to sensor any passwords)

it actually already does passively listen.  the UI doesn't connect automatically, though, thus the button.

IvoTr, 2019-07-05:
Hi Chris, 

Yes, of course. Just give me a few days and I’ll come back to you. The setup is in my home in Sofia but I’m in London at the moment. It’s pretty hard to live and work in two cities at the same time... :)

Regards, 
Ivo

---
Dr. Ivo Trayanov

Managing Director 
Haldane Systems Ltd.

E: i.trayanov@haldanesystems.co.uk
M: +359878777115 / +447796070060
W: http://www.haldanesystems.co.uk/

(Sent from my iPhone)

sidoh, 2019-08-06:
@IvoTr, any luck here?

sidoh, 2019-08-30:
Please let me know if you'd like to reopen the issue.  Closing due to inactivity.

closed 2019-08-30 by sidoh