sidoh/esp8266_milight_hub #487: Group 1 CCT not working

cometfish, 2019-07-15:
Using Group 1 on the milight hub no longer seems to work for my CCT lights on the pre-release version. Groups 2-4 are still working fine though. The physical remote with the same device ID is still working (so the light hasn't become unpaired).

I was on 1.10.0-dev5, and have just upgraded to 1.10.0-dev6 to see if that fixed it, but unfortunately the same thing is happening.

Sniffed packet from remote (which works):
```cct packet received (7 bytes):
Request type  : 5A
Device ID     : 26A6
b1            : 01
b2            : 0B
b3            : 26
Sequence Num. : 5F
```

Sniffed packet from milight hub (does not work):
```cct packet received (7 bytes):
Request type  : 5A
Device ID     : 26A6
b1            : 01
b2            : 0B
b3            : 08
Sequence Num. : 41
```

sidoh, 2019-07-15:
For the parts that matter, those are the same packets.  The labels aren't accurate for CCT... should probably fix that.

For CCT packets, `b3` is a sequence number (you'll notice that if you send multiple packets, it'll just increment).  The last byte is a checksum.

Things to try, if you haven't already:

* Try moving the ESP8266/nRF24 really close to the group 1 bulb and resending
* Try increasing the number of repeats in settings
* Keep the ESP8266 and nRF24 a reasonable distance apart.  They're both 2.4 GHz devices, and can interfere with one another.

cometfish, 2019-07-16:
Thanks for the quick reply!

> Try moving the ESP8266/nRF24 really close to the group 1 bulb and resending

The group 1 bulb is actually the closest of the four - less than a metre away. 

> Try increasing the number of repeats in settings

Hmm ok, I upped it to 500 and it did seem to help - the on/off worked maybe 50% of the time. It was slow though, wasn't lightning fast like the other bulbs. 

> Keep the ESP8266 and nRF24 a reasonable distance apart. They're both 2.4 GHz devices, and can interfere with one another.

Not sure how this would affect group 1 but not the others :)

When I get a chance in the next day or two I'll reflash back to 1.9 and see if that helps.
I previously had the hub running for several months on 1.9 and it was rock solid and fast - the bulbs and hub (and nRF24) are in the same spots they were then.

sidoh, 2019-07-17:
Got it.  It being slow means the bulb isn't picking it up until after several repeats, which supports the theory that the bulb is just missing packets.  It's also pretty strong evidence that there's not an issue with the structure of the packets.

It could be an antenna orientation thing.  Bulbs use a monopole antenna, which will not perform as well in all orientations.  

Stuff to try:

* Maybe try shifting the orientation of the bulb and/or nRF24 around?  I know you mentioned the devices are in the same spot, but I'm sure you know how fiddly electronics can be sometimes =)
* Move the group 1 bulb to a spot where others are working
* Just for the sake of being thorough, try 1.9.2 and see if it makes a difference.  There was a small change to the radio code in 1.10.0-dev4, but if it had any effect I expect that it would be all-or-nothing.  It also shouldn't apply to CCT.

cometfish, 2019-07-19:
> Maybe try shifting the orientation of the bulb and/or nRF24 around? I know you mentioned the devices are in the same spot, but I'm sure you know how fiddly electronics can be sometimes =)

Pleased to report this seemed to be the issue!
I moved the esp to next to my group 3 bulb, and group 1 started working fine (group 3 was fine too). Moved it back to where it was, near the group 1 bulb, and now group 1 responds everytime, fast.
Very odd. Thanks for your help :)

closed 2019-07-19 by cometfish

sidoh, 2019-07-20:
Funky.  Glad it's working!