sidoh/esp8266_milight_hub #491: Pairing/ Unpairing Bulbs via Web

JMacIV, 2019-07-22:
Hey Chris - Hope you have been well, man. 

Was trying to move some bulbs between some groups tonight, but could not get it work via Web UI. Went ahead and flashed my Node chip to 1.10-wip to see if that would work.

Continued to attempt to unpair via WebUI from both desktop and phone to no avail. (Yes, I was following the shut off power to bulb, restore power, press pair or unpair within 5 seconds.) 

I knew that I was on the correct groups and IDs because I was able to control them, however, they were not sending the pair and unpair. 

Fired up MQTT since I use it for integration with openHAB. Did power off and on procedure and then sent {"command":"unpair"} to "milight/commands/0x1B0C/rgb_cct/2". Sure enough, had instant success like that. 

Tried to move that same bulb to a new group on the same device ID, but Web UI buttons failed to work. Followed power off/on procedure, press Pair within 5 seconds, no dice. Repeated power off/on and then sent {"command":"pair"} to "milight/commands/0x1B0C/rgb_cct/1" on MQTT, success again. 

Got what I needed to done, but wanted to raise a ticket in case others run into same issue. FYI - This is same ESP chip and NRF setup that I have been using for months and months. 

Thanks - John

sidoh, 2019-07-22:
Huh.  REST and MQTT commands flow through the same code path, so one shouldn't work without the other working.  I was able to click on the button the UI and have it work.

Which browser are you using?  Does anything interesting show up in your browser's developer console?

JMacIV, 2019-07-22:
Super strange. I will take a look later tonight when I have some time. 

Browser was Safari on both laptop and phone (MacBook Pro and iPhone).

sidoh, 2019-07-22:
Just tested with safari as well and it seems to work.  this is after pressing the pair button --

[image]

poudenes, 2019-07-23:
Question as well about pair and unpair. When you do a unpair. Will only the bulb unpaired when you turn it on and then hit the unpair, of will all bulbs removed?

In official Milight app when you do a unpair, all bulbs are removed from that group.

JMacIV, 2019-07-23:
Man, that is strange on my Safari not working. I just enabled Sniffing on my hub and first press of unpair, packets were sent. I can’t explain why it didn’t work the other day, super strange. It was able to control the lights, but not unpair, so not sure. Either way, probably cannot recreate, so we can close. 

@poudenes - Bulbs will only be removed that have been power cycled within last 5 seconds. I just tested by sending unpair command to one of my groups with multiple bulbs. No bulbs were removed as I did not power cycle any of them :)

Are you sure that is how official Milight app works? It is supposed to only pair/unpair power cycled bulbs.

sidoh, 2019-07-27:
Thanks for the followup, @JMacIV.  Could've been some dumb javascript error.  I'll close this issue for now, but let me know if you run into it again.

@poudenes, espMH emulates exactly the real milight unpair process, which is just to send a command packet that almost always has another meaning (varies by remote type, but it's usually the "on" command).  The only way unpairing can work reliably in this way is if it's coupled with a recent physical power toggle.

closed 2019-07-27 by sidoh