sidoh/esp8266_milight_hub #495: FUT095 remote support (4 Zone RGBW)

kranendijk, 2019-07-29:
Super project!
Bought myself a FUT014 with FUR095 remote. For some reason the remote actions (on/off) are not picked up by the espMH: on/off status in Domoticz is not updated, mqtt settings verified. Controlling the bulb with remote/espMH/domoticz-dashticz works, just the status is not updated when using the remote.

Changed remote espMH name according to https://docs.google.com/spreadsheets/d/1eMNn5HmMzNc0LgRBQlcnDYeE7KKCKBPMd15dMbIpqfM/edit#gid=0. 

More info about the remote http://futlight.com/productdetails.aspx?id=173&typeid=143

If I can be of help by checking things within my knowledge: happy to help!

sidoh, 2019-07-30:
Are you sure both the remote and the espMH are working?  That remote is supported.

kranendijk, 2019-07-30:
I am able to control the bulb using esp8266 interface, remote and domoticz/dashticz. Only when using the remote I do not see any updates on the bulb's status (on/off) in domoticz. It seems then - since you indicate the remote is supported - I am having some error somewhere... What can I check? I see the status led on the ESP8266 blinking when pressing a button on the remote though.

Should this not be the place to address this (as I am quite new to the wole github thing) then let me know!

sidoh, 2019-07-30:
It sounds like the remote might have a different ID than the one you're using with espMH.  If that's the case, state for the remote will be getting published to a different topic.  If this ends up being true, your options are:

(A) Use the remote's ID

1. Unpair the bulb with the ID espMH is using (see the wiki for unpair instructions)
2. Make sure they're still working with the remote
3. Use the remote's ID in espMH, update domoticz accordingly

(B) Forward updates from the remote topic to the topic domoticz is using for state

[Here is an example of doing that with HomeAssistant](https://github.com/sidoh/esp8266_milight_hub/wiki/Using-Milight-Remote-with-HomeAssistant).

(A) is definitely more straightforward and less error-prone.

kranendijk, 2019-07-30:
Thanks a lot so far. Up until method a point 2 I understand. Getting the remote ID is beyond... I shoud use the REST endpoint (aka http://my_espMH_ip and the command "GET /gateway_traffic(/:rgbw)?" correct? I tried in soapUI, no luck...

sidoh, 2019-07-30:
Just enable sniffing in the UI with this button:

[image]

and press some buttons on your remote.  packets will show up with the device ID decoded.

kranendijk, 2019-07-30:
I must be doing something wrong?? Thanksfor the patience :)

[image]

sidoh, 2019-07-30:
did you click the button again?  that would make it stop sniffing.  after clicking, it should look like this:

[image]

(notice the button changes)

kranendijk, 2019-07-30:
Yes I did. Pressed almost all buttons on the remote and expected to see some code or... am I wrong in that assumption? I expect all components to be fully functional as I see rge led on the esp flicker when remote is pressed. Rf module should be functioning then.

sidoh, 2019-07-30:
Yes, if the LED is flickering it's seeing packets.  If they're not showing up in the UI it's either because listening is/was disabled, or because the web socket connection is failing.  Can you check the developer console for any errors?

As a workaround, you can request `http://<your-esp-ip>/gateway_traffic` and press a button on the remote.  This will display a single packet.

kranendijk, 2019-07-31:
Working, issue will be closed by me. For testing I attached the espMH to a power supply only, apparently not strong enough? I attached it to my laptop via USB, receiving signals in a sudden. No flicker when on power supply, flickering when on USB. So I attached the espMH to the rpi3's USB, found the remote ID, added it as device ID 0x<remote ID> and voila. It worked!

Great job in helping me out, thanks!

closed 2019-07-31 by kranendijk