sidoh/esp8266_milight_hub #496: Effects set in home assistant have no effect

razvanws, 2019-07-30:
I'm using v1.10.0-dev.8, FUT089 and home assistant and it seems there's an issue on triggering effects from home assistant interface.

This is my relevant hass config:
```
- platform: mqtt
  name: 'Living room downlights 1'
  qos: 1
  schema: json
  color_temp: true
  rgb: true
  brightness: true
  retain: true
  command_topic: milight/0x0/fut089/1
  state_topic: milight/states/0x0/fut089/1
  availability_topic: milight/client_status
  payload_available: 'connected'
  payload_not_available: 'disconnected'
  effect: true
  #Effect list 0=Full, 1=White, 2=RGB, 3=Full hard, 4=Disco, 5=Red Flash, 6=Green Flash, 7=Blue Flash, 8=White Flash]
  effect_list: [0, 1, 2, 3, 4, 5, 6, 7, 8]
```

When I trigger effect 2 in hass, the following mqtt messages are sent and nothing happens:
```
milight/0x0/fut089/1 {"state": "ON", "effect": "2"}
milight/updates/0x0/fut089/1 {"state":"ON"}
milight/states/0x0/fut089/1 {"state":"ON","brightness":125,"color_temp":370,"bulb_mode":"white","effect":"white_mode"}
```

However, if I trigger mode 2 in ESP8266 web interface, the following mqtt messages are sent:
```
milight/updates/0x0/fut089/1 {"mode":2}
milight/states/0x0/fut089/1 {"state":"ON","mode":2,"bulb_mode":"scene","effect":"2"}
```

So, it seems to me that in order for the effect to work, besides "effect": "2", home assistant should also send "bulb_mode": "scene". 
I'm not sure if home assistant is expected to send the "bulb_mode" attribute or the firmware should automatically set "bulb_mode": "scene" whenever "effect" != "white_mode".

Whenever I'm changing from white to color (in home assistant), "bulb_mode" is adjusted accordingly so I have a feeling this is really a bug on milight_hub software.

sidoh, 2019-07-30:
Does this work as expected with 1.9?

i think it's probably just not recognizing the `"effect"` key.  I refactored the command handler a bit in 1.10, and may have missed stuff like this.

razvanws, 2019-07-30:
I haven't tested 1.9 as I just bought the nodemcu and flashed the latest version. Do you want me to test 1.9?

sidoh, 2019-07-30:
Yes please. :)

razvanws, 2019-07-30:
For some reason 1.9.2 doesn't build for me, do you have any idea?
```
.pio/libdeps/nodemcuv2/WebSockets_ID549/src/WebSocketsClient.cpp: In member function 'void WebSocketsClient::connectedCb()':
.pio/libdeps/nodemcuv2/WebSockets_ID549/src/WebSocketsClient.cpp:749:22: error: 'class axTLS::WiFiClientSecure' has no member n
amed 'setInsecure'
         _client.ssl->setInsecure();

```

sidoh, 2019-07-30:
yes, it's a bug in an upstream library.  patched a fix in 1.10, but didn't backport to 1.9.

you don't need to compile yourself -- just use the precompiled version:

https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.9.2

razvanws, 2019-07-30:
I can confirm that changing effects from hass interface works perfectly with 1.9.2.

sidoh, 2019-07-30:
Great.  Thanks for reporting!

I'll add a test recreating the issue and push a fix in the next few days.

sidoh, 2019-07-31:
Fixed in [1.10.0-rc.1](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.10.0-rc.1).

razvanws, 2019-07-31:
I can confirm that the issue was fixed in 1.10.0-rc.1. Thanks a lot!

strumf666, 2019-08-01:
Thanks for the fix, works for me as well.

razvanws, 2019-08-01:
By the way, I think this fix triggered some other bug that color doesn't "stick". Steps to reproduce.
1. Select an effect (0 let's say)
2. Select white-mode effect
3. Choose an rgb color from hass color picker
4. The light will change to that color and then it will return to white mode
5. Now same thing happens if you try to change the color from esp webpage.

I tried again 1.9.2 and this bug doesn't replicate on 1.9.2.

sidoh, 2019-08-01:
It's probably a different bug.  Don't think this would've triggered behavior like this.

Could you please open another issue? :)

PR #497 (2019-07-31): Bugfix: handle effect key
Fixes #496.  This was a regression in the v1.10 branch.
closed 2019-08-30 by sidoh