sidoh/esp8266_milight_hub #499: Added support for HEX color support in payload?

MrSiO, 2019-07-31:
Is there a way (or consider as a feature request) to use the hex color values (#FFFFFF) in the MQTT color payload instead of RGB?

Many MQTT dashboards use HEX natively.

Thx for considering.

sidoh, 2019-08-02:
Seems fine, although I think it'd have to be prefixed with the `#`.  It already supports a string in the form of `r,g,b`, and parsing would be a bit gross without the prefix.  

Would you need it to be two-way?  It's pretty easy to make it write-only, and a bit harder to make it read/write.  Write-only means you can send commands in this format, but state updates sent by the hub cannot be.

MrSiO, 2019-08-02:
Write only would be a great.
I did a test and yes the # is included in the payload:

Eg:
#db8921

Thx for considering!

sidoh, 2019-08-12:
Will be in the next 1.10 release.  Added a field as well so it's read/write.

MrSiO, 2019-08-14:
Excellent! Thx alot!

closed 2019-08-30 by sidoh