sidoh/esp8266_milight_hub #500: problems controlling brightness dual white cct lamps

hcu00, 2019-07-31:
**Thumbs up. Super project! Thanks**  :-)

I have MiLight FUT011, 017 and 019 dual white lamps, which I control via UDP (v6) from HASSIO. If I pair them as RGB+CCT and adjust it in lights.yaml config to rgbww, brightness works as expected (with 255 steps).  
However, if I pair them as CCT and change it in the lights.yaml config to white, adjusting brightness gives problems. If I adjust it with the slider, the adjustment is much larger than expected. Also hard values don't work well. I tried using 100 and 255 steps.
The problems remain when I use MQTT to control them.

Any idea what is going wrong?

greetings from the Netherlands, Herco

sidoh, 2019-07-31:
If you _can_ use RGB+CCT, definitely use it.  The CCT protocol is far more limited.  FUT091 is also significantly better and is CCT-only.

The biggest reason that the CCT protocol sucks is that there is no packet to directly set brightness or white temperature.  Instead, there are commands to increase or decrease it.  espMH hacks around this by spamming decrease packets so that there's a known value, and then sending the appropriate number of increase packets.

In order to accomplish this, it has to assume the number of intervals.  The device you're using might have a different number than the ones are testing with.

I think FUT011 and FUT017 are not compatible with the better remote types, but FUT019 is.  You might be stuck with CCT for the first two.

Sidebar: if you're using HomeAssistant, highly recommend that you switch to MQTT: https://github.com/sidoh/esp8266_milight_hub/wiki/HomeAssistant

hcu00, 2019-08-01:
Thanks for the quick and clear answer. FUT011 and 017 work also with the FUT091 remote controller. So, I will try pairing them that way, while accessing it by MQTT.
The reason I prefer not to use RGB+CCT is that it shows the color graph while manually adjusting the lights in HASSIO.

sidoh, 2019-08-06:
Great.  Sounds like you've got it handled, but please let me know if you'd like to reopen the issue.

closed 2019-08-06 by sidoh

hcu00, 2019-08-06:
Hi Chris, 

MQTT worked, however I returned to UDP because I use a 900 seconds simultaneously color-temperature and brightness transition in the morning and evening. Besides that, when turning on, one light should remain at 25% instead of 100%. This does not work with MQTT (yet). 

The problems I had were solved by soldering a 47mF 16V capacitor to the back of the NRF24 (antenna model). It allowed me to lower the radio-repeats to 10. (I have not tested lower values yet). I also added a longer delay in the home assistant scripts after each transition and before turning that light off. My first impression is that it works flawlessly. :-)

I am really happy with it.  Thanks man for all your efforts and your professional and cooperative attitude. :-)  

greetings,

Herco