sidoh/esp8266_milight_hub #501: bulb flicker while transitioning

badstraw360, 2019-07-31:
I really love that we can use transitions now, you're amazing!  =D

If I dim the lights from 100% to 0% (off) using transitions it works perfectly, but if I go from off to 100% it seems like there is a notorious flickering light effect while it's ramping up the brightness. 

Sometimes it goes straight to 100% like if it's ignoring the command. 

Should I enable the debug in hass or you can reproduce it ? 

Thanks again for your hard work.

sidoh, 2019-08-02:
The transitions seem to work as expected when I interface with the hub directly:

### Transition off

Command:

```
$ curl -X PUT -H 'content-type: application/json' -d '{"status":"off","transition":2}' '10.133.8.194/gateways/0x118d/rgb_cct/1?blockOnQueue=true'
```

State updates from MQTT:

```
milight_test/state/0x118D/rgb_cct/1 {"status":"ON","level":100,"color_temp":370,"bulb_mode":"white","kelvin":100,"effect":"white_mode"}
milight_test/state/0x118D/rgb_cct/1 {"status":"ON","level":60,"color_temp":370,"bulb_mode":"white","kelvin":100,"effect":"white_mode"}
milight_test/state/0x118D/rgb_cct/1 {"status":"ON","level":20,"color_temp":370,"bulb_mode":"white","kelvin":100,"effect":"white_mode"}
milight_test/state/0x118D/rgb_cct/1 {"status":"OFF","level":0,"color_temp":370,"bulb_mode":"white","kelvin":100,"effect":"white_mode"}
```

### Transition on

Command:

```
$ curl -X PUT -H 'content-type: application/json' -d '{"status":"on","transition":2}' '10.133.8.194/gateways/0x118d/rgb_cct/1?blockOnQueue=true'
```

MQTT:

```
milight_test/state/0x118D/rgb_cct/1 {"status":"ON","level":0,"color_temp":370,"bulb_mode":"white","kelvin":100,"effect":"white_mode"}
milight_test/state/0x118D/rgb_cct/1 {"status":"ON","level":20,"color_temp":370,"bulb_mode":"white","kelvin":100,"effect":"white_mode"}
milight_test/state/0x118D/rgb_cct/1 {"status":"ON","level":80,"color_temp":370,"bulb_mode":"white","kelvin":100,"effect":"white_mode"}
milight_test/state/0x118D/rgb_cct/1 {"status":"ON","level":100,"color_temp":370,"bulb_mode":"white","kelvin":100,"effect":"white_mode"}
```

Yes, if you could attach some debug logs + a way to reproduce, would be great.  Thanks!

badstraw360, 2019-08-05:
Sorry for the late reply, It was because I migrated everything to a NUC with ubuntu. Im using hassio in a docker container, and I'm still experiencing the same issue. 

The easiest way to reproduce it is to use the service dev tool in home assistant: 

```
{
  "entity_id": "light.room",
  "transition": 20,
  "brightness_pct": 80
}
```

Should I enable the logger in HASS? let me know if I need to compile it to enable the debug flag, I'll be happy to help as much as I can.

sidoh, 2019-08-06:
Ah, I see what's going on.

It's generating two transitions which are fighting with each other -- one from `{"status":"ON"}` and another from `{"level":80}`.  The semantics for a command like `{"status":"ON","transition":20}` are "turn the light on, and transition from last known brightness to 100%."

I think the right behavior is to skip generating a transition for the `status` field if a `brightness` field is also specified.

Thanks for reporting!

sidoh, 2019-08-06:
fixed in the 1.10.0-wip branch.  i'll tag a release in the next day or two if there are no additional changes.

badstraw360, 2019-08-08:
Yay!!! I'm testing 1.10.0-rc.3 and now it's dimming very smoothly, however If I turn off the light in HASS and then use the service dev tool with the following: 

```
{
  "entity_id": "light.room",
  "transition": 20,
  "brightness_pct": 80
}
```

The light jumps straight to 80% like it's ignoring the "transition" command. But if I set it to dim the light to 0% first, and then 80% it works. 

```
{
  "entity_id": "light.room",
  "transition": 20,
  "brightness_pct": 0
}
```

```
{
  "entity_id": "light.room",
  "transition": 20,
  "brightness_pct": 80
}
```

It gives me the impression that it's ignoring the state from previous brightness, but I still have to try it using an automation. 

Thanks again.

badstraw360, 2019-08-08:
I'm trying this automation: 

```
-  alias: 'test'
   trigger:
     - platform: state
       entity_id: light.room
       from: 'off'
       to: 'on'
   action:     
     - service: scene.turn_on
       entity_id: scene.dim

```

```
- name: dim
  entities: 
    light.hall:
      state: on
      transition: 20
      brightness_pct: 100 
```

When I run the automation ```light.hall``` jumps straight to 100% without transitioning.
The transition works if I set light.hall to 1% before running the automation.

sidoh, 2019-08-09:
The semantics of a command like `{"brightness":80,"status":"ON","transition":20}` are:

1. Turn the bulb ON
2. Transition from the current brightness to 80

So if the previous brightness was 80, nothing would happen.  If it was 100, it would transition _down_ from 100 to 80.

It sounds like this is not what you expect?

badstraw360, 2019-08-09:
If the light is off and I want it to transition to 80%, then it turns on at full brightness.

So it works fine now as long as the light is on before the transition.

sidoh, 2019-08-09:
> If the light is off and I want it to transition to 80%, then it turns on at full brightness.

Does this mean:

1. Transition from 0% to 80% brightness over the specified transition duration
2. Afterwards, immediately go to 100%?

Not sure I follow why that's an intuitive way to interpret the command.

> So it works fine now as long as the light is ON before the transition. So I guess it has to do with the light.turn_on command, because if the light is already on then it would transition down from 100 to 80 like it's supposed to.

`light.turn_on` just sends the `{"state":"ON"}` command.

When the bulb turns on, it'll be at whatever brightness it was at when it was last on.  So if you have some interaction like so:

1. Turn light on
2. Set brightness to 60%
3. Turn light off
4. Send `light.turn_on` with `{"brightness_pct": 80,transition": 5}`

the bulb will turn on at 60% brightness and transition from 60% to 80% brightness over 5 seconds.

It does not seem unreasonable to expect (4) to first set brightness to 0 and then transition up to 80%.  But honestly it doesn't seem totally unreasonable for it to have the current behavior either.

> Is there a possibility to go from OFF to 80%?

The easiest way with HASS is to either: (1) make sure the bulb is at 0% brightness when it's turned off, or (2) set it to 0% brightness immediately when it turns on.

The transition API supports what you want, but HASS' `mqtt` light component is limited to defining transitions where the start state is unspecified.  So we either assume the start state is some value by convention or use the current value (which is what's done now).

poudenes, 2019-08-11:
I guess transition is only working with 1.10.0 firmware?
was trying this also but use 1.9.2

sidoh, 2019-08-11:
Yes, it's a new feature in 1.10.

sidoh, 2019-08-12:
@badstraw360, I decided that the behavior you were expecting is probably more intuitive, so I made the change.  Will be live in 1.10.0-rc.4.

badstraw360, 2019-08-12:
> @badstraw360, I decided that the behavior you were expecting is probably more intuitive, so I made the change. Will be live in 1.10.0-rc.4.

Great thanks!!! I'm testing it right now..! 

According to the readme:

>You can transition to a given status or state. For example,
>{"status":"ON","transition":10}
>will turn the bulb on, immediately set the brightness to 0, and then transition to brightness=255 >over 10 seconds. If you specify a brightness value, the transition will stop there instead of 255.

I was trying to achieve this:

-light state = OFF >>transition>>80%  in 20 seconds

The light should turn on, set brightness to 0, and fade to brightness 80 in 20 seconds. 

But instead of that, this happens:

-light state = OFF >>>> 100% in 1 second

The light won't fade, it turns ON at 100 brightness in one second. 

Actually this code won't work either, it won't fade to 100% so I don't know if I'm making a mistake or something: 

Service: light.turn_on
entity: light.room

```
{
  "entity_id": "light.room",
  "transition": 10
}
```

It works if I go from full brightness to off though.

sidoh, 2019-08-13:
hahaha, I know what's going on.  Travis-CI is gas-lighting me.

I just spent 20 minutes squinting at the integration tests I wrote that were supposed to ensure this behavior was being implemented:

```ruby

    it 'should transition from off -> on from 0 to 100, even when there is a last known brightness' do
      seen_updates = {}
      @client.patch_state({status: 'ON', brightness: 99}, @id_params)
      @client.patch_state({status: 'OFF'}, @id_params)

      @mqtt_client.on_update(@id_params) do |id, message|
        message.each do |k, v|
          seen_updates[k] ||= []
          seen_updates[k] << v
        end
        seen_updates['brightness'] && seen_updates['brightness'].last == 255
      end

      @mqtt_client.patch_state(@id_params, {state: 'ON', transition: 1.0})

      @mqtt_client.wait_for_listeners

      transitions_are_equal(
        expected: calculate_transition_steps(start_value: 0, end_value: 255, duration: 1000),
        seen: seen_updates['brightness'],
        # Allow some variation for the lossy level -> brightness conversion
        allowed_variation: 4
      )
    end
```

tests were passing, and it seemed like it was perfectly replicating what you were doing.

Turns out it was.

I had nuked an old release because I wanted to push this change into it.  Travis had already uploaded the artifacts from the old build, though, and was refusing to clobber the artifacts it had already uploaded.

That'll teach me for re-using tags that have successfully built...

I just killed all of the artifacts and restarted the Travis build.  Hopefully that fixes it.

sidoh, 2019-08-13:
Confirmed that the 1.10.0-rc.4 artifact that _was_ up had the old behavior, and that the current one has the new behavior.  Please try again.

badstraw360, 2019-08-14:
Now it always transitions from Brightness = 0 which is nice if the bulb is off, but if I set an automation in Hass to dim the bulb from 100% to 20 % brightness, it goes to 0% and then transitions to 20%. 
In my opinion it was a good idea to make the bulb transition from previous brightness, because it could go from 100%>1% and 1% to 100% . The only issue was if the bulb was turned off, because it wouldn’t transition unless I set the bulb to 0% or 1% manually in hass. 

Is there an easy way to change this behavior?

poudenes, 2019-08-14:
Maybe a check: of bulb is off then start transition from 0%

Otherwise it's on and it have already a brightness level

sidoh, 2019-08-26:
Ah, right.  I keep forgetting that the only way HomeAssistant allows you to transition is with the `light.turn_on` service, which translates to a `{"status":"on"}` command (this is sort of silly, honestly).

Yes, it makes sense to skip the set-to-0 behavior if the light is already on.

I've been away from home for the last couple of weeks, and will probably still be a bit slow on making this change.  This is probably the last behavior that's keeping me from the 1.10 final release, though, so it'll hopefully happen soon =)

sidoh, 2019-08-30:
Should be fixed in the 1.10.0 release.

closed 2019-08-30 by sidoh

badstraw360, 2019-08-31:
I'm still experiencing the same issue with the latest version, but now the difference is that while it's transitioning, sometimes the bulb flickers and stops before reaching the brightness level. 

It can be reproduced easilyusing the service dev tool in HASS, just try to adjust the bulb to different brightness levels keeping the same 20 seconds transition. 

Let me know if I need to open another issue please, or if I can help in some way.

sidoh, 2019-08-31:
Are you able to share MQTT logs?  I'm not able to reproduce this.

Here's me doing what you're suggesting, with the MQTT logs on the side.  Seeing nothing but smooth transitions from one value to another.  The lamp is right next to me, and is behaving as I'd expect.

[image]

Is it possible you have more than one transition running at a time?  This would definitely cause the flickering behavior you're describing.  You can see active transitions using the `/transitions` API route.

badstraw360, 2019-09-01:
Should I compile the firmware and enable the debug flag? It happens when I'm using the HomeKit app look: 

[image]

badstraw360, 2019-09-01:
Should I compile the firmware and enable the debug flag? It happens when I'm using the HomeKit app look:
 
[image]

> Are you able to share MQTT logs? 

Sure, but I have a lot of MQTT components. How do I enable the logger for the lights only. 
I'm using this code:

```
logger:
  logs:
    homeassistant.components.mqtt: debug
```

> Is it possible you have more than one transition running at a time? This would definitely cause the flickering behavior you're describing. You can see active transitions using the `/transitions` API route.

I'm using only one transition at a time. I wish I could help in some way.

sidoh, 2019-09-01:
This really looks like something is sending a transition command while one is already active.  In the original bug, the software itself was doing this because it was enqueuing transitions for both the "status = on" and "brightness = X" commands, but it now de-duplicates when those appear in the same command.

> Sure, but I have a lot of MQTT components. How do I enable the logger for the lights, not everything?.

Use `mosquitto_sub`:

```
mosquitto_sub -h your-mqtt-server -u username -P password -t 'milight/#'
```

where `milight/` is the prefix you care about.

> I'm using only one transition at a time. I wish I could help in some way.

The MQTT logs will be helpful.  As I said in the last post, the output of `/transitions` while you're seeing the flickering would help too.

badstraw360, 2019-09-02:
> This really looks like something is sending a transition command while one is already active. In the original bug, the software itself was doing this because it was enqueuing transitions for both [...]

Probably, maybe HomeKit has something to do with it.. 

is it ok if I enable the debug like this? 
```
logger:
  logs:
    homeassistant.components.mqtt: debug
  ```

Btw Im getting an "Error: Connection refused " when I run the command if I try to connect to the ip that I got from the Mosquitto broker addon which is 192.168.1.190, but if I try to connect using the same ip that I'm using for hassio I just get a black screen, but I don't receive any output when I turn the bulbs on or off. Like it's waiting for an mqtt message. 

I'm gonna have to check why I'm I can't connect to my MQTT broker.

badstraw360, 2019-09-03:
Here is the output: 

```
lights/commands/0xF63/rgb_cct/2 {"state": "ON", "brightness": 255}
lights/commands/0xF63/rgb_cct/2 {"state": "ON", "brightness": 2}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":10,"level":4,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":13,"level":5,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/commands/0xF63/rgb_cct/2 {"state": "ON", "brightness": 12}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":15,"level":6,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":255,"level":100,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":3,"level":1,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":20,"level":8,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":13,"level":5,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":28,"level":11,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":36,"level":14,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":41,"level":16,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/commands/0xF63/rgb_cct/2 {"state": "ON", "brightness": 35}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":48,"level":19,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":54,"level":21,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/commands/0xF63/rgb_cct/2 {"state": "ON", "brightness": 40}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":61,"level":24,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":69,"level":27,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/commands/0xF63/rgb_cct/2 {"state": "ON", "brightness": 53}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":74,"level":29,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":36,"level":14,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":82,"level":32,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":87,"level":34,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":89,"level":35,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":97,"level":38,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":99,"level":39,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":102,"level":40,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":107,"level":42,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":117,"level":46,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":122,"level":48,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":130,"level":51,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
[... cut ...]
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":242,"level":95,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":204,"level":80,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/commands/0xF63/rgb_cct/2 {"state": "ON", "brightness": 204}
lights/states/0xF63/rgb_cct/2 {"state":"ON","status":"ON","brightness":204,"level":80,"kelvin":80,"color_temp":327,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}

```

sidoh, 2019-09-03:
Something else is sending commands while the transition is happening.  What is it?  This is almost certainly your problem.

badstraw360, 2019-09-04:
It’s HomeKit I’m completely sure, I can’t open it and force press the icon to see the transition in real time from the app without experiencing a weird flickering effect.

Maybe there is a delay between HomeKit and Hass and the sliders tries to compensate the changes.

diplix, 2019-09-13:
> The transitions seem to work as expected when I interface with the hub directly:
>
> ### Transition off
> Command:
>
> ```
> $ curl -X PUT -H 'content-type: application/json' -d [...]

if i understand this right, every state-change during transition is beeing re-transmitted via status-messages? shouldn’t there be radio silence while doing a transition? no other light-platform i know monitors the transition state: you just fire the command and let the light/platform handle the transistion by itself. after the transistion is done (or after it started?), the **final state** should be transmited, but nothing more.

when a light platform gets status updates for each step of a transition interference is bound to happen, because every linked platform and component acts on it (frontend state updates, homekit state updates, log and history entries are made for each step).

wouldn’t that be a solution to let the milight hub handle the transition silently or did i misunderstand something?

sidoh, 2019-09-13:
> It’s HomeKit I’m completely sure, I can’t open it and force press the icon to see the transition in real time from the app without experiencing a weird flickering effect.
> 
> Maybe there is a delay between HomeKit and Hass and the slider tries to compensate the changes.

@badstraw360 -- sounds like doing this causes it to send command packets?  Command packets are definitely going to interfere with transitions.  The only way I see around this is:

1. Ignore packets that have an active transition (this seems like it'd be really confusing)
2. Cancel a transition when a command is received for the same device (also seems like it'd be confusing).

> I think I'm experiencing this same issue in Home Assistant. My MiLights are going crazy when they transition from the Circadian Lighting Custom Component and might have burnt out my radio (at least, it seems fishy to me that it died a few days after this behavior). Started after I upgraded my milight-hub firmware.
> 
> What logs would be useful for me to provide?

@jkms -- Yikes!  For what it's worth, I've not managed to burn a radio yet, and I've let the nRF blast packets for hours at a time.

MQTT logs during the transition would help, but a script that allows me to reproduce the issue myself is 100x more useful.

> if i understand this right, every state-change during transition is beeing re-transmitted via status-messages? shouldn’t there be radio silence while doing a transition? no other light-platform i [...]

@diplix, state is not updated for every packet.  It's governed by a throttle configuration which defaults to 1000ms.  If you have _updates_ turned on, then yes, you'll get an update for every packet.  But updates are meant to have these semantics.

I'm not sure I see how this is going to cause issues.  Things like HomeAssistant are more than capable of handling one update per second.  What kind of interference are you imagining?

My personal expectation wrt to state during transitions is to see that the transition is happening after I run it.