sidoh/esp8266_milight_hub #502: Doestn't connect to 5W RGB WW

Toubster, 2019-08-01:
**First of all, thank you for this great project!**  
I have already tried a few hours to solve the problem myself, but unfortunately that was not possible.  
I have done everything according to instructions and also the Troublesooting Guide, unfortunately it still does not work.  
I tried it with an NodeMCU ESP8266v2 & 3. This I connect to the NRF24L01+.  
Packets are sent and received by the ESP, does anyone have any idea what else it can have?
The desired files are attached.

Thanks in advance.

[MilightHub-Output.zip](https://github.com/sidoh/esp8266_milight_hub/files/3457888/MilightHub-Output.zip)

sidoh, 2019-08-02:
Hi,

Are the ESP8266s able to see packets sent from one another?

Which bulbs are you using?  Can you identify the product number from [this page](https://www.futlight.com/productlist.aspx?typeid=101)?  Which protocol are you trying to use?

Toubster, 2019-08-02:
Hi,

yes, the ESP8266s see the packets from the other one.  

Product: FUT018
5W GU10 RGBW LED Spotlight

Do you mean the UDP Version 5/6? I tried it with both of them over UDP Port 1900.

sidoh, 2019-08-02:
I mean the espMH remote type:

[image]

don't bother with UDP until you have the bulbs responding.  try controlling them directly from the UI.

I assume you've already paired the bulbs with either a remote or espMH?

Toubster, 2019-08-02:
I use the RGBW Remote Type.

Yeah, I tried it also without UDP only in the UI.

Before I had paired the bulbs with a remote and now I would like to pair them with the espMH. And now the bulbs are unpaired from all.

sidoh, 2019-08-02:
Does your remote show up as RGBW when you're sniffing packets from it?

Try using RGB+CCT, FUT018 is compatible with it as well.

If bulbs are not responding, double-check that they're unpaired with the remote.

Toubster, 2019-08-02:
Yes.

Ok thanks, I will test this in the evening.

The bulbs are unpaired because they only light up red and the remote can't change anything.

Toubster, 2019-08-02:
Hi,

i have tested now with both ESPs on the RGB+CCT Type and it doest't work.

sidoh, 2019-08-03:
Does espMH pick up packets from the remote?

Toubster, 2019-08-04:
Yes

sidoh, 2019-08-04:
I don't have a whole lot to go off of here -- gonna need some more details or at least a way to reproduce the issue if you'd like help.

sidoh, 2019-08-04:
Things to try:

1. Enable listening mode while sending the pair command -- make sure packets are getting sent (please paste the output)
2. Paste output while listening to remote

Toubster, 2019-08-04:
Send Packets from the ESPMHv2:  
[image]

Received Packets from the ESPMHv3 on the ESPMH2:  
[image]

Received Packets from the Remote:  
[image]

Do you need more output from listening?

sidoh, 2019-08-06:
That looks like pretty strange behavior from the remote.  Could either be an alternative version or a knockoff.

What buttons were you pressing when the remote packets were intercepted?

What do the packets look like when you run through the pairing procedure with the remote?

Toubster, 2019-08-06:
This Button I pushed.
[image]

These are the packets from the remote at pairing process.
[image]

JHeijting, 2019-08-08:
I have exactly the same problem. The bulb does react to remote type "RGB". It will than turn on and off. But whenever I sniff the original signal (I use the ibox at this momen), than I receive a RGBW packet:
[image]

sidoh, 2019-08-08:
Perhaps try the suggestion I left in this comment:

https://github.com/sidoh/esp8266_milight_hub/issues/498#issuecomment-519580691

The packet from the remote looks weird, but I think that's just because it's sending the last-known values for color, brightness, etc.  If you try changing the color, then pressing the on/off button, does the value for `b1` change?

sidoh, 2019-08-26:
@Toubster, any luck here?

Toubster, 2019-08-26:
Sry for no reaction from my side...

I have tested it and yes you are right, the value of b1 changes if another color is on.

On Packet (White Color):
[image]

Turn the color on blue:
[image]

On Packet while the color was blue:
[image]

But I would try the steps from the issue #498 then I write here again.

Toubster, 2019-09-21:
Hi again,
now I have tested the steps of #498 and with luck(and new Hardware [Kuman nRF24L01+PA+LNA](https://www.amazon.de/gp/product/B06WD17WLS/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1) ) the bulbs are connected to the ESPMH.
But I have another Problem. The connection between the bulbs and the ESPMH is very unstable.
I have tried the new Hardware with other antennas, but nothing has improved.

Do you have any idea about this problem?

sidoh, 2019-11-16:
Sorry for the terribly late response here.  Have had a hard time keeping up with the issue backlog.

Hard for me to say without being able to reproduce it.  If the fundamental problem is instability, the cause is likely either:

* Bad nRF
* Faulty wiring
* Insufficient power

closed 2020-11-01 by sidoh