sidoh/esp8266_milight_hub #503: Color doesn't stick

razvanws, 2019-08-01:
After updating to v1.10.0-rc.1, color doesn't "stick" anymore. Steps to reproduce below:

1. Select an effect (0 let's say)
2. Select white-mode effect
3. Choose an rgb color from hass color picker or esp webpage
4. The light will change to that color and then it will return to white mode 1 second later
I flashed 1.9.2 and this bug doesn't replicate on 1.9.2.

sidoh, 2019-08-01:
Are you using MQTT?  Can you provide some logs?  Easiest way is to attach with `mosquitto_sub`.

razvanws, 2019-08-01:
**v1.9.2** (which is not affected by this bug)

Effect 0
```
milight/0x0/fut089/1 {"state": "ON", "effect": "0"}
milight/updates/0x0/fut089/1 {"state":"ON"}
milight/updates/0x0/fut089/1 {"mode":0}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","mode":0,"bulb_mode":"scene","color":{"r":255,"g":255,"b":255},"effect":"0"}
```

White mode
```
milight/0x0/fut089/1 {"state": "ON", "effect": "white_mode"}
milight/updates/0x0/fut089/1 {"state":"ON"}
milight/updates/0x0/fut089/1 {"command":"set_white"}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","brightness":255,"level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
```

Color change
```
milight/0x0/fut089/1 {"state": "ON", "color": {"r": 255, "g": 0, "b": 0}}
milight/updates/0x0/fut089/1 {"state":"ON"}
milight/updates/0x0/fut089/1 {"hue":0}
milight/updates/0x0/fut089/1 {"saturation":100}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","brightness":255,"level":100,"color":{"r":255,"g":0,"b":0},"bulb_mode":"color"}
```

Color change
```
milight/0x0/fut089/1 {"state": "ON", "color": {"r": 127, "g": 255, "b": 0}}
milight/updates/0x0/fut089/1 {"state":"ON"}
milight/updates/0x0/fut089/1 {"hue":90}
milight/updates/0x0/fut089/1 {"saturation":100}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","brightness":255,"level":100,"color":{"r":127,"g":255,"b":0},"bulb_mode":"color"}
```

**v1.10.0-rc.1**

Effect 0
```
milight/0x0/fut089/1 {"state": "ON", "effect": "0"}
milight/updates/0x0/fut089/1 {"state":"ON"}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","brightness":255,"level":100,"color":{"r":127,"g":255,"b":0},"bulb_mode":"color"}
milight/updates/0x0/fut089/1 {"mode":0}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","mode":0,"bulb_mode":"scene","color":{"r":255,"g":255,"b":255},"effect":"0"}
```

White mode
```
milight/0x0/fut089/1 {"state": "ON", "effect": "white_mode"}
milight/updates/0x0/fut089/1 {"state":"ON"}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","mode":0,"bulb_mode":"scene","color":{"r":255,"g":255,"b":255},"effect":"0"}
milight/updates/0x0/fut089/1 {"command":"set_white"}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","brightness":255,"level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
```

Color change (it jumps to white)
```
milight/0x0/fut089/1 {"state": "ON", "color": {"r": 255, "g": 0, "b": 0}}
milight/updates/0x0/fut089/1 {"state":"ON"}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","brightness":255,"level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
milight/updates/0x0/fut089/1 {"hue":0}
milight/updates/0x0/fut089/1 {"hue":90}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","brightness":255,"level":100,"color":{"r":127,"g":255,"b":0},"bulb_mode":"color"}
milight/updates/0x0/fut089/1 {"saturation":100}
milight/updates/0x0/fut089/1 {"command":"set_white"}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","brightness":255,"level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
```

Color change (it jumps to white again)
```
milight/0x0/fut089/1 {"state": "ON", "color": {"r": 0, "g": 0, "b": 255}}
milight/updates/0x0/fut089/1 {"state":"ON"}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","brightness":255,"level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
milight/updates/0x0/fut089/1 {"hue":240}
milight/updates/0x0/fut089/1 {"hue":90}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","brightness":255,"level":100,"color":{"r":127,"g":255,"b":0},"bulb_mode":"color"}
milight/updates/0x0/fut089/1 {"saturation":100}
milight/updates/0x0/fut089/1 {"command":"set_white"}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","brightness":255,"level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
```

sidoh, 2019-08-01:
Perfect, exactly what I needed.  Thank you!

sidoh, 2019-08-02:
Hmm... I'm not able to reproduce on 1.10.0-rc.1:

```
$ curl -s 10.133.8.158/about | jq -r .version
1.10.0-rc.1
```

MQTT log, sending exactly the same commands:

```
# Clear state
milight/states/0x5/fut089/1 (null)

# Effect
milight/5/fut089/1 {"state": "ON", "effect": "0"}
milight/updates/0x5/fut089/1 {"state":"ON"}
milight/states/0x5/fut089/1 {"state":"ON","color":{"r":255,"g":255,"b":255}}
milight/updates/0x5/fut089/1 {"mode":0}
milight/states/0x5/fut089/1 {"state":"ON","bulb_mode":"scene","color":{"r":255,"g":255,"b":255},"effect":"0"}

# White mode
milight/5/fut089/1 {"state": "ON", "effect": "white_mode"}
milight/updates/0x5/fut089/1 {"state":"ON"}
milight/states/0x5/fut089/1 {"state":"ON","bulb_mode":"scene","color":{"r":255,"g":255,"b":255},"effect":"0"}
milight/updates/0x5/fut089/1 {"command":"set_white"}
milight/states/0x5/fut089/1 {"state":"ON","bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}

# Color
milight/5/fut089/1 {"state": "ON", "color": {"r": 255, "g": 0, "b": 0}}
milight/updates/0x5/fut089/1 {"state":"ON"}
milight/states/0x5/fut089/1 {"state":"ON","bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
milight/updates/0x5/fut089/1 {"hue":0}
milight/updates/0x5/fut089/1 {"saturation":100}
milight/states/0x5/fut089/1 {"state":"ON","bulb_mode":"color","color":{"r":255,"g":0,"b":0}}

# Color change
milight/5/fut089/1 {"state": "ON", "color": {"r": 0, "g": 0, "b": 255}}
milight/updates/0x5/fut089/1 {"state":"ON"}
milight/states/0x5/fut089/1 {"state":"ON","bulb_mode":"color","color":{"r":255,"g":0,"b":0}}
milight/updates/0x5/fut089/1 {"hue":240}
milight/updates/0x5/fut089/1 {"saturation":100}
milight/states/0x5/fut089/1 {"state":"ON","bulb_mode":"color","color":{"r":0,"g":0,"b":255}}
```

In every case, the packets I expect to be sent are sent, and the resulting state at the end is as expected.

Do you have some automation set up that is reading and relaying messages from MQTT, or perhaps multiple hub instances listening to the same topics?

If you can get some kind of shell script together that reliably reproduces the problem, that might be helpful.  Maybe it's a timing thing?  (don't see why this would matter, though)

razvanws, 2019-08-02:
Thanks for looking into it.
I'm not using any kind of automation (it's a new install) and I only have one hub. Please see below the mosquitto_pub commands I used to reproduce the issue. 
Unfortunately it reproduces each time in my case.

`mosquitto_pub -h 10.10.111.4 -u mosquitto -P xxxxxx -v -t 'milight/0x0/fut089/1' -m '{"state": "ON", "effect": "1"}'
`
`mosquitto_pub -h 10.10.111.4 -u mosquitto -P xxxxxx -t 'milight/0x0/fut089/1' -m '{"state": "ON", "effect": "white_mode"}'
`
`mosquitto_pub -h 10.10.111.4 -u mosquitto -P xxxxxx -t 'milight/0x0/fut089/1' -m '{"state": "ON", "color": {"r": 0, "g": 0, "b": 255}}'
`

Maybe I need to somehow "factory reset" the hub? 
Is such a thing even exists?

sidoh, 2019-08-02:
I'm running the script and still seeing results that I expect:

```
milight/0x0/fut089/1 {"state": "ON", "effect": "1"}
milight/0x0/fut089/1 {"state": "ON", "effect": "white_mode"}
milight/0x0/fut089/1 {"state": "ON", "color": {"r": 0, "g": 0, "b": 255}}
milight/updates/0x0/fut089/1 {"state":"ON"}
milight/states/0x0/fut089/1 {"state":"ON","bulb_mode":"color","color":{"r":0,"g":0,"b":255}}
milight/updates/0x0/fut089/1 {"mode":1}
milight/updates/0x0/fut089/1 {"state":"ON"}
milight/updates/0x0/fut089/1 {"command":"set_white"}
milight/updates/0x0/fut089/1 {"state":"ON"}
milight/updates/0x0/fut089/1 {"hue":240}
milight/states/0x0/fut089/1 {"state":"ON","bulb_mode":"color","color":{"r":0,"g":0,"b":255}}
milight/updates/0x0/fut089/1 {"saturation":100}
milight/states/0x0/fut089/1 {"state":"ON","bulb_mode":"color","color":{"r":0,"g":0,"b":255}}
```

Do the logs above look wrong to you?

razvanws, 2019-08-02:
It's looking perfectly good on your end. That's weird...

sidoh, 2019-08-02:
Weird indeed.  Definitely need to be able to reproduce in order to debug.

Seeing this:

```
milight/updates/0x0/fut089/1 {"command":"set_white"}
```

means that either something instructed the espMH to send the set white command, or it intercepted a set white command from something else.  espMH processes commands it receives in order, and packets that result from the command are also processed in order.

can i see your `/settings`?  make sure to sensor passwords.

next step is probably serial logs.  if nothing interesting appears, might need to compile with debug flags.

PetricaM, 2019-08-02:
Hi,

Tried to reproduce the issue, however it doesn't happen for my 1.10.0 rc1 install (and I had some weird stuff going on in the past :)) ) 

Some ideas: do you happen to use retain flag? Is the gateway online the whole time? Do you have a physical remote to compare the result against?

You might try with another NRF24L01 or use the tips in the wiki to  stabilize the board (use a good power supply, add a capacitor, etc). I had previous problems with NRF24L01 boards that caused the gateway to reboot each time a package that used saturation was sent. When the gateway rebooted, the last payload was applied (as it was sent with retain flag).

razvanws, 2019-08-02:
@sidoh I'll show you the /settings later today

@PetricaM I'm using the retain flag. The gateway is online the whole time. I also tried to unplug/plug it and the issue persists. As far as I can see, the gateway is not rebooting because if I keep a ping open to it's IP address, no packet is dropped at any time.
However, I think it's a good idea to try a different power supply. Currently I'm using a Samsung 2A microusb charger. I'll try to use a different one and also try to power the board directly from the usb port of the macbook pro.

I'll get back to you with the results later today.

razvanws, 2019-08-02:
So, I tried again with a different power supply, including power directly from the laptop's usb and and it replicates each time. Here there are my /settings:

```
{
	"admin_username": "razvan",
	"admin_password": "xxxxxxxxx",
	"ce_pin": 4,
	"csn_pin": 15,
	"reset_pin": 0,
	"led_pin": -2,
	"radio_interface_type": "nRF24",
	"packet_repeats": 50,
	"http_repeat_factor": 1,
	"auto_restart_period": 0,
	"mqtt_server": "10.10.111.4:1883",
	"mqtt_username": "mosquitto",
	"mqtt_password": "xxxxxxxx",
	"mqtt_topic_pattern": "milight/:device_id/:device_type/:group_id",
	"mqtt_update_topic_pattern": "milight/updates/:device_id/:device_type/:group_id",
	"mqtt_state_topic_pattern": "milight/states/:device_id/:device_type/:group_id",
	"mqtt_client_status_topic": "milight/client_status",
	"simple_mqtt_client_status": true,
	"discovery_port": 48899,
	"listen_repeats": 3,
	"state_flush_interval": 10000,
	"mqtt_state_rate_limit": 500,
	"packet_repeat_throttle_sensitivity": 0,
	"packet_repeat_throttle_threshold": 200,
	"packet_repeat_minimum": 3,
	"enable_automatic_mode_switching": true,
	"led_mode_wifi_config": "Fast toggle",
	"led_mode_wifi_failed": "On",
	"led_mode_operating": "Slow blip",
	"led_mode_packet": "Flicker",
	"led_mode_packet_count": 3,
	"hostname": "milight-hub",
	"rf24_power_level": "MAX",
	"rf24_listen_channel": "LOW",
	"wifi_static_ip": "10.10.111.6",
	"wifi_static_ip_gateway": "10.10.111.1",
	"wifi_static_ip_netmask": "255.255.255.0",
	"packet_repeats_per_loop": 10,
	"home_assistant_discovery_prefix": "",
	"wifi_mode": "n",
	"rf24_channels": ["LOW", "MID", "HIGH"],
	"device_ids": [],
	"gateway_configs": [],
	"group_state_fields": ["state", "status", "brightness", "level", "color", "mode", "color_temp", "bulb_mode", "computed_color", "effect"],
	"group_id_aliases": {}
}

```

Here it is the result of a color change:
```
milight/0x0/fut089/1 {"state": "ON", "color": {"r": 255, "g": 0, "b": 0}}
milight/updates/0x0/fut089/1 {"state":"ON"}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","brightness":255,"level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
milight/updates/0x0/fut089/1 {"hue":0}
milight/updates/0x0/fut089/1 {"hue":104}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","brightness":255,"level":100,"color":{"r":68,"g":255,"b":0},"bulb_mode":"color"}
milight/updates/0x0/fut089/1 {"saturation":100}
milight/updates/0x0/fut089/1 {"command":"set_white"}
milight/states/0x0/fut089/1 {"state":"ON","status":"ON","brightness":255,"level":100,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
```

If you want to send me a "special" version with some debug code inside, I'll be happy to test and report back.

sidoh, 2019-08-02:
Could you check serial logs as well?  Just use `pio device monitor` while the ESP8266 is plugged into your computer.

razvanws, 2019-08-02:
This is the output while changing colors:

```
pio device monitor

--- Available ports:
---  1: /dev/cu.Bluetooth-Incoming-Port 'n/a'
---  2: /dev/cu.BoseQuietComfort35-SPPD-2 'n/a'
---  3: /dev/cu.BoseQuietComfort35-SPPD-4 'n/a'
---  4: /dev/cu.usbserial-1410 u'USB2.0-Serial'
---  5: /dev/cu.wchusbserial1410 u'USB2.0-Serial'
--- Enter port index or full name: /dev/cu.wchusbserial1410
--- Miniterm on /dev/cu.wchusbserial1410  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�␌�xWA␌���$��␟��ERROR: Failed to connect to MQTT server

*WM: Adding parameter
*WM: staticIP
*WM: Adding parameter
*WM: netmask
*WM: Adding parameter
*WM: gateway
We have a static IP: 10.10.111.6
*WM:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 10.10.111.6
*WM: Using last saved values, should be faster
*WM: Connection result:
*WM: 3
*WM: IP Address:
*WM: 10.10.111.6
Wifi connected succesfully

Setup complete (version 1.10.0-rc.1)

```

sidoh, 2019-08-02:
Is this the output when you run the script, or just after a boot?

razvanws, 2019-08-02:
When I run the script. It seems that nothing changes while I run the commands.

sidoh, 2019-08-02:
Gotcha, at least means no crashing is happening.

To enable debug logging, just uncomment the lines here:

https://github.com/sidoh/esp8266_milight_hub/blob/v1.10.0-wip/platformio.ini#L41

razvanws, 2019-08-02:
I'll do that and report back.

sidoh, 2019-08-02:
(in case it's helpful - you can compile and flash an image using `pio run -e d1_mini --target upload`)

razvanws, 2019-08-02:
Do you have any idea how to save the output of device monitor to a file? I cannot scroll up o down and I cannot copy paste the results...

sidoh, 2019-08-02:
`tee` them to a file, maybe?

```
pio device monitor | tee serial_output.txt
```

razvanws, 2019-08-02:
Whenever I try to redirect to a file with pio, I get:
`UnicodeEncodeError: 'ascii' codec can't encode character u'\ufffd' in position 0: ordinal not in range(128)
`

razvanws, 2019-08-02:
I managed to do it. Please see the pastebin below:
https://pastebin.com/Pka0113F

sidoh, 2019-08-02:
Thanks!  Mind doing the same thing without STATE_DEBUG?  It's really noisy and I think will be unhelpful here.

razvanws, 2019-08-02:
Please note that last time I tried to change it to a color:
```
Patching existing state with: {"hue":240}
GroupState::patch: State changed: {"bulb_mode":"color","hue":240}
```

Then, the next logs are from when it jumped back to white on it's own. The lines near the end of the file represent an action which was not trigger by me:
```
Patching existing state with: {"command":"set_white"}
GroupState::patch: State changed: {"bulb_mode":"white","effect":"white_mode"}
Raw data: 00000000 30000800
Patching existing state with: : {"bulb_mode":"white","effect":"white_mode"}
Raw data: 00000000 30000800
Patching existing state with: : {"bulb_mode":"white","effect":"white_mode"}
Raw data: 00000000 30000800

```

razvanws, 2019-08-02:
Sure, let me take a new session of logs.

razvanws, 2019-08-02:
Here it is the new pastebin without STATE_DEBUG:
https://pastebin.com/jQJWag2i

sidoh, 2019-08-03:
@razvanws, are you using the retain flag on commands sent?  This will likely produce unexpected results, although I don't think I see evidence suggesting that's your problem.  Retain makes sense for state -- it's the last known state and is persistent.  Commands are incomplete and should be ephemeral.

Some of the prints are broken for some reason.  Can you try with this image?

[esp8266_milight_hub_1.10.0-rc.1.debug.bin.gz](https://github.com/sidoh/esp8266_milight_hub/files/3464098/esp8266_milight_hub_1.10.0-rc.1.debug.bin.gz)

Here's the diff if you want to compile yourself:

```
diff --git a/lib/MQTT/MqttClient.cpp b/lib/MQTT/MqttClient.cpp
index 2c2e686..c7265b9 100644
--- a/lib/MQTT/MqttClient.cpp
+++ b/lib/MQTT/MqttClient.cpp
@@ -213,7 +213,8 @@ void MqttClient::publishCallback(char* topic, byte* payload, int length) {
   memcpy(cstrPayload, payload, sizeof(byte)*length);
 
 #ifdef MQTT_DEBUG
-  printf("MqttClient - Got message on topic: %s\n%s\n", topic, cstrPayload);
+  Serial.printf("MqttClient - Got message on topic: %s\n", topic);
+  Serial.println(cstrPayload);
 #endif
 
   char topicPattern[settings.mqttTopicPattern.length()];
diff --git a/lib/MiLight/MiLightClient.cpp b/lib/MiLight/MiLightClient.cpp
index 7942bff..e8ba197 100644
--- a/lib/MiLight/MiLightClient.cpp
+++ b/lib/MiLight/MiLightClient.cpp
@@ -301,6 +301,9 @@ void MiLightClient::update(JsonObject request) {
     this->updateBeginHandler();
   }
 
+  Serial.print(F("MiLightClient - Handling command: "));
+  serializeJson(request, Serial);
+
   const JsonVariant status = this->extractStatus(request);
   const uint8_t parsedStatus = this->parseStatus(status);
   const JsonVariant jsonTransition = request[RequestKeys::TRANSITION];
diff --git a/platformio.ini b/platformio.ini
index 15db945..810d36e 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -38,11 +38,11 @@ build_flags =
   -D RICH_HTTP_REQUEST_BUFFER_SIZE=2048
   -D RICH_HTTP_RESPONSE_BUFFER_SIZE=2048
   -Idist -Ilib/DataStructures
-# -D STATE_DEBUG
-# -D DEBUG_PRINTF
-# -D MQTT_DEBUG
+  -D STATE_DEBUG
+  -D DEBUG_PRINTF
+  -D MQTT_DEBUG
+  -D DEBUG_CLIENT_COMMANDS
 # -D MILIGHT_UDP_DEBUG
-# -D STATE_DEBUG
 
 [env:nodemcuv2]
 platform = ${common.platform}

```

zyrorl, 2019-08-06:
this is definitely a bug. I can replicate it every time.

Steps to replicate:

Set light to white using: 
```{"command": "set_white"}```

Set light to red using:
```{"color":{"r": 255, "g":0, "b":0}```

Result: Colour flicks to red, then back to white.

This appears to work however:

```{"hue": 0}``` - sets to red, and stays red, doesn't jump to white.

However ```{"hue": 0, "saturation": 100}``` jumps to red, then white.

It appears there's some buggy logic somewhere.

Hope this helps !

razvanws, 2019-08-06:
@sidoh Indeed I'm using the retain flag but as you said, I don't think it's related to the behaviour. For the time being, I'm unable to further debug but as soon as I have the board again, I'll try the suggested diff.

PetricaM, 2019-08-06:
> However `{"hue": 0, "saturation": 100}` jumps to red, then white. If you send them as separate commands (hue, then saturation), it works correctly, but this is a crappy solution that takes longer to apply settings with.

Do you have "Enable automatic mode switching" set to enable in Settings->Radio?

sidoh, 2019-08-06:
@PetricaM, good call.  I have forgotten about this setting a few times because I dislike the behavior and have never used it myself.  It _should_ be off by default.

@razvanws and @zyrorl, please check the setting Petrica recommended.

sidoh, 2019-08-06:
I can replicate this with automatic mode switching turned on.  If you don't need it, turn it off and it'll hopefully fix.

That feature is confusing, but still shouldn't behave that way.  The intention is to switch back into your current mode if you issue a command that switches you to another mode.  So sending a `color` command while you're in white mode should not trigger it.  I'll look into it.

sidoh, 2019-08-06:
Ah, I know what the problem is.  And honestly it's probably not the right thing to fix it for the sake of this feature.  

Summary:

* Previous to 1.10, packets were processed synchronously, which meant that packet formatters could assume that state for a bulb was up to date.  
* In 1.10, packets are enqueued and processed asynchronously.  So a packet which was enqueued before the current packet will not have affected internal state unless it's already been processed.  
* The `color` command enqueues two packets (one for hue, one for saturation) immediately, meaning when the saturation packet is processed, it won't see that the bulb will have switched to color mode by the time it's processed (it just sees the bulb's current state -- white mode).  So if color switching is enabled, it'll get triggered.

The solutions I see are:

1. Make the packet queue packet generator callbacks instead of raw packets.  This avoids the weird semantics around state, but introduces additional complexity and overhead.
2. Just write off the color switching feature as incompatible with the `color` command.
3. Temporarily override color switching when processing `color` commands.  This is a case-specific workaround that avoids the new state model.  Honestly feels pretty gross, but will probably work just fine.

If lots of people use the color switching feature, i can look into (1) or (3).  Otherwise, I'm inclined to choose (2).

razvanws, 2019-08-06:
Nice find! I disabled Automatic mode switching and it works as expected.

sidoh, 2019-08-11:
Gonna go with (2).  I'm also updating the label and help message for this setting to make it sound less inviting.  I do not expect this is a setting most people want to have enabled.

```javascript
{
    tag:   "enable_automatic_mode_switching",
    friendly: "Switch to previous mode after saturation/color commands",
    help: "For RGBWW bulbs (using RGB+CCT or FUT089), commands that adjust color saturation or white temperature "
      + "will switch into the appropriate mode, make the change, and switch back to previous mode.  WARNING: this "
      + "feature is not compatible with \"color\" commands.",
}
```

[image]

Thanks for reporting, @razvanws!

closed 2019-08-11 by sidoh