sidoh/esp8266_milight_hub #506: MQTT no LWT

DasQwarx, 2019-08-09:
Hi,
i use 1.10.0-rc.3 (nodemcuv2) on FHEM own MQTT-server.
my problem is, at 
Settings --> MQTT --> MQTT Client Status Topic
you write from LWT, but there is no LWT only i found a "status" like connenced or disconnected. 
is it possible to change this from "status" to "LWT"

sidoh, 2019-08-09:
Hello,

MQTT Client Status Topic _is_ LWT.  It's using it under the hood, just does a little bit more to guarantee that any subscriber to this topic sees:

* `connected` if the hub is actively connected to the MQTT server
* `disconnected` if the hub either cleanly or uncleanly disconnected from MQTT

This is in "simple" mode.  with "detailed" mode, a JSON object is published containing a bunch of additional information.

[image]

What are you trying to accomplish that this feature doesn't solve?

sidoh, 2019-08-09:
For reference, the behavior is almost precisely what is [outlined here under "best practices."](https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament/)

DasQwarx, 2019-08-09:
Sorry for my Bad written english, but i think we Talk from 2 different things. 
In json u said „Status“ => LWTmessage
But i Need 
„LWT => LWTmessage

sidoh, 2019-08-09:
Hard to tell from their docs, Tasmota seems to do roughly the same thing I'm doing in "Simple" mode.  Publish either "online" or "offline" to a single topic:

https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant-%E2%80%90-Manual-Config

Does "Simple" mode not do what you want?

If you can point me at a standard I should be following, I'm happy to take a look.

DasQwarx, 2019-08-09:
The payload Is correct (connected or disconnected)
But the topic „Status“ Is wrong, it should be „LWT“

I make some Screenshots from my other devices in FEHM, where Is the right topic.

sidoh, 2019-08-09:
"status" makes a lot more sense to me as a field name.  It's not the LWT.  LWT is the MQTT feature that allows a message to be sent on the client's behalf in the case that it disconnects uncleanly.  Something consuming this doesn't care about LWT, it cares whether or not the client is connected, right?

DasQwarx, 2019-08-09:
but a lot devices did it with the topic LWT.
-my 3D Printer (octoprint)
-Tasmota
-ESPeasy
-OwnTracks
....

and other devices use status for real status e.g. on or off

but i have made an readings mapping as workaround for me and its works fine. (quickanddirty)

closed 2019-08-10 by DasQwarx

sidoh, 2019-08-10:
Glad you were able to find a workaround.

Definitely happy to consider a change.  Hope I wasn't coming across as combative -- just trying to find the right thing =)

odis68, 2022-04-27:
https://github.com/sidoh/esp8266_milight_hub/issues/506#issuecomment-520016516
Right.
But this status is needed when the hub hangs!
I join the guys, I really want to realize this status. Today it is important because. it can be seen in a smart home that the hub is frozen.