sidoh/esp8266_milight_hub #507: HASS and updates group 0

poudenes, 2019-08-10:
Hi,

When you use the milight plugin for DOmoticz, and I turn on group 1,2,3 and 4, also group 0 will change status. When turn off group 0 all other groups go off as well.

Now im working to change to HASS. I have add all bulbs, also group 0
When I do the same, the groups 1,2,3 and 4 stay with their status in HASS

There is no MQTT update topic pattern option in HAS

is there a way to fix this with the options in HASS?

sidoh, 2019-08-11:
Hiya @poudenes.

You're right.  When a command for Group 0 is sent, states for individual bulbs are _not_ sent out (although state _is_ updated internally).  I made this decision intentionally because it seemed like an up to 8x increase in response traffic could cause performance issues.

My guess is that Domoticz was handling this intelligently because it had a plugin that knew that Group 0 was the "all" group with milight.  The HASS integration everyone uses is just the generic JSON mqtt light, which does not understand that Group 0 means anything special.

As a workaround, you can probably adapt this example to fan out state updates from Group 0 to the constituent groups:

https://github.com/sidoh/esp8266_milight_hub/wiki/Using-Milight-Remote-with-HomeAssistant#configure-hass

I think it'd look something like this (I haven't tested it):

```yaml
automation:
- alias: MiLight Forwarder
  initial_state: True
  trigger:
    platform: mqtt
    topic: milight/state/0x1111/rgb_cct/0
  action:
    - service: mqtt.publish
      data_template:
        topic: "milight/state/0x1111/rgb_cct/1"
        payload_template: >
          {{ trigger.payload }}
    - service: mqtt.publish
      data_template:
        topic: "milight/state/0x1111/rgb_cct/2"
        payload_template: >
          {{ trigger.payload }}
# ... Same thing for groups 3 and 4 ...
```

poudenes, 2019-08-11:
Hi,

Thanks for the nice explanation.
I will create the automation and see how it work.

I'll let you know!!

closed 2019-08-11 by poudenes

poudenes, 2020-05-07:
Hi @sidoh,

With the corona lockdown in Netherlands I was investigate this more. I had some funny issues with bulbs last week. When I turn off group 0, then internal all groups 1,2,3 and 4 get their state but don't send them out (prevent lot of traffic over mqtt)

Funny thing that will happen (and we had this funny problem)
When we turn off all the lights in the house I send a turn off command to all groups 0.
All bulbs go off. The patch workaround you post will help to turn off the bulbs by groups in HA to turn them off also. But they not sending mqtt out to Hub.

What will happen then, Group 0 is off. In MQTT some groups retained in MQTT database as "on"
Now the funny thing pop up:

When the Milight Hub reset or HA reset because of a issue it reads the retained MQTT status, that means for groups other then 0 sometimes a "on" status. And then the lights turn on.

Also in our bedroom at 2AM haha... 

Now im searching for a solution how to trigger this. Because would be nice to have group 0, 1, 2, 3 and 4 the correct status in MQTT database all the time...

Cheers
Peter

p.s. hope you understand what I mean haha

poudenes, 2020-05-11:
Created a very dirty way to turn all lights (every Bulb) to off at the night
After a scene has stopped it will turn every individual bulb to OFF Status.

Better then risk what I wrote above .. haha
```
    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAF/rgb_cct/1"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAF/rgb_cct/2"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAF/rgb_cct/3"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAA1/rgb_cct/1"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAA1/rgb_cct/2"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAA1/rgb_cct/3"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAA2/rgb_cct/1"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAA2/rgb_cct/2"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAA2/rgb_cct/3"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAA/rgb_cct/1"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAA/rgb_cct/2"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAA/rgb_cct/4"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAA/rgb_cct/3"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAB/rgb_cct/1"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAB/rgb_cct/2"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAC/rgb_cct/1"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAC/rgb_cct/2"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAC/rgb_cct/3"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAC/rgb_cct/4"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAD/rgb_cct/1"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAD/rgb_cct/2"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAD/rgb_cct/3"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAD/rgb_cct/4"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAE/rgb_cct/1"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAE/rgb_cct/2"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAAAE/rgb_cct/3"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAABB/rgb_cct/1"
        payload: "{'state': 'OFF'}"

    - service: mqtt.publish
      data_template:
        topic: "milight/0xAABB/rgb_cct/2"
        payload: "{'state': 'OFF'}"

```