sidoh/esp8266_milight_hub #508: RGBW lights in Home assistant don't have effects available

strumf666, 2019-08-11:
Effects available are white_mode and night_mode, but not 0-8. In milight hub all efects are available.

sidoh, 2019-08-11:
Hi @strumf666, how did you set up espMH with HomeAssistant?

strumf666, 2019-08-11:
I use mqtt with autodiscovery, no manual light configuration in configuration.yaml.

poudenes, 2019-08-11:
I have this in my config:

- platform: mqtt
  name: "Dressoir All"
  schema: json
  command_topic: "milight/0xAAAF/rgb_cct/0"
  state_topic: "milight/states/0xAAAF/rgb_cct/0"
  qos: 1
  optimistic: false
  retain: true
  brightness: true
  rgb: true
  color_temp: true
  effect: true
  effect_list: [zero, one, two, tree, four, five, six, seven, eight, white_mode, night_mode]

have all effects, include the white_mode and night_mode

sidoh, 2019-08-11:
Ah yeah, the numbered effects aren't supplied in the discoverable config.  I'll add them.

FUT020, FUT096 (RGB), and FUT007 (CCT) don't support these, though, so they won't be advertised for those remote types.

sidoh, 2019-08-11:
Got it.  Yeah -- currently the numbered effects are not published.  I'll add them.  Will be in the next release.

closed 2019-08-30 by sidoh