sidoh/esp8266_milight_hub #509: transitioning 0 to X

Heimweh76, 2019-08-15:
I'm trying to drive the lamp up and down with the function "transitioning". I have tried it with the following MQTT command:
"status":"on","transition":"20","brightness":"200" (excerpt from fhem)

the lamp goes on and off but without the 20 seconds ramp. What am I doing wrong?

sidoh, 2019-08-26:
Try `{"transition":20}` without the quotes.

sidoh, 2019-08-29:
let me know if that doesn't work.

closed 2019-08-29 by sidoh

Heimweh76, 2019-08-29:
Hello Chris,

i didn't have much time to try the command, but i tried it quickly and
couldn't get it to work.
I switched off the lamp with the remote control and then tried to dim up
with the following command:

 {"status":"on","transition":20}

and i tried with a additional brightness value:

 {"status":"on","transition":20,"brightness":"200"}

But the bulp turns quickly to on.

I can't rule out making a mistake. I will try it in the next days if the
weather is worse again :)

Greetings to SF, Daniel

sidoh, 2019-08-29:
Are you using one of the 1.10 versions?

Heimweh76, 2019-08-29:
I use the 1.9.2 (nodemcuv2) from 2019-06-14

sidoh, 2019-08-29:
Transitions were added in 1.10.

Heimweh76, 2019-08-29:
oooooooh - I'm sorry for wasting your time :( I will make the update as
soon as possible and give you feedback

Heimweh76, 2019-08-30:
Hello Chris,

I did the update yesterday and tried again. It's working now. the dimming
is a bit jerky - you can clearly see the dimming steps. So no soft dimming.
Is this due to my light source or maybe the hardware (bridge)?

Greetz from Germany, Daniel