sidoh/esp8266_milight_hub #512: Transition

poudenes, 2019-08-31:
### Describe the bug

When you use transition on multiple group lights (livingroom, bedroom, kitchen) the transition is doing ok. But when the group_all turns off sometimes it stay on in HA, sometimes it goes off.

### Steps to reproduce

I used this test YAML to do some tests before I implement the transition into my scenes:

`- alias: TEST BUTTON ON
  trigger:
    - entity_id: switch.newkaku_014e5906_a
      platform: state
      to: 'on'

  action:
    - service: light.turn_on
      data:
        entity_id:
          - light.dressoir_all
          - light.livingroom_all
          - light.kitchen_all
          - light.bedroom_all
        brightness_pct: 100

- alias: TEST BUTTON OFF
  trigger:
    - entity_id: switch.newkaku_014e5906_a
      platform: state
      to: 'off'

  action:
    - service: light.turn_off
      data:
        entity_id:
          - light.dressoir_all
          - light.livingroom_all
          - light.kitchen_all
          - light.bedroom_all
        transition: 10`

### Expected behavior

I was expecting that the async will help to handle all the commands over MQTT correct.
No blocking issues when sending lot of commands.

### Setup information
RPi 3 B+ with Debian Stretch, Home Assistant (0.97.2) running in docker config
System runs on a SSD with a Wifi 5Ghz connection

#### Firmware version
Version
1.10.0
Release Date
2019-08-30T16:47:40Z

#### Output of http://milight-hub.local/about

```{"firmware":"milight-hub","version":"1.10.0","ip_address":"192.168.100.70","reset_reason":"Software/System restart","variant":"nodemcuv2","free_heap":11712,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":116}}
```

#### Output of http://milight-hub.local/settings
```
{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":25,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.100.24:1883","mqtt_username":"poudenes","mqtt_password":"XXXXXXXXXX","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:hex_device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/states/:hex_device_id/:device_type/:group_id","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":0,"state_flush_interval":0,"mqtt_state_rate_limit":50,"packet_repeat_throttle_sensitivity":10,"packet_repeat_throttle_threshold":1000,"packet_repeat_minimum":10,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow toggle","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"MilightHub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"192.68.100.70","wifi_static_ip_gateway":"192.168.100.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","rf24_channels":["LOW","MID","HIGH"],"device_ids":[43681,43682,43683,43684,43690,43691,43692,43693,43694,43695,43706,43707],"gateway_configs":[[43690,5980,6],[43691,5981,6],[43692,5982,6],[43693,5983,6],[43694,5984,6],[43695,5985,6],[43681,5986,6],[43682,5987,6],[43683,5988,6],[43706,5989,6],[43684,5990,6],[43707,5991,6]],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type"],"group_id_aliases":{"Balcony":["rgb_cct",43684,0],"Bathroom":["rgb_cct",43693,0],"Bedroom":["rgb_cct",43690,0],"Bedroom2":["rgb_cct",43691,0],"BedroomAll":["rgb_cct",43706,0],"Dressoir":["rgb_cct",43695,0],"Hallway":["rgb_cct",43683,0],"Kitchen":["rgb_cct",43682,0],"Livingroom":["rgb_cct",43692,0],"Sink":["rgb_cct",43681,0],"Toilet":["rgb_cct",43694,0],"WallWindow":["rgb_cct",43707,0]}}
```

### Additional context
Maybe I have to delete the gateway configs ? that was used by Domoticz in past and not used anymore in HA I guess?

sidoh, 2019-08-31:
Async helps with throughput, but there's still a limit.  Your `/about` output suggests you're dropping packets:

```json
"queue_stats":{"length":0,"dropped_packets":116}
```

I understand roughly what you're trying to do, could you do me a favor and translate this into a script that I can run to reproduce?  Easiest thing would be a bunch of mosquitto_pub commands.

The suggestions on the [Performance Tuning](https://github.com/sidoh/esp8266_milight_hub/wiki/Performance-Tuning) wiki page still apply.  I'm not entirely convinced this is a bug, for what it's worth. :)

poudenes, 2019-08-31:
I will do some test later de week. I had a crash on the Hub, plug out the power and put it back... was not my intention. And after reboot the whole config was gone.... So had to do a restore... but at same time I must leave the house to my work... haha :) had no time to test more.

So I will come back to this coming week :)

poudenes, 2019-09-04:
Did some tests. Everything is working. The problem is that I turn on 4 groups (dressoir, living room, kitchen and bedroom) total 18 bulbs. When turn on group_0 then the patch to update group 1,2,3,4 also activates. 

What I think is when I do a transition on group_0 then there will be so many state updates to group 1,2,3,4. This give a big delay. At the end I see all bulbs get the right status in HA. But it takes some time.

Its not a problem, I love to test the hub maximum to see what it can handle :)

closed 2019-09-04 by poudenes

sidoh, 2019-09-04:
Totally fair to push the boundaries.  I can only promise so much, though :)

There are a couple of additional settings I can expose that will help with performance tuning.  Namely, default transition period (it's currently really low -- optimized for smooth transitions rather than throughput).  But might not get around to that for a while.