sidoh/esp8266_milight_hub #513: Longer transitions support

badstraw360, 2019-08-31:
Is there any chance of making longer transitions, for example 10 minutes, instead of 1 minute only? (it doesn't work for more than 60 seconds) 

That would be great to make a sunset / dawn effect.

poudenes, 2019-08-31:
You can also do this in your script. I had it in past with Domoticz dZvents

sidoh, 2019-08-31:
There is not a limit on the length of transitions.  Duration is measured in seconds, and you should be able to pass any value (within reason).  Under the hood, this is converted to milliseconds.  The datatype being used is `size_t`, which is a 32-bit unsigned integer.  So the maximum transition time should be on the order of a month.  

What are you trying that isn't working?

badstraw360, 2019-09-01:
Im using the service dev tool in HASS:

```
{
  "entity_id": "light.room",
  "transition": 600,
  "brightness_pct": 100
}

```

Instead of fading in 10 minutes, the transition lasts 1 minute. So If it's longer than 60 seconds it always transitions in 1 minute, but if it's less then it works as expected.

sidoh, 2019-09-01:
Ah, you're definitely right.  Something is getting screwed up with the period calculation.

sidoh, 2019-09-01:
Yeah, I see what's going on.  It's parameterizing on period, which is never specified when using homeassistant-controlled transitions.  So the default of 225ms is used, meaning the longest possible duration without manually specifying a period is roughly 100x that, or 22.5s.

I'll move things around to force a given duration.

sidoh, 2019-09-01:
Hoping this commit does the trick: https://github.com/sidoh/esp8266_milight_hub/commit/8529d7ac96cf81052767b5d407c7f5dcba70998c

sidoh, 2019-09-01:
Should be fixed in 1.10.1.  Thanks for reporting.

closed 2019-09-01 by sidoh