sidoh/esp8266_milight_hub #514: How does the packet queue handel repeats?

starkillerOG, 2019-08-31:
I saw a new stable version (1.10.0) just came out which implements a packet queue.
In the code i see that the max queue length is 20:
 `#define MILIGHT_MAX_QUEUED_PACKETS 20`

I have my packet repeats set to 150, now if i issue a single turn on command, will this count as 1 in the queue or as 150?
In other words will 130 of my 150 repeats be dropped because the queue is full after 20 packets?

sidoh, 2019-08-31:
No.  Enqueued packets are not expanded in the queue.  The meaning there is a maximum of 20 enqueued command packets, before considering repeats.  There's a setting that controls the number of repeats handled in a single pass (see the README for details).

Reference code:

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/PacketSender.cpp#L33-L43

closed 2019-08-31 by sidoh

starkillerOG, 2019-08-31:
Thank you for the clear explanation and quick replay!
That is indeed the behaviour that you would expact and what I would want, perfect.

Keep up the amazing work!