sidoh/esp8266_milight_hub #515: Lost settings after update

poudenes, 2019-09-02:
### Describe the bug

After latest update Current Version 1.10.1 (nodemcuv2) it lost his settings.

### Steps to reproduce

Update from 1.10.0 to 1.10.1 gives me the problem

### Expected behavior

not loosing the settings

After upload backup from 1.10.0 there is nothing happening, but after upload backup from 1.9.2 I had my config back. Maybe something to do with the device label field?

sidoh, 2019-09-03:
I can't reproduce.  Are you sure you uploaded the right binary?  Uploading the wrong one can corrupt SPIFFS.

poudenes, 2019-09-04:
I did some checks. And the hostname was like @#$}{":><%$# instead of MilightHub.
So download the firmware again and update again. Now hostname is correct (checked on TimeCaspule) maybe this was the problem also losing the settings.

sidoh, 2019-09-04:
This is still pretty strange -- if anything, I'd expect it to reset to defaults and maybe having screwy behavior when saving.

This does sound like the mostly likely cause, though.  If you notice it again, let me know.

closed 2019-09-04 by sidoh