sidoh/esp8266_milight_hub #516: problem with transitions and scenes with multiple bulbs

badstraw360, 2019-09-03:
When I try to run an automation that includes multiple bulbs, some of them won't transition to the next state or they transition partially.

### Steps to reproduce

```
automation:
  - alias: "SONOS play"
      trigger:
        - platform: state
          entity_id: media_player.sonos
          from: 'idle'
          to: 'playing'
      action:
        - service: scene.turn_on
          entity_id: scene.normal
        - delay: '00:00:02'   
        - service: scene.turn_on
          entity_id: scene.party
```

```
scenes:
  - name: party
    entities:
      light.kitchen:
        state: on
        brightness_pct: 50
        transition: 20
      light.hall:
        state: on
        brightness_pct: 50
        transition: 20
      light.room:
        state: on
        brightness_pct: 50
        transition: 20

  - name: normal
    entities:
      light.kitchen:
        state: on
        brightness_pct: 100
      light.hall:
        state: on
        brightness_pct: 100
      light.room:
        state: on
        brightness_pct: 100
  
```

mqtt output:

```

lights/commands/0x5555/rgb_cct/2 {"state": "ON", "brightness": 255}
lights/commands/0x1111/rgb_cct/4 {"state": "ON", "brightness": 255}
lights/commands/0x0/rgb_cct/2 {"state": "ON", "brightness": 255}
lights/states/0x5555/rgb_cct/2 {"state":"ON","status":"ON","brightness":255,"level":100,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x5555/rgb_cct/2 {"state":"ON","status":"ON","brightness":255,"level":100,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x0/rgb_cct/2 {"state":"ON","status":"ON","brightness":255,"level":100,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/commands/0x5555/rgb_cct/2 {"state": "ON", "transition": 20, "brightness": 127}
lights/commands/0x1111/rgb_cct/4 {"state": "ON", "transition": 20, "brightness": 127}
lights/commands/0x0/rgb_cct/2 {"state": "ON", "transition": 20, "brightness": 127}
lights/states/0x5555/rgb_cct/2 {"state":"ON","status":"ON","brightness":255,"level":100,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x5555/rgb_cct/2 {"state":"ON","status":"ON","brightness":252,"level":99,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x1111/rgb_cct/4 {"state":"ON","status":"ON","brightness":252,"level":99,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x0/rgb_cct/2 {"state":"ON","status":"ON","brightness":252,"level":99,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x5555/rgb_cct/2 {"state":"ON","status":"ON","brightness":250,"level":98,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x1111/rgb_cct/4 {"state":"ON","status":"ON","brightness":250,"level":98,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
[... 35 lines cut ...]
```

### Expected behavior

Bulbs ```light.room```  ```light.hall``` ```light.kitchen``` should fade in 20 seconds to 50 % brightness. 

#### Firmware version

1.10.1

#### Output of http://milight-hub.local/about

```
{"firmware":"milight-hub","version":"1.10.1","ip_address":"192.168.xxxx","reset_reason":"External System","variant":"nodemcuv2","free_heap":15608,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":438}}
```

#### Output of http://milight-hub.local/settings

```
{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168xxxx:1883","mqtt_username":"xxxxx","mqtt_password":"xxxxx","mqtt_topic_pattern":"lights/commands/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"lights/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow toggle","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","rf24_channels":["LOW","MID","HIGH"],"device_ids":[0,4369,8738,13107,21845,39321,3939],"gateway_configs":[],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect"],"group_id_aliases":{}}
```

### Additional context

I noticed that I'm dropping packets, so I tried to update MQTT state rate limit to 800 ms but it wouldn't save any setting.

badstraw360, 2019-09-03:
okay I changed the settings according to performance tuning: 

```
{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168xxxx1883","mqtt_username":"xxxx","mqtt_password":"xxxx","mqtt_topic_pattern":"lights/commands/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"lights/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":0,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":10,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow toggle","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","rf24_channels":["LOW","MID","HIGH"],"device_ids":[0,4369,8738,13107,21845,39321,3939],"gateway_configs":[],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect"],"group_id_aliases":{}}
```
But I'm still having the same problem.. 

```

lights/commands/0x5555/rgb_cct/2 {"state": "ON", "brightness": 255}
lights/commands/0x1111/rgb_cct/4 {"state": "ON", "brightness": 255}
lights/commands/0x0/rgb_cct/2 {"state": "ON", "brightness": 255}
lights/states/0x5555/rgb_cct/2 {"state":"ON","status":"ON","brightness":255,"level":100,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x5555/rgb_cct/2 {"state":"ON","status":"ON","brightness":255,"level":100,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x0/rgb_cct/2 {"state":"ON","status":"ON","brightness":255,"level":100,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/commands/0x5555/rgb_cct/2 {"state": "ON", "transition": 20, "brightness": 127}
lights/commands/0x1111/rgb_cct/4 {"state": "ON", "transition": 20, "brightness": 127}
lights/commands/0x0/rgb_cct/2 {"state": "ON", "transition": 20, "brightness": 127}
lights/states/0x5555/rgb_cct/2 {"state":"ON","status":"ON","brightness":255,"level":100,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x5555/rgb_cct/2 {"state":"ON","status":"ON","brightness":252,"level":99,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x1111/rgb_cct/4 {"state":"ON","status":"ON","brightness":252,"level":99,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x0/rgb_cct/2 {"state":"ON","status":"ON","brightness":252,"level":99,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x5555/rgb_cct/2 {"state":"ON","status":"ON","brightness":250,"level":98,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x1111/rgb_cct/4 {"state":"ON","status":"ON","brightness":250,"level":98,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x0/rgb_cct/2 {"state":"ON","status":"ON","brightness":247,"level":97,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x0/rgb_cct/2 {"state":"ON","status":"ON","brightness":247,"level":97,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x5555/rgb_cct/2 {"state":"ON","status":"ON","brightness":242,"level":95,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x1111/rgb_cct/4 {"state":"ON","status":"ON","brightness":242,"level":95,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x0/rgb_cct/2 {"state":"ON","status":"ON","brightness":242,"level":95,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x1111/rgb_cct/4 {"state":"ON","status":"ON","brightness":240,"level":94,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x5555/rgb_cct/2 {"state":"ON","status":"ON","brightness":237,"level":93,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x1111/rgb_cct/4 {"state":"ON","status":"ON","brightness":237,"level":93,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x5555/rgb_cct/2 {"state":"ON","status":"ON","brightness":232,"level":91,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
[... cut ...]
lights/states/0x5555/rgb_cct/2 {"state":"ON","status":"ON","brightness":130,"level":51,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x5555/rgb_cct/2 {"state":"ON","status":"ON","brightness":128,"level":50,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
lights/states/0x0/rgb_cct/2 {"state":"ON","status":"ON","brightness":128,"level":50,"kelvin":98,"color_temp":366,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}

```

badstraw360, 2019-09-03:
Ok, it only works as expected if I disable some of the transmit channels under the setting nRF24 Send Channels, but it becomes very unresponsive to the point that it ignores some commands, bulbs won't turn on sometimes or they won't fade manually. But I get dropped_packets to zero.

sidoh, 2019-09-03:
Glad you found the source.  

The things I can do on my end are:

1. Increase the max queue size
2. Increase the default period (it's currently 225ms), or expose a setting
3. Make the nRF radio code better (I don't have the time or motivation to do this right now not get into this, though).

(2) will probably help the most.  If it's too high, though, your transitions will appear choppy.

badstraw360, 2019-09-04:
Yeah it seems that if I try to dim more than one bulb at the same time, it loses packets so it only works every time if it’s one bulb only. 

If you increase the default period, would that affect how fast the bulbs turn off and on? 

Maybe I can try different settings for performance tuning and let you know if they work better, do you have a suggestion I can try?

sidoh, 2019-09-13:
Sorry for the slow reply.

No, it won't affect how quickly the transition runs.  That's controlled by the duration.  The period controls how granular the updates are.  In other words, the smoothness of the transition.

This setting is not currently exposed -- it's hard-coded.  I'll have to do some plumbing to get it in settings.

badstraw360, 2019-09-19:
That would be amazing!! I’d love to see if that makes it work better with multiple bulbs..

sidoh, 2019-09-22:
The setting is live in [1.10.3](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.10.3).

sidoh, 2019-09-27:
Gonna close this issue, please let me know if I've missed something.

closed 2019-09-27 by sidoh