sidoh/esp8266_milight_hub #518: Help with Platformio

feder-ico, 2019-09-13:
Hi, 
I am very new with Platfomio and as recomended I want to use it to upload the sketch to a NODEMCU.
If I download the ZIP file I and open the project I get a lot of errors, so I cloned the project using the "Clone Git Project" option and it compiles and upload with no problems but if I make any change and build and upload again the changes don´t apply.
For instance, if I change the <title> in esp8266_milight_hub\web\src\index.html file and then I build and upload, the title I see when I open http://milight-hub.local is the original and not the one I moddified.

Can anyone help me by saying what I am doing wrong?

Thanks a lot!

sidoh, 2019-09-14:
Check the top of the compile logs and you'll probably see some warnings about npm packages not being present on your system.  It allows the build to continue with the pre-packaged dist binary.

The webpage is built using gulp.  It's packed into one file, compressed, and transformed into a PROGMEM header file.

This is the script that builds the webpage: https://github.com/sidoh/esp8266_milight_hub/blob/master/.build_web.py

feder-ico, 2019-09-14:
Hi @sidoh ,
Thanks for your response.
I really don't see anything in the logs, but frankly I don´t know where to look. Would you be so kind to take a look?

Here is the complete log of the compile: https://www.dropbox.com/s/70h6jhu14y31sf1/comple_log.txt?dl=0 

Thanks in advance.

Best regards.

sidoh, 2019-09-14:
pio might be swallowing them.

The web portion of the build is handled here:

https://github.com/sidoh/esp8266_milight_hub/blob/master/platformio.ini#L28

try replicating what this script is doing directly.  eg on linux/osx:

```
$ npm install && ./node_modules/.bin/gulp
npm WARN esp8266-milight-hub-web@1.0.0 No description
npm WARN esp8266-milight-hub-web@1.0.0 No repository field.

audited 1824 packages in 1.607s
found 8 vulnerabilities (2 low, 6 high)
  run `npm audit fix` to fix them, or `npm audit` for details
[12:06:20] Using gulpfile ~/code/me/arduino/Milight/web/gulpfile.js
[12:06:20] Starting 'clean'...
[12:06:20] Finished 'clean' after 3.16 ms
[12:06:20] Starting 'buildfs_inline'...
[12:06:21] Finished 'buildfs_inline' after 373 ms
[12:06:21] Starting 'buildfs_embeded'...
[12:06:21] Finished 'buildfs_embeded' after 30 ms
[12:06:21] Starting 'default'...
[12:06:21] Finished 'default' after 28 μs
```

The issue is almost certainly that you need to install node.js and npm.

feder-ico, 2019-09-14:
Thanks Chris, you were right, I just installed node.js and it works now!

**Thanks a lot!** and great job!

closed 2019-09-14 by feder-ico