sidoh/esp8266_milight_hub #519: Cannot build with PlatformIO and Win10

dersch81, 2019-09-16:
Hi, i can build on Windows10 with PlatformIO. Python 3.7 is installed locally. 

`   
Processing nodemcuv2 (platform: espressif8266@~1.8; framework: arduino; board: nodemcuv2)
-------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
INFORMATION: Es konnten keine Dateien mit dem angegebenen
Muster gefunden werden.
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 1.8.0 > NodeMCU 1.0 (ESP-12E Module)

HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), tool-esptool 1.413.0 (4.13), framework-arduinoespressif8266 2.20402.4 (2.4.2)
Fatal Python error: initfsencoding: unable to load the file system codec
  File "c:\users\dsteingaesser.dl\.platformio\penv\lib\encodings\__init__.py", line 123
    raise CodecRegistryError,\
                            ^
SyntaxError: invalid syntax

Current thread 0x0000b474 (most recent call first):
OSError: 'python3 .get_version.py' exited -1073740791:
  File "C:\users\dsteingaesser.dl\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 126:

    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\dsteingaesser.DL\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\dsteingaesser.DL\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\users\dsteingaesser.dl\.platformio\platforms\espressif8266@1.8.0\builder\main.py", line 316:
    target_elf = env.BuildProgram()
  File "C:\Users\dsteingaesser.DL\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\dsteingaesser.dl\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 107:
    env.ProcessFlags(env.get("BUILD_FLAGS"))
  File "C:\Users\dsteingaesser.DL\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\dsteingaesser.dl\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 189:
    env.Append(**env.ParseFlagsExtended(flags))
  File "C:\Users\dsteingaesser.DL\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\dsteingaesser.dl\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 152:
    for key, value in env.ParseFlags(str(raw)).items():
  File "C:\Users\dsteingaesser.DL\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 800:
    do_parse(arg)
  File "C:\Users\dsteingaesser.DL\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 676:
    arg = self.backtick(arg[1:])
  File "C:\Users\dsteingaesser.DL\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 594:
    raise OSError("'%s' exited %d" % (command, status))
========================= [FAILED] Took 1.43 seconds =========================
 `

sidoh, 2019-09-17:
I do not have a Windows machine to test with, unfortunately.  Honestly this looks like an issue with your environment, though.  Try searching for the error:

```
Fatal Python error: initfsencoding: unable to load the file system codec
File "c:\users\dsteingaesser.dl.platformio\penv\lib\encodings_init_.py", line 123
raise CodecRegistryError,
^
SyntaxError: invalid syntax
```

itProfi, 2019-09-25:
Same problem, confirm:
````
Executing task: C:\Users\User\.platformio\penv\Scripts\platformio.exe run --environment d1_mini <

Processing d1_mini (platform: espressif8266@~1.8; framework: arduino; board: d1_mini)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
Attempting to build webpage...
npm WARN esp8266-milight-hub-web@1.0.0 No description
npm WARN esp8266-milight-hub-web@1.0.0 No repository field.

b'audited 1824 packages in 3.334s\nfound 8 vulnerabilities (2 low, 6 high)\n  run `npm audit fix` to fix them, or `npm audit` for details\n'
fs.js:27
const { Math, Object } = primordials;
                         ^

ReferenceError: primordials is not defined
    at fs.js:27:26
    at req_ (d:\milight\web\node_modules\natives\index.js:143:24)
    at Object.req [as require] (d:\milight\web\node_modules\natives\index.js:55:10)
    at Object.<anonymous> (d:\milight\web\node_modules\graceful-fs\fs.js:1:37)
    at Module._compile (internal/modules/cjs/loader.js:868:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:879:10)
    at Module.load (internal/modules/cjs/loader.js:731:32)
    at Function.Module._load (internal/modules/cjs/loader.js:644:12)
    at Module.require (internal/modules/cjs/loader.js:771:19)
    at require (internal/modules/cjs/helpers.js:68:18)
b''
Encountered error CalledProcessError building webpage: Command '['node_modules\\.bin\\gulp.cmd']' returned non-zero exit status 1.
WARNING: Failed to build web package. Using pre-built page.
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 1.8.0 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), framework-arduinoespressif8266 2.20402.4 (2.4.2), tool-esptool 1.413.0 (4.13)
"python3" ­Ґ пў«пҐвбп ў­гваҐ­­Ґ© Ё«Ё ў­Ґи­Ґ©
Є®¬ ­¤®©, ЁбЇ®«­пҐ¬®© Їа®Ја ¬¬®© Ё«Ё Ї ЄҐв­л¬ д ©«®¬.
OSError: 'python3 .get_version.py' exited 1:
  File "C:\Users\User\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 126:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\User\.platformio\platforms\espressif8266@1.8.0\builder\main.py", line 316:
    target_elf = env.BuildProgram()
  File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\User\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 107:
    env.ProcessFlags(env.get("BUILD_FLAGS"))
  File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\User\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 189:
    env.Append(**env.ParseFlagsExtended(flags))
  File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\User\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 152:
    for key, value in env.ParseFlags(str(raw)).items():
  File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 807:
    do_parse(arg)
  File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 676:
    arg = self.backtick(arg[1:])
  File "C:\Users\User\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 594:
    raise OSError("'%s' exited %d" % (command, status))
====================================================================================== [FAILED] Took 5.13 seconds ======================================================================================

Environment    Status    Duration
-------------  --------  ------------
nodemcuv2      IGNORED
d1_mini        FAILED    00:00:05.130
esp12          IGNORED
esp07          IGNORED
huzzah         IGNORED
d1_mini_pro    IGNORED
````

sidoh, 2019-09-26:
I don't have a Windows machine to test with, so gonna need some help.

Can you translate this into English?  Looks like something between your terminal and the github comment messed up the encoding.

> PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), framework-arduinoespressif8266 2.20402.4 (2.4.2), tool-esptool 1.413.0 (4.13)
> "python3" ­Ґ пў«пҐвбп ў­гваҐ­­Ґ© Ё«Ё ў­Ґи­Ґ©
> Є®¬ ­¤®©, ЁбЇ®«­пҐ¬®© Їа®Ја ¬¬®© Ё«Ё Ї ЄҐв­л¬ д ©«®¬.
> OSError: 'python3 .get_version.py' exited 1:

I'm guessing you just don't have python3 installed.

gJaouenFr, 2019-10-01:
Hello,

I meet troubles too with platformio compilation of this project under windows 10.

System : windows 10 Pro Build 1903
VSC : 1.38.1
pio home : 2.3.3
pio core : 4.1.0b1

You'll find verbose output in attached file.

Best regards,
Guillaume
[platformio run verbose output.txt](https://github.com/sidoh/esp8266_milight_hub/files/3678749/platformio.run.verbose.output.txt)

sidoh, 2019-10-05:
looks like a different issue than the others.  Maybe check this out:

https://docs.platformio.org/en/latest/faq.html#unicodedecodeerror-non-ascii-characters-found-in-build-environment

not entirely sure where it'd be coming from.

Are you wanting to make changes to the source?  If not, you might want to use one of the pre-compiled binaries:

https://github.com/sidoh/esp8266_milight_hub/releases

localhost61, 2019-10-12:
@sidoh, @gJaouenFr
FYI, under Windows OS, when python3 is not installed and I want to build current master branch, I get this (in French):  
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 1.8.0 > WeMos D1 R2 & mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), framework-arduinoespressif8266 2.20402.4 (2.4.2), tool-esptool 1.413.0 (4.13)
'python3' n'est pas reconnu en tant que commande interne
ou externe, un programme ex‚cutable ou un fichier de commandes.
OSError: 'python3 .get_version.py' exited 1:
```
In my case python 3.6.8 is installed together with python 2.7.16. Python 2 is not in the path and is reached using py.exe launcher: 
```
>py -2 --version
Python 2.7.16
>py -3 --version
Python 3.6.8
>python --version
Python 3.6.8
```
So I edited my local copy of platformio.ini and removed the char '3' in line 32:  
`python .get_version.py`
And got this :-) :
```
Building .pio\build\d1_mini\firmware.bin
Retrieving maximum program size .pio\build\d1_mini\firmware.elf
Checking size .pio\build\d1_mini\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  50.2% (used 41128 bytes from 81920 bytes)
PROGRAM: [=====     ]  47.5% (used 496000 bytes from 1044464 bytes)
==================================== [SUCCESS] Took 30.76 seconds ====================================
Environment    Status    Duration
-------------  --------  ------------
nodemcuv2      IGNORED
d1_mini        SUCCESS   00:00:30.756
esp12          IGNORED
```

localhost61, 2019-10-12:
@dersch81 I'm still using Win8.1, not Win10 but your folder looks weird :  
 `c:\users\dsteingaesser.dl.platformio\penv\lib\encodings_init_.py`
for me, a '\' is missing in front of `.platformio` provided that `dsteingaesser.dl` is your username.

itProfi, 2019-11-03:
Dear friends! Don't waste you time! It is not compilling in Windows 8 and Windows 10! Absolutly not.

localhost61, 2019-11-03:
@itProfi Really?
```
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=====     ]  50.2% (used 41116 bytes from 81920 bytes)
PROGRAM: [=====     ]  47.5% (used 495836 bytes from 1044464 bytes)
====================================================================================================================== [SUCCESS] Took 30.10 seconds ======================================================================================================================

Environment    Status    Duration
-------------  --------  ------------
nodemcuv2      SUCCESS   00:00:30.736
d1_mini        SUCCESS   00:00:29.872
esp12          SUCCESS   00:00:30.102
esp07          SUCCESS   00:00:31.309
huzzah         SUCCESS   00:00:29.786
d1_mini_pro    SUCCESS   00:00:30.102
======================================================================================================================= 6 succeeded in 00:03:01.907 =======================================================================================================================
```
```
D:\RnD\gitHub\MiLight_hub>git rev-parse --short HEAD
7c1e6ad
D:\RnD\gitHub\MiLight_hub>ver
Microsoft Windows [version 6.3.9600]
D:\RnD\gitHub\MiLight_hub>python --version
Python 3.7.4
D:\RnD\gitHub\MiLight_hub>py -3 .get_version.py
-DMILIGHT_HUB_VERSION=1.10.3-3-g7c1e6ad
D:\RnD\gitHub\MiLight_hub>python .get_version.py
-DMILIGHT_HUB_VERSION=1.10.3-3-g7c1e6ad
D:\RnD\gitHub\MiLight_hub>python3 .get_version.py
'python3' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.
```
So instead of trolling, report here your error messages and we may be able to help you fine tune your setup. As a minimal requirement, Python v3.* need to be installed in your system and accessible to PlatformIO.

localhost61, 2019-11-03:
@dersch81 Did you solve your issue?
Early versions of Python v3.7.2 had a distribution [issue](https://bugs.python.org/issue35596) that exhibited the same behavior.
The referred thread should give you some inputs to troubleshoot your issue.
The File System codec is related to .pyc object files that are .py files pre-encoded using vcruntime140.dll for faster execution on Windows, and in that distribution this .dll occurred twice hence the error. Removing the .pyc files located in `c:\users\dsteingaesser.dl.platformio\penv\Lib\site-packages\__pycache__` should temporary solve the problem but I suggest you a better solution that solved all my issues related to python.
- close VSCode if a session is already running
- make a backup of `c:\users\dsteingaesser.dl.platformio` and then remove this folder
- launch VSCode IDE, it will reinstall latest PlatformIO and latest python v3.7 in a breeze.
- make sure that no other python installation appears in the path variable, that would interfere.
- if you need other packages inside PlatformIO, like pycurl, you'll have to install them again with `pip install ...`
- in a platformio Terminal test if python is running properly and try to build milight_hub again
- if you don't get better result or miss a functionality, then delete your .platformio folder again and restore it from the backup.

dersch81, 2019-11-03:
Hi, yes i could solve it
 I just changed pyhton3 to pyhton and could build it. I confirmed it with windows 7 and 10.

closed 2019-11-03 by dersch81

localhost61, 2019-11-04:
@sidoh As python v3 is not mandatory to run  `.get_version.py` please replace !python3 with !python in `platformio.ini` so that it can be compatible with Windows OS.

sidoh, 2019-11-04:
You're welcome to make the change locally, but Python 2 is EOL in two months.  Gonna stick with python3.

sidoh, 2019-11-04:
Let me know if I'm misunderstanding.  I think `python` is supposed to always refer to `python2`.

localhost61, 2019-11-04:
No!
In .\python37 folder there is a python.exe too, here is a screenshot of the folder embedded in platformio:
[image]
When you install latest python v2 it doesn't modify the environment path, so it won't disturb a preexisting v3 installation.