sidoh/esp8266_milight_hub #520: Transition to off not working, possibly because brightness is returned as 0

VossenTech, 2019-09-17:
### Describe the bug

I set up all of my turn off commands with a transition time of 1 second to avoid blinding myself in the night if I need to turn something on. This seems to work well except that if I turn on a light after it transitions to off the initial state seems to have a brightness of 0.

{"state":"ON","status":"ON","brightness":0,"level":0,"kelvin":65,"color_temp":294,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}

When it is turned on in this state the turn off command with transition time will not turn off the light, I see the off message being sent via MQTT but the bulb state is never changed or updated.

If I send the turn off command without the transition time then the bulb turns off as expected.

### Steps to reproduce

Turn off any bulb using a transition and then turn it on again using only the turn on command (don't specify brightness). This should produce a bulb state of ON with brightness of 0.

Then try to turn it off using the transition command again. Result should be no change in the bulb and no state update messages.

### Expected behavior

Two expected behaviors:
1. When turned on the bulb should report an accurate brightness level instead of 0
2. When told to transition to off the bulb should turn off

### Setup information

#### Firmware version
1.10.1 (nodemcuv2)

#### Output of http://milight-hub.local/about

{"firmware":"milight-hub","version":"1.10.1","ip_address":"REDACTED","reset_reason":"Exception","variant":"nodemcuv2","free_heap":7976,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":723}}

#### Output of http://milight-hub.local/settings

{"admin_username":"REDACTED","admin_password":"REDACTED","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"REDACTED:1883","mqtt_username":"REDACTED","mqtt_password":"REDACTED","mqtt_topic_pattern":"milight/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"milight/states/:device_id/:device_type/:group_id","mqtt_client_status_topic":"","simple_mqtt_client_status":false,"discovery_port":48899,"listen_repeats":0,"state_flush_interval":0,"mqtt_state_rate_limit":0,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Off","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"","wifi_mode":"n","rf24_channels":["MID","HIGH"],"device_ids":[1,32,33,34,36,37,38,39,48,102,580,2311,32776],"gateway_configs":[[2311,8000,6],[1,8001,6]],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type"],"group_id_aliases":{}}

### Additional context

sidoh, 2019-09-17:
I'm able to reproduce the second issue, and that's definitely a bug.  Thanks for filing.  I'll look at fixing it when I have more time.

I might see where you're coming from with the first issue:

> When turned on the bulb should report an accurate brightness level instead of 0

but 0 _is_ the correct brightness level when a bulb is turned off with a transition.  All a transition is is an automated ramp down to brightness 0 and a hook to switch the bulb off after it's done.

Can you clarify what you're expecting to happen here, and how it diverges from what you see?

VossenTech, 2019-09-17:
I guess it was just a misunderstanding on my part with what 0 represents. I was expecting a brightness of 0 to mean off and 1 would be the minimum value for on.

The value of 0 doesn't actually cause any other issues that I am aware of, just thought it might be an explanation as to why the transition off didn't work.

Also thanks for your work on this project, its greatly appreciated.

closed 2019-09-18 by sidoh

sidoh, 2019-09-18:
should be fixed in 1.10.2.  thanks again!

commit e92c643e (2019-09-18): Fix #520 -- bug where 0-step status transition had max_int length