sidoh/esp8266_milight_hub #522: HomeKit Support

magichew, 2019-09-18:
Is there any scope for integrating this with HomeKit. 

Perhaps building on this project?

https://github.com/RavenSystem/esp-homekit-devices

sidoh, 2019-09-21:
Certainly open to it, although I'm entirely unfamiliar with the platform and would want some help navigating it.

The libraries I'm seeing in your link are for the RTOS SDK, which is a different (and I believe incompatible) platform.  The Arduino SDK is built on top of the non-OS SDK.

buenyamin-olgun, 2019-09-23:
Is maybe homebridge with mqtt a valid solution?

sidoh, 2019-09-29:
Thanks @buenyamin-olgun.  Yes, using one of the existing integration modes would be preferable.

sidoh, 2019-10-05:
@magichew, any thoughts here?  Is there a way to use either the REST API or the MQTT integration?

magichew, 2019-11-04:
> @magichew, any thoughts here? Is there a way to use either the REST API or the MQTT integration?

Hi sorry, I really don’t have any technical knowledge, I’m just a user of things that way more talented people have put together. 

I’ve had milight working via homebridge but I’m hoping to achieve HomeKit integration without Homebridge.

I have the firmware that I linked to above on several sonoff devices and its working well, I hoped there may have been a way to build HomeKit support in to this custom milight hub concept but I’m afraid all I can contribute to this is the idea.

Zer0x00, 2020-04-16:
Finally I've managed to enhance a homebridge plugin so that it can utilize all of the lovely EMH features.

https://github.com/normen/homebridge-milighthub-platform/