sidoh/esp8266_milight_hub #523: Lost settings after update from 1.10.1 to 1.10.2

ReinierGielen, 2019-09-18:
After updating to 1.10.2 with web UI all settings are gone.
And uploading the backuped settings give a succes message, but there are still no settings

### Steps to reproduce
Started with version 1.10.1
In web UI did "check for updates"
Clicked download firmware.
Updated through webUI
Settings are gone.
Through web UI tried Restore settings without result.

### Expected behavior

After an update setrtings should automatically be restored.
After a restore the stored settings should be rerstored.

#### Firmware version
Stared with 1.10.1 now 1.10.2

ReinierGielen, 2019-09-18:
I think the request payload is missing in the restore request

sidoh, 2019-09-18:
Sounds an awful lot like you uploaded a binary compiled for a different platform.  My guess is you're on a `d1_mini` and you uploaded `d1_mini_pro` or something like that.

sidoh, 2019-09-18:
> I think the request payload is missing in the restore request

[It's not](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/WebServer/MiLightHttpServer.cpp#L217-L225), but it might look that way if the ESP8266 is crashing, which might happen if SPIFFS gets resized without a format.  This would happen if you upload the wrong binary.

ReinierGielen, 2019-09-18:
Ik uploaded the nodemcuv2 binary. Ik have a nodemcuv3. This worked fine on previous updates

sidoh, 2019-09-18:
The [only change in 1.10.2](https://github.com/sidoh/esp8266_milight_hub/commit/e92c643e2f87f0c29e7bce5003a96a8b6a365253) is trivial and wouldn't have caused this kind of behavior:

The only time I've seen the behavior you're describing is when you upload a binary compiled with a [ldscript](https://github.com/esp8266/Arduino/tree/master/tools/sdk/ld) different than the one you had previously. ldscripts are responsible for partitioning the onboard flash into program and SPIFFS space.

You can try wiping the device with `esptool.py erase_flash` and re-uploading from scratch.  This will probably fix it.

If you can find a reliable way to reproduce this I'm happy to look into it.  But I flashed my milight hub install with 1.10.2 last night and haven't had any problems.

poudenes, 2019-09-19:
update here went ok as well.. first backup my settings for "In case"

morrisb74, 2019-09-19:
Mine lost the mqtt username.
But was easily fixed when I found out.

ReinierGielen, 2019-09-19:
I re-flashed my nodemcu with physical connection. It started fine.
Updates over the web UI also work fine.

The only thing is restoring the backuped settings from web UI  is not working in any version.

Is there another way to restore backedup settings ?

sidoh, 2019-09-19:
Hmm... this should definitely work.  There's [a test](https://github.com/sidoh/esp8266_milight_hub/blob/master/test/remote/spec/settings_spec.rb#L42-L82) for it.  The [web UI](https://github.com/sidoh/esp8266_milight_hub/blob/master/web/src/index.html#L505) is definitely calling the correct API endpoint.

I also just tried doing it through the UI myself, and it seems to be working fine:

[image]

Are you able to check if there's anything interesting going on in serial logs when you try to restore settings?

ReinierGielen, 2019-09-19:
Which logs are you referring to

sidoh, 2019-09-19:
Serial logs.  You can access them with `pio device monitor` while the ESP8266 is plugged into USB.

ReinierGielen, 2019-09-19:
I managed to restore the system by manually inputting the data from the settings file.
I don't have pio device monitor installed but I'll attempt to retreive the logs.
I will not make another attempt to use the restore option as this erases the settings and upsets my wife.

Except for the restore your project works briljantly

sidoh, 2019-09-19:
it shouldn't really matter, but what browser are you using?

sidoh, 2019-09-20:
definitely understand not wanting to screw up your setup.  anything you can do to help me replicate the problem is much appreciated.  I've tested on two separate setups on both Firefox and Chrome and haven't been able to reproduce. :(

ReinierGielen, 2019-09-20:
I am using chrome.
I'll be carefull doing the next update. At least I now know how to restore the backup.
Read the JSON and type it in manually. 
That is alricht if it does not happen to often. ;-)

sidoh, 2019-09-20:
I'm reasonably sure there's something kind of screwy about your ESP8266, although I'm not sure what.  Have you used it for anything else before?

Would really like to reproduce the issue -- should never happen.

ReinierGielen, 2019-09-20:
No nothing before. I have purchased it for this goal.
Can it be that is a nodemcu V3 and the build is for a V2 ?
Or maybe the Chinese send me a cheap clone .

sidoh, 2019-09-20:
v2 and v3 are both fine.  it's possible, but the only way in which i think it'd matter is if you have less than 4MB of flash.  You can inspect this with `esptool.py flash_id`:

```
$ esptool.py flash_id
esptool.py v2.7
Found 2 serial ports
Serial port /dev/cu.usbserial-1443230
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 2c:3a:e8:2e:fd:b3
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

anyway --

If you're willing to try one more potentially destructive thing, here's a suggestion that might narrow it down:

1. You can back up and restore your entire ESP8266 image using `esptool.py`.  [Here's a tutorial](https://steve.fi/Hardware/backup-and-restore/).  Hopefully this would avoid you losing your settings again.
2. Erase the entire image with `esptool.py erase_flash`
3. Re-setup the wifi
4. Try restoring settings (also watch serial logs with `pio device monitor` to see if anything interesting shows up

If this works like I suspect it will, you'll have restored settings and no manual re-entry required.  If it doesn't, you can restore your backed up flash image using the process mentioned in (1).

ReinierGielen, 2019-09-21:
Do you have a windows 10 version of the eptool.py ?

sidoh, 2019-09-21:
It's just a python library, it'll work on windows.  (just install with pip)

MunkeyBalls, 2019-09-23:
Not having exactly the same issue (restore works), but since 10.0.1 I'm having a problem that might be related so I'll chime in anyway. 

Since 10.0.1 I notice that I am losing settings after being up and running a for a while (few hours/day). The first time I had to run around the house, re-pair all the bulbs, but luckily was smart enough to take a backup. In probably less than a day, settings were gone again. Restoring the backup luckily works. I've tried restoring a few times, but always ended up losing my settings after a while.

I've tried redownloading/reflashing the image a few times. Also used the esptool to erase_flash thinking something might have gotten left over, but nothing worked. In the end flashed back to  10.0.0-dev4 and all is working again. It's been up and running for around a week. 

Loving the new MQTT Hass discovery and aliases btw, great addition! I will try and flash 10.0.2 to see if I am still having the same problem. 

On a totally unrelated note.. Would it be possible to add something to prevent the hub from (re)-pairing a bulb when someone flips the power switch at the same time another light gets a message to turn on/off ? I've noticed after a while some bulbs seem to merge with others, so 1 lightId all of a sudden controls multiple lights. Usually happens with the bulbs which are used with a wall switch, that get tied to bulbs which are triggered by PIR switches.

sidoh, 2019-09-23:
That's bizarre.  Is there any part of your setup that is automatically updating settings?  Is it possible for you to capture some serial logs while running a problematic version?  Might help reveal what's going on.

> On a totally unrelated note.. Would it be possible to add something to prevent the hub from (re)-pairing a bulb when someone flips the power switch at the same time another light gets a message to [...]

It's silly, but bulb pair packets are the same as ON commands.  So as long as something is sending ON commands at the same time that you're physically flipping power on, there's no way to prevent the bulbs from interpreting that as a pair command.

MunkeyBalls, 2019-09-24:
> 
> 
> That's bizarre. Is there any part of your setup that is automatically updating settings? Is it possible for you to capture some serial logs while running a problematic version? Might help reveal what's going on.
> 

So after about day on 10.0.2 the settings disappeared again. 

Can I build my own version with the debug flags enabled, or do I need to hook it up to nodeMCU to the PC and monitor it that way (which would be a pain since it only happens after about a day).

In the meantime I'll have a look at "esptool.py flash_id" to see if there's something funny about this nodeMCU. Weird thing is, the old version works fine, so it must be due to something added between 10.0.0-dev4  and 10.0.1

> It's silly, but bulb pair packets are the same as ON commands. So as long as something is sending ON commands at the same time that you're physically flipping power on, there's no way to prevent the [...]

I was already afraid it'd be something like this.  Gives me a bad excuse to go all out on the PIR sensors where I don't have them yet and add electroshock to the physical switches to discourage usage :)

MunkeyBalls, 2019-09-24:
Here's esptool.py flash_id output, nothing odd there I think

```
Found 1 serial ports
Serial port COM4
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:f3:eb:1a:c7:31
Uploading stub...
Running stub...
Stub running...
Manufacturer: 68
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

sidoh, 2019-09-24:
Thanks for all of that detail.  Great -- serial logs with debug flags would be super useful.  Appreciate the effort!

ReinierGielen, 2019-09-24:
It happend tot me too. For the first time suddenly all settings are gone.

ReinierGielen, 2019-09-25:
Strangest thing, not the same settings file is loaded in exactly the same way all messages look the same.
But now it is actually working

sidoh, 2019-09-26:
There's definitely something screwy going on with SPIFFS.  I'm starting to suspect something in the compile toolchain changed the ldscript without telling me.  Could've been a platformio upgrade or something.

Doesn't really explain the behavior where it works for a while and then doesn't, though...

MunkeyBalls, 2019-09-26:
The version I built from source is still running after 2 days. Could it maybe be something with the binaries?

sidoh, 2019-09-26:
Yeah, potentially.  Can you paste `pio --version`?

The output of `pio run -e nodemcuv2 --verbose` would be helpful as well.

Thanks for your help.

MunkeyBalls, 2019-09-28:
Never mind, settings were gone again this morning. 
I'll have to figure out first how to monitor serial from the pi though before I can get you any logs.

sidoh, 2019-09-28:
periodic output of /about might be interesting as well.

MunkeyBalls, 2019-09-28:
This is after restoring the backup, I'm guessing the free_heap is what you want me to keep an eye on?

{"firmware":"milight-hub","version":"1.10.3","ip_address":"192.168.2.51","reset_reason":"Exception","variant":"nodemcuv2","free_heap":16736,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}

MunkeyBalls, 2019-09-28:
Currently rolling back to older versions to try and figure out on which release exactly the issue got introduced. 

Running now on 1.10.0-dev.7

sidoh, 2019-09-28:
yeah, exactly.   I think my idle free heap is ~17 KB, so yours looks fine there.  Possible you're using some feature that I don't normally use that leaks memory.

output of `/settings` would be helpful too (make sure to sensor passwords)

this is meaningful as well:

> "reset_reason":"Exception"

sidoh, 2019-09-28:
Oops, submitted that earlier than I meant to.  `{"reset_reason":"Exception"}` means something is crashing, so Serial output would likely be interesting.  Can't think of a reason that would clear settings, though.

MunkeyBalls, 2019-09-28:
Hooked up the nodemcu with latest version to a pi and its logging to file with all the debug flags enabled. If it fails again hopefully I can catch the log.

sidoh, 2019-09-28:
Excellent.  Thanks for your help.

MunkeyBalls, 2019-09-28:
Here's my settings btw
```
{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"192.168.2.52","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"milight/commands/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"milight/state/:device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":0,"state_flush_interval":10000,"mqtt_state_rate_limit":900,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Off","led_mode_wifi_failed":"Off","led_mode_operating":"Off","led_mode_packet":"Off","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"n","default_transition_period":500,"rf24_channels":["LOW","MID","HIGH"],"device_ids":[1,2,3,4,5,6,7,8],"gateway_configs":[],"group_state_fields":["state","brightness","mode","color_temp","bulb_mode","computed_color","effect"],"group_id_aliases":{"bathroom_milight_fut105":["rgb_cct",3,1],"bedroom_milight_fut105":["rgb_cct",4,1],"deathstar_milight_fut016":["rgbw",5,1],"dinner_table_milight":["rgb_cct",2,1],"kitchen2":["cct",7,1],"kitchen3":["cct",8,1],"kitchen_milight_mr16_1":["cct",6,1],"peoplecave_milight_fut105":["rgb_cct",1,1]}}
```
Just checked the /about again, free heap is a lot lower and there's a new exception type. 
```
{"firmware":"milight-hub","version":"unknown","ip_address":"192.168.2.51","reset_reason":"Software Watchdog","variant":"nodemcuv2","free_heap":7968,"arduino_version":"2_4_2","queue_stats":{"length":0,"dropped_packets":0}}
```

sidoh, 2019-09-28:
definitely looks like a memory leak of some kind.  Did you happen to catch serial logs? WDT reset is watchdog timer reset. It means some blocking code runs for too long and the system gets hard reset. With this software, it's usually the nRF library busy waiting for an SPI response, usually due to spotty wiring connections. But it's possible it's memory related. 

If you're able to find some set of actions that trigger the memory to go down consistently, that would be most helpful. 

I was pretty lazy with the queued packet code. It creates a lot of objects, but they're shared_ptrs, so should be getting cleaned up.

MunkeyBalls, 2019-09-28:
Seems like my log file is being overwritten ever 30 or so minutes, but I think I managed to catch a crash.

``` 
09/28/2019 23:17:22 MqttClient - Got message on topic: milight/commands/0x1/rgb_cct/1
09/28/2019 23:17:22 MqttClient - device 0001, group 1
09/28/2019 23:17:53 MqttClient - Got message on topic: milight/commands/0x1/rgb_cct/1
09/28/2019 23:17:53 MqttClient - device 0001, group 1
09/28/2019 23:18:13 MqttClient - Got message on topic: milight/commands/0x1/rgb_cct/1
09/28/2019 23:18:13 MqttClient - device 0001, group 1
09/28/2019 23:18:13 Enqueuing packet
09/28/2019 23:18:13 Switching to next packet, 1 packets in queue
09/28/2019 23:18:13 Sending packet (10 repeats): 
09/28/2019 23:18:13 00 DB E1 24 66 CA A6 66 A4 
09/28/2019 23:18:13 Elapsed: 47
09/28/2019 23:18:13 Sending packet (10 repeats): 
09/28/2019 23:18:13 00 DB E1 24 66 CA A6 66 A4 
09/28/2019 23:18:13 Elapsed: 45
09/28/2019 23:18:13 Sending packet (10 repeats): 
09/28/2019 23:18:13 00 DB E1 24 66 CA A6 66 A4 
09/28/2019 23:18:13 Elapsed: 46
09/28/2019 23:18:13 Sending packet (10 repeats): 
09/28/2019 23:18:13 00 DB E1 24 66 CA A6 66 A4 
09/28/2019 23:18:13 Elapsed: 49
09/28/2019 23:18:13 Sending packet (10 repeats): 
09/28/2019 23:18:13 00 DB E1 24 66 CA A6 66 A4 
09/28/2019 23:18:13 Elapsed: 46
09/28/2019 23:18:13 Patching existing state with: {"state":"OFF"}
09/28/2019 23:18:13 GroupState::patch: State changed: {"state":"OFF"}
09/28/2019 23:18:13 Raw data: 40000000 70000000
09/28/2019 23:18:13 Patching existing state with: : {"state":"OFF"}
09/28/2019 23:18:13 Raw data: 40000000 70000000
09/28/2019 23:18:13 
09/28/2019 23:18:13 Patching existing state with: : {"state":"OFF"}
09/28/2019 23:18:13 Raw data: 40000000 70000000
09/28/2019 23:18:13 
09/28/2019 23:18:14 Clearing fields.  Current state: {}
09/28/2019 23:18:14 Raw data: 00000000 00000000
09/28/2019 23:18:14 Other state: {"state":"OFF"}
09/28/2019 23:18:14 Raw data: 40000000 70000000
09/28/2019 23:18:14 Result: {}
09/28/2019 23:18:14 Raw data: 00000000 00000000
09/28/2019 23:18:14 MqttClient - publishing update to milight/state/0x1/rgb_cct/1
09/28/2019 23:21:32 MqttClient - Got message on topic: milight/commands/0x1/rgb_cct/1
09/28/2019 23:21:32 MqttClient - device 0001, group 1
09/28/2019 23:21:32 Enqueuing packet
09/28/2019 23:21:32 Flushing dirty state for 0x0001 / 1 / rgb_cct
09/28/2019 23:21:32 Switching to next packet, 1 packets in queue
09/28/2019 23:21:32 Sending packet (10 repeats): 
09/28/2019 23:21:32 00 DB E1 24 66 D1 A7 66 A8 
09/28/2019 23:21:32 Elapsed: 49
09/28/2019 23:21:32 Sending packet (10 repeats): 
09/28/2019 23:21:32 00 DB E1 24 66 D1 A7 66 A8 
09/28/2019 23:21:32 Elapsed: 47
09/28/2019 23:21:32 Sending packet (10 repeats): 
09/28/2019 23:21:32 00 DB E1 24 66 D1 A7 66 A8 
09/28/2019 23:21:33 Elapsed: 46
09/28/2019 23:21:33 Sending packet (10 repeats): 
09/28/2019 23:21:33 00 DB E1 24 66 D1 A7 66 A8 
09/28/2019 23:21:33 Elapsed: 46
09/28/2019 23:21:33 Sending packet (10 repeats): 
09/28/2019 23:21:33 00 DB E1 24 66 D1 A7 66 A8 
09/28/2019 23:21:33 Elapsed: 46
09/28/2019 23:21:33 Patching existing state with: {"state":"ON"}
09/28/2019 23:21:33 GroupState::patch: State changed: {"state":"ON"}
09/28/2019 23:21:33 Raw data: 40000001 70000000
09/28/2019 23:21:33 Patching existing state with: : {"state":"ON"}
09/28/2019 23:21:33 Raw data: 40000001 70000000
09/28/2019 23:21:33 
09/28/2019 23:21:33 Patching existing state with: : {"state":"ON"}
09/28/2019 23:21:33 Raw data: 40000001 70000000
[... 287 lines cut ...]
09/28/2019 23:33:26   Wrote 128 bytes
09/28/2019 23:33:26   Wrote 128 bytes
09/28/2019 23:33:26   Wrote 36 bytes
09/28/2019 23:33:26 HomeAssistantDiscoveryClient: adding discoverable device: kitchen_milight_mr16_1...
09/28/2019 23:33:26   topic: homeassistant/light/milight_hub_1754929/cct_0x6_1/config
09/28/2019 23:33:27 config: {"schema":"json","name":"kitchen_milight_mr16_1","command_topic":"milight/commands/0x6/cct/1","state_topic":"milight/state/0x6/cct/1","device":{"manufacturer":"esp8266_milight_hub","sw_version":"unknown","identifiers":[1754929,6,"cct",1]},"availability_topic":"milight/client_status","payload_available":"connected","payload_not_available":"disconnected","brightness":true,"effect":true,"effect_list":["night_mode"],"color_temp":true}
09/28/2019 23:33:27 Printing message in parts because it's too large for the packet buffer (434 bytes)  Wrote 128 bytes
09/28/2019 23:33:27   Wrote 128 bytes
09/28/2019 23:33:27   Wrote 128 bytes
09/28/2019 23:33:27   Wrote 50 bytes
09/28/2019 23:33:27 HomeAssistantDiscoveryClient: adding discoverable device: peoplecave_milight_fut105...
09/28/2019 23:33:27   topic: homeassistant/light/milight_hub_1754929/rgb_cct_0x1_1/config
09/28/2019 23:33:28 config: {"schema":"json","name":"peoplecave_milight_fut105","command_topic":"milight/commands/0x1/rgb_cct/1","state_topic":"milight/state/0x1/rgb_cct/1","device":{"manufacturer":"esp8266_milight_hub","sw_version":"unknown","identifiers":[1754929,1,"rgb_cct",1]},"availability_topic":"milight/client_status","payload_available":"connected","payload_not_available":"disconnected","brightness":true,"effect":true,"effect_list":["night_mode","white_mode","0","1","2","3","4","5","6","7","8"],"rgb":true,"color_temp":true}
09/28/2019 23:33:28 Printing message in parts because it's too large for the packet buffer (509 bytes)  Wrote 128 bytes
09/28/2019 23:33:28   Wrote 128 bytes
09/28/2019 23:33:28   Wrote 128 bytes
09/28/2019 23:33:28   Wrote 125 bytes
09/28/2019 23:33:28 HomeAssistantDiscoveryClient: Removing discoverable devices...
09/28/2019 23:33:28 MqttClient - Got message on topic: milight/commands/0x1/rgb_cct/1
09/28/2019 23:33:28 MqttClient - device 0001, group 1
09/28/2019 23:33:28 Couldn't fetch state for 0x0001 / 1 / rgb_cct in the cache, getting it from persistence
09/28/2019 23:33:54 MqttClient - Got message on topic: milight/commands/0x1/rgb_cct/1
09/28/2019 23:33:54 MqttClient - device 0001, group 1

```

sidoh, 2019-09-29:
thank you!

Looks like the WDT reset is just because the Serial debug logs for the discovery client take too long to print.  Just changed the baudrate from the default (9600) to 115200  and that went away.  Could you do that as well?  Look for `Serial.begin(9600)` in `main.cpp`.  You'll need to start serial logs with `pio device monitor -b 115200` as well.

Was initially suspecting that the leak was in the discovery client, but doesn't seem to be the case.  Set this up:

<details><summary>Heap trend with successive MQTT server reboots</summary>
<p>

```
$ for i in $(seq 1 100); do docker start mosquitto>/dev/null; curl -s 10.133.8.194/about | jq '.free_heap'; sleep 10; docker stop mosquitto>/dev/null; done
16248
16168
16432
16592
16424
15504
16328
16328
16248
15920
16328
16424
16424
16424
16096
16248
16248
15328
16328
16328
16480
16096
15176
16248
15936
16264
16328
16424
16424
16480
15768
15408
16328
16328
16384
16328
14736
16096
16248
16304
16248
15408
16328
16000
16368
16328
16264
16320
16248
15408
16328
16328
16384
16328
15408
16328
16248
16248
16344
16328
16096
15768
16328
15504
16424
16424
15824
16328
14736
16328
16264
16304
16328
15504
16368
16368
16424
16328
15176
16248
16120
16488
16328
15504
16424
16096
16032
16328
15352
16272
16272
```

</p>
</details>

There's no downwards trend.  There are a few times when it dips into the 14Ks, but it pops back up.  So it's at least not this alone.

You might try tossing `tee` on your command to log to a file if your scrollback buffer is eating logs.  I'm using:

```
pio device monitor -b 115200 | tee -a pio_log.txt
```

something interesting from your logs -- it looks like you've got some kind of scheduled command being sent roughly every minute to 0x1,rgb_cct,1.  is this expected?

sidoh, 2019-09-29:
Added a heap monitor to HomeAssistant:

```
- name: "ESP8266 Milight Hub - Free Heap"
  platform: rest
  resource: http://10.133.8.158/about
  value_template: '{{ value_json.free_heap }}'
  unit_of_measurement: 'Bytes'
```

MunkeyBalls, 2019-09-29:
I was using a python script to read the serial output, after I changed to baud 115200 I started getting the folowing error: 

```
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```

I noticed the same error when trying to read the device with 
```
pio device monitor -b 115200 | tee -a pio_log.txt
```

Worked my way around it with a try / catch and it's working, but the output is unreadable at times.
```
09/29/2019 23:22:35 : {0
09/29/2019 23:22:36 w d09/29/2019 23:23:06 MqttClient - Got message on topic: milight/commands/0x3/rgb_cct/1
09/29/2019 23:23:06 MqttClient - device 0003, group 1
09/29/2019 23:23:06 Enqueuing packet
09/29/2019 23:23:06 Enqueuing packet
09/29/2019 23:23:06 Switching to next packet, 2 (10 repeats): 
09/29/2019 23:23:06 00 DB E1 22 63 F6repeats): 
09/29/2019 23:23:06 00 DB66 BF 
09/29/2019 23:23:06 repeats): 
09/29/2019 23:23:06 
09/29/2019 23:23:06 re66Elre66Elre66Elre66Elre66Eltaou{}Pa{}
09/29/2019 23:23:06 : {0
09/29/2019 23:23:07 w 09/29/2019 23:23:37 c: 1
09/29/2019 23:23:37 moa:is":80ti"c00. 00"bawt:00e 
09/29/2019 23:23:37 Pa{}
09/29/2019 23:23:37 : {0
09/29/2019 23:24:08 1
09/29/2019 23:24:08 00e 
09/29/2019 23:24:08 66Elre66Elre66Elre66Elre66Eltate wiou{}Pa{}
09/29/2019 23:24:08 0
09/29/2019 23:24:09 at0000w 09/29/2019 23:24:39 MqttCc:1
09/29/2019 23:24:39 66Elrep66Elare6Elapsed: 50
09/29/2019 23:24:39 Sending packet (10 repeats): 
09/29/2019 23:24:39 is":80ti"c00. 000"bawt:00e 
09/29/2019 23:24:39 Elre66Elre66Elare66Eltaou{}Pat{}
09/29/2019 23:24:39 : {0
09/29/2019 23:24:40 at0000w d09/29/2019 23:25:10 1
09/29/2019 23:25:10 awt:00e 
09/29/2019 23:25:10 Elataou{}Pa{}
09/29/2019 23:25:10 0
09/29/2019 23:25:11 w 
``` 
 
Probably some settings in my script, but I didn't find a magic combination yet
```
ser = serial.Serial(
 port='/dev/ttyUSB0',
 baudrate = 115200,
 parity=serial.PARITY_NONE,
 stopbits=serial.STOPBITS_ONE,
 bytesize=serial.EIGHTBITS,
 timeout=1
)
counter=0
```

The heap monitor is great idea, I'll keep an eye on the graph and hope I can read the log around those times.

sidoh, 2019-09-29:
You might try passing `-p /dev/ttyUSB0`.  I'm guessing that PlatformIO is using the same Serial library under the hood considering it's Python, so it should, in theory, work :)

Here's the full command I'm using:

```
$ pio device monitor -b 115200 | ts '[%Y-%m-%d %H:%M:%S]' | tee -a /tmp/pio_logs.txt
--- Miniterm on /dev/cu.usbserial-1443230  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[2019-09-29 16:45:54] MqttClient - connecting
[2019-09-29 16:45:54] ERROR: Failed to connect to MQTT server
[2019-09-29 16:45:59] MqttClient - connecting
[2019-09-29 16:45:59] ERROR: Failed to connect to MQTT server
```

`ts` is from [moreutils](https://joeyh.name/code/moreutils/), in case it's helpful.

thanks again for your help!

MunkeyBalls, 2019-09-30:
This is what I mean, notice the missing characters and finally just crashing altogether.
---
dietpi@DietPi:~$ pio device monitor -b 115200 | ts '[%Y-%m-%d %H:%M:%S]' | tee -a /tmp/pio_logs.txt
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[2019-09-30 08:07:39] qc1
[2019-09-30 08:07:39] MqttClient - device 0003, gup
[2019-09-30 08:07:39] Fhing dirty state for 0x0003  ExeDEreats):
[2019-09-30 08:07:39] 00 DB E1 22 63 E1 5D 6lrts):

--- exit ---
Exception in thread rx:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/serial/tools/miniterm.py", line 445, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
---

MunkeyBalls, 2019-09-30:
> 
> something interesting from your logs -- it looks like you've got some kind of scheduled command being sent roughly every minute to 0x1,rgb_cct,1. is this expected?

That would probably be flux https://www.home-assistant.io/components/flux/

sidoh, 2019-09-30:
Ah, got it.  Yeah, Serial only works with one connection.

Ok, cool.  Thanks again.

MunkeyBalls, 2019-09-30:
It crashed again shortly after, seems like the connection isn't really stable. Trying to make something in bash with your command to keep retrying.

MunkeyBalls, 2019-09-30:
Here's how the heap size graph is looking currently. Stable when nothing is happening, but a a huge dip when I started playing around with the lights / serial log. 

[image]

MunkeyBalls, 2019-09-30:
Settings dissapeared now, heap size is still low and not recovering.
<details>
  <summary>Here's the rest of the log.</summary>

```
[2019-09-30 08:57:22] MqttClient - publishing updaeMqttClient - Got message on topic: milight/commands/0x1/rgb_cct/1
[2019-09-30 08:57:22] MqttClient - device 0001, group 1
[2019-09-30 08:57:22] Flushing dirty state for 0x0001 / 1 / rgb_cct
[2019-09-30 08:57:22] Enqueuing packet
[2019-09-30 08:57:22] Enqueuing packet
[2019-09-30 08:57:22] Switching to next packet, 2 packets in queue
[2019-09-30 08:57:22] Sending packet (10 repeats): 
[2019-09-30 08:57:22] 00 DB E1 24 63 E1 94 66 26 
[2019-09-30 08:57:22] Elapsed: 45
[2019-09-30 08:57:22] Sending packet (10 repeats): 
[2019-09-30 08:57:22] 00 DB E1 24 63 E1 94 66 26 
[2019-09-30 08:57:22] Elapsed: 48
[2019-09-30 08:57:22] Sending packet (10 repeats): 
[2019-09-30 08:57:22] 00 DB E1 24 63 E1 94 66 26 
[2019-09-30 08:57:22] Elapsed: 45
[2019-09-30 08:57:22] Sending packet (10 repeats): 
[2019-09-30 08:57:22] 00 DB E1 24 63 E1 94 66 26 
[2019-09-30 08:57:22] Elapsed: 46
[2019-09-30 08:57:22] Sending packet (10 repeats): 
[2019-09-30 08:57:22] 00 DB E1 24 63 E1 94 66 26 
[2019-09-30 08:57:22] Elapsed: 47
[2019-09-30 08:57:22] Patching existing state with: {"hue":25}
[2019-09-30 08:57:22] GroupState::patch: State changed: {"bulb_mode":"color","hue":25}
[2019-09-30 08:57:22] Raw data: 88001200 30000800
[2019-09-30 08:57:22] Patching existing state with: : {"bulb_mode":"color","hue":25}
[2019-09-30 08:57:22] Raw data: 88001200 30000800
[2019-09-30 08:57:22] 
[2019-09-30 08:57:22] Patching existing state with: : {"bulb_mode":"color","hue":25}
[2019-09-30 08:57:22] Raw data: 88001200 30000800
[2019-09-30 08:57:22] 
[2019-09-30 08:57:22] Clearing fields.  Current state: {}
[2019-09-30 08:57:22] Raw data: 00000000 00000000
[2019-09-30 08:57:22] Other state: {"bulb_mode":"color","hue":25}
[2019-09-30 08:57:22] Raw data: 88001200 30000800
[2019-09-30 08:57:22] Result: {}
[2019-09-30 08:57:22] Raw data: 00000000 00000000
[2019-09-30 08:57:22] MqttClient - publishing update to milight/state/0x1/rgb_cct/1
[2019-09-30 08:57:22] Switching to next packet, 1 packets in queue
[2019-09-30 08:57:22] Sending packet (10 repeats): 
[2019-09-30 08:57:22] 00 DB E1 24 61 C6 95 66 0A 
[2019-09-30 08:57:22] Elapsed: 47
[2019-09-30 08:57:22] Sending packet (10 repeats): 
[2019-09-30 08:57:22] 00 DB E1 24 61 C6 95 66 0A 
[2019-09-30 08:57:22] Elapsed: 47
[2019-09-30 08:57:22] Sending packet (10 repeats): 
[2019-09-30 08:57:22] 00 DB E1 24 61 C6 95 66 0A 
[2019-09-30 08:57:22] Elapsed: 46
[2019-09-30 08:57:22] Sending packet (10 repeats): 
[2019-09-30 08:57:22] 00 DB E1 24 61 C6 95 66 0A 
[2019-09-30 08:57:22] Elapsed: 49
[2019-09-30 08:57:22] Sending packet (10 repeats): 
[2019-09-30 08:57:22] 00 DB E1 24 61 C6 95 66 0A 
[2019-09-30 08:57:22] Elapsed: 48
[2019-09-30 08:57:22] Patching existing state with: {"saturation":13}
[2019-09-30 08:57:22] GroupState::patch: State changed: {}
[2019-09-30 08:57:22] Raw data: 000D0000 30000100
[2019-09-30 08:57:22] Patching existing state with: : {}
[2019-09-30 08:57:22] Raw data: 000D0000 30000100
[2019-09-30 08:57:22] 
[2019-09-30 08:57:22] Patching existing state with: : {}
[2019-09-30 08:57:22] Raw data: 000D0000 30000100
[2019-09-30 08:57:22] 
[2019-09-30 08:57:22] Clearing fields.  Current state: {}
[2019-09-30 08:57:22] Raw data: 00000000 00000000
[2019-09-30 08:57:22] Other state: {}
[... 1033 lines cut ...]
[2019-09-30 09:06:45] Sending packet (10 repeats): 
[2019-09-30 09:06:45] 00 DB E1 24 66 CA FC 66 FA 
[2019-09-30 09:06:45] Elapsed: 47
[2019-09-30 09:06:45] Patching existing state with: {"state":"OFF"}
[2019-09-30 09:06:45] GroupState::patch: State changed: {"state":"OFF"}
[2019-09-30 09:06:45] Raw data: 40000000 70000000
[2019-09-30 09:06:45] Patching existing state with: : {"state":"OFF"}
[2019-09-30 09:06:45] Raw data: 40000000 70000000
[2019-09-30 09:06:45] 
[2019-09-30 09:06:45] Patching existing state with: : {"state":"OFF"}
[2019-09-30 09:06:45] Raw data: 40000000 70000000
[2019-09-30 09:06:45] 
[2019-09-30 09:06:45] Clearing fields.  Current state: {}
[2019-09-30 09:06:45] Raw data: 00000000 00000000
[2019-09-30 09:06:45] Other state: {"state":"OFF"}
[2019-09-30 09:06:45] Raw data: 40000000 70000000
[2019-09-30 09:06:45] Result: {}
[2019-09-30 09:06:45] Raw data: 00000000 00000000
[2019-09-30 09:06:45] MqttClient - publishing update to milight/state/0x1/rgb_cct/1
[2019-09-30 09:14:49] Unhandled websocket event: 10
[2019-09-30 09:22:03] Couldn't fetch state for 0x0000 / 1 / rgb_cct in the cache, getting it from persistence
[2019-09-30 09:22:03] Flushing dirty state for 0x0000 / 1 / rgb_cct
[2019-09-30 09:22:07] Unhandled websocket event: 10
```
</details>

MunkeyBalls, 2019-10-01:
Small update, I've been playing around trying to figure out which version introduced the memory leak.

The blue dotted lines are the moments where I've restarted the hub or installed a different version.

As you can see I only ran the 1.10.0-rc.4 version for a few minutes because it went downhill fast. I reverted to 1.10.0-rc.3 and it's been working well so far. If 1.10.0-rc.3 keeps going well I will reinstall 1.10.0-rc.4 to confirm that's the version where the leak was introduced. 

What I also noticed is that the dev4 version has more available memory in general.

[image]

sidoh, 2019-10-01:
Thanks, this is helpful.  There's not a ton of meaningful stuff new in rc.4, but the HASS discovery messages got a bit bigger.

Note that it's not terribly concerning that memory decreases quickly, as long as it recovers to normal levels.

Mine has been slowly heading down as well, but has been nowhere near as dramatic:

[image]

[image]

what are you using for graphing there?  looks much better than HASS' builtin stuff.

sidoh, 2019-10-01:
Could you maybe try running for a bit with HASS discovery disabled?  Your devices should still work, as it does not currently clean up after itself when disabled.

MunkeyBalls, 2019-10-01:
> 
> what are you using for graphing there? looks much better than HASS' builtin stuff.

I'm using grafana and influxDB, it's really great.

You can setup home-assistant to automagically log all your sensor data to influxDB. You can then use grafana to make beautiful dashboards like this:

https://play.grafana.org/d/000000012/grafana-play-home?orgId=1
Looks nothing like my dashboards though :)

I'll reinstall the latest version and turn off discovery to monitor what happens. Thanks again for the great project! 👍

sidoh, 2019-10-05:
Thanks for the data.  Mine has still holding steady, which I'm super confused by.  It might just be that the way I use the hub doesn't trigger the issue.  My rough daily routine doesn't involve a lot more than turning up to 5 groups of bulbs on at the same time.

Are you able to share what sorts of scenes/routines/commands you've got set up?  Might help me reproduce the leak.

> When I looked at the times on the graph I was suprised to see the memory started tanking at a time when noone was home (and no light states should have changed)... So I checked out the log from home-assistant and noticed another strange thing. The lights seem to be going unavailable constantly. I don't think this used to happen much on the older version, I'm going to reinstall the last working version and keep an eye on the logbook.

This is definitely interesting.  Serial logs from around the same time would be very helpful if you've got them.

Going unavailable means that the MQTT client is losing its connection with your broker.  I tried [forcing this problem here](https://github.com/sidoh/esp8266_milight_hub/issues/523#issuecomment-536327528), but it didn't reproduce the symptoms (i.e., no memory leak).  So my guess is that this is a symptom and not a cause.

I think serial logs at the times you're seeing (A) the lights going unavailable, and (B) the heap is tanking are going to be our best bet.

Really appreciate your help here!

sidoh, 2019-10-05:
btw -- having all Serial debug logs enabled probably does degrade performance.  you might try disabling DEBUG_PRINTF and STATE_DEBUG in particular.  I don't expect the output from them to be useful, and they're both really noisy.

ReinierGielen, 2019-10-05:
My setup has been running fine the last couple of weeks. Ik even used the restore a couplet of times. 
Ik currently user 10.2
So i have no data that Will help

sidoh, 2019-10-06:
Just tried running a stress test script without any significant effect (see chart and data below).  

Summary of what it does:

1. Turns groups on so state changes are registered
1. Spams each group, all at the same time.  The packet queue is flooded
1. Waits for the packet queue to flush
1. Measures heap

<details>
<summary>$ cat stress_test.sh</summary>
<p>

```bash
#!/usr/bin/env bash

set -eo pipefail

for ix in $(seq 1 8); do
  mosquitto_pub -h localhost -t "milight_test/commands/0x118d/fut089/${group}" -m '{"status":"ON"}'
done

for ix in $(seq 1 100); do
  for i in $(seq 1 10); do 
    for group in $(seq 1 8); do
      mosquitto_pub -h localhost -t "milight_test/commands/0x118d/fut089/${group}" -m "{\"hue\":$i}"
    done
  done

  printf "%3d" $ix
  n=0

  while [ $(curl -s 10.133.8.194/about | jq '.queue_stats.length') -gt 0 ]; do
    n=$(($n+1))
    printf "."
    sleep 1
  done

  printf "%$((15-$n))s" ""

  printf "%d // %d\n" $(curl -s 10.133.8.194/about | jq '.free_heap') $(curl -s 10.133.8.158/about | jq '.free_heap')
done
```

</p>
</details>

<details>
<summary>./stress_test.sh</summary>
<p>

```
  1........       16776 // 17344
  2.......        16776 // 17344
  3......         16928 // 17496
  4......         16928 // 17496
  5.......        17032 // 17600
  6.......        16792 // 17360
  7.......        16984 // 17552
  8.......        16984 // 17552
  9.......        16984 // 17552
 10.......        16984 // 17552
 11.......        16984 // 17552
 12.......        16984 // 17552
 13.......        16776 // 17344
 14.......        16992 // 17560
 15.......        17104 // 17672
 16.......        17104 // 17672
 17.......        16728 // 17296
 18.......        16400 // 16968
 19.......        17008 // 16584
 20.......        16912 // 17480
 21.......        16944 // 17512
 22.......        16944 // 17512
 23.......        16984 // 17552
 24.......        16984 // 17552
 25.......        16928 // 17496
 26.......        16928 // 17496
 27.......        16600 // 17496
 28.......        17032 // 17600
 29.......        17032 // 17600
 30.......        16792 // 17360
 31.......        16016 // 16584
 32.......        16016 // 16584
 33.......        16016 // 16584
 34.......        16720 // 17048
 35.......        16720 // 17288
 36.......        16720 // 17288
 37.......        16928 // 17496
 38.......        16776 // 17344
 39.......        17032 // 17600
 40.......        17032 // 17600
 41.......        17032 // 17600
 42.......        17032 // 17600
 43.......        17032 // 17600
 44.......        17032 // 17600
 45.......        17032 // 17600
 46.......        16704 // 17600
 47.......        16432 // 17672
 48.......        16928 // 17496
 49.......        16928 // 17496
 50.......        16256 // 17496
 51.......        17032 // 17600
 52.......        17032 // 17600
 53.......        17032 // 17600
 54.......        17032 // 17272
 55.......        17032 // 17600
 56.......        16944 // 17512
 57.......        16704 // 17512
 58.......        16016 // 16584
 59.......        16928 // 17496
 60.......        17104 // 17672
 61.......        16864 // 17432
 62.......        17032 // 17600
 63.......        17032 // 17600
 64.......        16704 // 17272
 65.......        17096 // 17664
 66.......        17096 // 17664
 67.......        17112 // 17344
 68.......        17112 // 17344
 69.......        17112 // 17344
 70.......        17112 // 17344
 71.......        17072 // 17496
 72.......        17072 // 17496
 73.......        17072 // 17496
 74.......        17104 // 17672
 75.......        17104 // 17672
 76.......        17104 // 17672
 77.......        16776 // 17344
 78.......        16776 // 17344
 79.......        16776 // 17344
 80.......        16776 // 17344
 81.......        16776 // 17344
 82.......        16928 // 17576
 83.......        16928 // 17576
 84.......        16600 // 17248
 85.......        17032 // 17600
 86.......        17032 // 17600
 87.......        17104 // 17672
 88.......        16864 // 17672
 89.......        17104 // 17672
 90.......        17104 // 17672
 91.......        16944 // 17512
 92.......        16944 // 17512
 93.......        16944 // 17512
 94.......        17008 // 17496
 95.......        16888 // 17496
 96.......        16912 // 17600
 97.......        16744 // 17312
 98.......        16984 // 17552
 99.......        16984 // 17552
100.......        16984 // 17552
```
</p>
</details>

[image]

@MunkeyBalls, hoping you'll be able to find a routine or script that'll reproduce the leak.

MunkeyBalls, 2019-10-07:
I had to revert back to an old version for the WAF

Last night I reinstalled the latest version and ran some logs. Sorry for the missing datetime stamp, I migrated to a different PI and forgot about it.

Here's the log:
```
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
MqttClient - Connecting to: 192.168.2.254
MqttClient - connecting
ERROR: Failed to connect to MQTT server
*WM: Adding parameter
*WM: staticIP
*WM: Adding parameter
*WM: netmask
*WM: Adding parameter
*WM: gateway
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.2.51
Wifi connected succesfully

Setup complete (version unknown)
MqttClient - connecting
MqttClient - subscribing to topic: milight/commands/+/+/+
MqttClient - Successfully connected to MQTT server
HomeAssistantDiscoveryClient: Sending discoverable devices...
HomeAssistantDiscoveryClient: adding discoverable device: bathroom_milight_fut105...
  topic: homeassistant/light/milight_hub_1754929/rgb_cct_0x3_1/config
config: {"schema":"json","name":"bathroom_milight_fut105","command_topic":"milight/commands/0x3/rgb_cct/1","state_topic":"milight/state/0x3/rgb_cct/1","device":{"manufacturer":"esp8266_milight_hub","sw_version":"unknown","identifiers":[1754929,3,"rgb_cct",1]},"availability_topic":"milight/client_status","payload_available":"connected","payload_not_available":"disconnected","brightness":true,"effect":true,"effect_list":["night_mode","white_mode","0","1","2","3","4","5","6","7","8"],"rgb":true,"color_temp":true}
Printing message in parts because it's too large for the packet buffer (507 bytes)  Wrote 128 bytes
  Wrote 128 bytes
  Wrote 128 bytes
  Wrote 123 bytes
HomeAssistantDiscoveryClient: adding discoverable device: bedroom_milight_fut105...
  topic: homeassistant/light/milight_hub_1754929/rgb_cct_0x4_1/config
config: {"schema":"json","name":"bedroom_milight_fut105","command_topic":"milight/commands/0x4/rgb_cct/1","state_topic":"milight/state/0x4/rgb_cct/1","device":{"manufacturer":"esp8266_milight_hub","sw_version":"unknown","identifiers":[1754929,4,"rgb_cct",1]},"availability_topic":"milight/client_status","payload_available":"connected","payload_not_available":"disconnected","brightness":true,"effect":true,"effect_list":["night_mode","white_mode","0","1","2","3","4","5","6","7","8"],"rgb":true,"color_temp":true}
Printing message in parts because it's too large for the packet buffer (506 bytes)  Wrote 128 bytes
  Wrote 128 bytes
  Wrote 128 bytes
  Wrote 122 bytes
HomeAssistantDiscoveryClient: adding discoverable device: deathstar_milight_fut016...
  topic: homeassistant/light/milight_hub_1754929/rgbw_0x5_1/config
config: {"schema":"json","name":"deathstar_milight_fut016","command_topic":"milight/commands/0x5/rgbw/1","state_topic":"milight/state/0x5/rgbw/1","device":{"manufacturer":"esp8266_milight_hub","sw_version":"unknown","identifiers":[1754929,5,"rgbw",1]},"availability_topic":"milight/client_status","payload_available":"connected","payload_not_available":"disconnected","brightness":true,"effect":true,"effect_list":["night_mode","white_mode","0","1","2","3","4","5","6","7","8"],"rgb":true}
Printing message in parts because it's too large for the packet buffer (481 bytes)  Wrote 128 bytes
  Wrote 128 bytes
  Wrote 128 bytes
  Wrote 97 bytes
HomeAssistantDiscoveryClient: adding discoverable device: dinner_table_milight...
  topic: homeassistant/light/milight_hub_1754929/rgb_cct_0x2_1/config
config: {"schema":"json","name":"dinner_table_milight","command_topic":"milight/commands/0x2/rgb_cct/1","state_topic":"milight/state/0x2/rgb_cct/1","device":{"manufacturer":"esp8266_milight_hub","sw_version":"unknown","identifiers":[1754929,2,"rgb_cct",1]},"availability_topic":"milight/client_status","payload_available":"connected","payload_not_available":"disconnected","brightness":true,"effect":true,"effect_list":["night_mode","white_mode","0","1","2","3","4","5","6","7","8"],"rgb":true,"color_temp":true}
Printing message in parts because it's too large for the packet buffer (504 bytes)  Wrote 128 bytes
  Wrote 128 bytes
  Wrote 128 bytes
  Wrote 120 bytes
HomeAssistantDiscoveryClient: adding discoverable device: kitchen2...
  topic: homeassistant/light/milight_hub_1754929/cct_0x7_1/config
config: {"schema":"json","name":"kitchen2","command_topic":"milight/commands/0x7/cct/1","state_topic":"milight/state/0x7/cct/1","device":{"manufacturer":"esp8266_milight_hub","sw_version":"unknown","identifiers":[1754929,7,"cct",1]},"availability_topic":"milight/client_status","payload_available":"connected","payload_not_available":"disconnected","brightness":true,"effect":true,"effect_list":["night_mode"],"color_temp":true}
Printing message in parts because it's too large for the packet buffer (420 bytes)  Wrote 128 bytes
  Wrote 128 bytes
  Wrote 128 bytes
[... 4350 lines cut ...]
  Wrote 128 bytes
  Wrote 128 bytes
  Wrote 122 bytes
HomeAssistantDiscoveryClient: adding discoverable device: deathstar_milight_fut016...
  topic: homeassistant/light/milight_hub_1754929/rgbw_0x5_1/config
config: {"schema":"json","name":"deathstar_milight_fut016","command_topic":"milight/commands/0x5/rgbw/1","state_topic":"milight/state/0x5/rgbw/1","device":{"manufacturer":"esp8266_milight_hub","sw_version":"unknown","identifiers":[1754929,5,"rgbw",1]},"availability_topic":"milight/client_status","payload_available":"connected","payload_not_available":"disconne [... cut ...]
config: {"schema":"json","name":"kitchen2","command_topic":"milight/commands/0x7/cct/1","state_topic":"milight/state/0x7/cct/1","device":{"manufacturer":"esp8266_milight_hub","sw_version":"unknown","identifiers":[1754929,7,"cct",1]},"availability_topic":"milight/client_status","payload_available":"connected","payload_not_available":"disconnected","brightness":true,"effect":true,"effect_list":["night_mode"],"color_temp":true}
Printing message in parts because it's too large for the packet buffer (420 bytes)  Wrote 128 bytes
  Wrote 128 bytes
  Wrote 128 bytes
  Wrote 36 bytes
HomeAssistantDiscoveryClient: adding discoverable device: kitche

[16 more replies]

closed 2020-07-01 by sidoh